OpenKinect/libfreenect2 #343: [Bug][Windows] Starting two devices causes runtime error

chipweinberger, 2015-07-22:
This error can be reproduced with the code below on windows. Built today with VS 2013 on windows 8.1. Protonect works flawlessly. Device enumeration finds both devices and both devices succeed in opening.

But when a second device is started an access violation at libfreenect2.cpp line 461 (  ir_camera_params_.fx = ir_p->fx;) gets thrown.

	int main(int argc, char *argv[])
	{
	
	
	  libfreenect2::Freenect2 freenect2;
	  libfreenect2::Freenect2Device *dev = 0;
	  libfreenect2::PacketPipeline *pipeline = 0;
	
	  libfreenect2::Freenect2Device *dev2 = 0;
	  libfreenect2::PacketPipeline *pipeline2 = 0;
	
	  dev = freenect2.openDevice(0);
 	 dev2 = freenect2.openDevice(1);
	  
	
	  dev->start();
 	 dev2->start();//error gets thrown here
	
 	 
	  dev->stop();
	  dev->close();

	dev->stop();
	  dev->close();
	
	  return 0
}

chipweinberger, 2015-07-24:
This can be resolved by starting two processes of libfreenect. One for each device. I believe the issue is race conditions in libusb but I am not sure. At the very least error checking code should be handled to the data returned from libusb to prevent a crash here. This issue should remain open IMO.

floe, 2015-07-24:
Not sure if this is the reason for this specific bug, but you definitely need to call `freenect2.enumerateDevices()` before you open them.

Pk1HAlpha, 2015-08-31:
Same issue here.
I've made some testing, if both the devices are started the program fails.
However, if the first one is started and then closed,  then the second one can be started without problems. Obviously this does not solve you problem..

I've solved, as suggeste in an other thread, creating 2 different processes. In this configuration I manage to get both the sensor working (just RGB, if I try to get also the DEPTH the second sensors stop the acquisition.. i think it is a bus problem)
(Notebook: Alienware M14X R2) 

Now I am looking for a PC with multiple PCI to test multiple kinect with depth acquisition..

floe, 2015-11-09:
Is this fixed with 0.1-rc1? If not, please re-open.

closed 2015-11-09 by floe