OpenKinect/libfreenect2 #346: can I get the intrinsic parameters of the kinect depth camera

huayi, 2015-07-23:
Hi can I get the intrinsic parameters of the kinect depth camera, e.g. focal length, principal point and distortion parameters. thanks a lot!

HenningJ, 2015-07-23:
There's a method doing exactly that:
[`libfreenect2::Freenect2Device::IrCameraParams getIrCameraParams()`](https://github.com/OpenKinect/libfreenect2/blob/master/examples/protonect/include/libfreenect2/libfreenect2.hpp#L84)

huayi, 2015-07-24:
struct ColorCameraParams
    {
        float fx, fy, cx, cy;

        float shift_d, shift_m;

        float mx_x3y0; // xxx
        float mx_x0y3; // yyy
        float mx_x2y1; // xxy
        float mx_x1y2; // yyx
        float mx_x2y0; // xx
        float mx_x0y2; // yy
        float mx_x1y1; // xy
        float mx_x1y0; // x
        float mx_x0y1; // y
        float mx_x0y0; // 1

        float my_x3y0; // xxx
        float my_x0y3; // yyy
        float my_x2y1; // xxy
        float my_x1y2; // yyx
        float my_x2y0; // xx
        float my_x0y2; // yy
        float my_x1y1; // xy
        float my_x1y0; // x
        float my_x0y1; // y
        float my_x0y0; // 1
    };
    struct ColorCameraParams
    {
        float fx, fy, cx, cy;
    
        float shift_d, shift_m;
    
        float mx_x3y0; // xxx
        float mx_x0y3; // yyy
        float mx_x2y1; // xxy
        float mx_x1y2; // yyx
        float mx_x2y0; // xx
        float mx_x0y2; // yy
        float mx_x1y1; // xy
        float mx_x1y0; // x
        float mx_x0y1; // y
        float mx_x0y0; // 1
    
        float my_x3y0; // xxx
        float my_x0y3; // yyy
        float my_x2y1; // xxy
        float my_x1y2; // yyx
        float my_x2y0; // xx
        float my_x0y2; // yy
        float my_x1y1; // xy
        float my_x1y0; // x
        float my_x0y1; // y
        float my_x0y0; // 1
    };
hi. what's the meaning of the parameters  "shift_d, shift_m" and "mx_x3y0 ... my_x0y0". can i get the Rotation Martrix and Translation Vector between the depth camera coordinate and the color camera coordinate.  whether can I get these parameters from Kinect-for-windows-v2?

floe, 2015-07-24:
No, you can't get these parameters directly, they are not provided by the device. If you want to perform registration between color and depth camera image, you might want to look at the `Registration` class which does exactly that (see `Protonect.cpp` for how to use it).

huayi, 2015-07-25:
Thank you very much !

huayi, 2015-07-29:
hi. what's the relationship among "Camera Space", "Depth Space" and "Color Space" in Kinect2?

closed 2016-01-27 by xlz