OpenKinect/libfreenect2 #348: [Runtime] Depth stream loop/hang

AliShug, 2015-07-24:
I've been screwing around with the library for a few days now and I'm consistently getting an issue where the DepthPacketStreamParser seems to be going out of sync with the incoming data - there are fairly frequent dropped packets and missed subsequences beforehand, but it invariably ends up hanging the entire application. This starts silently, then with infinite spam of:

`[DepthPacketStreamParser::onDataReceived] subpacket too large`

When I've gone into debug mode, it looks like it's no longer detecting any footers in the data. At that point it just endlessly fills and throws away the working buffer.

The RGB parsing and processing threads keep going strong, but obviously don't update the displayed image as everything in the main thread (and depth processing thread) grinds to a halt waiting for a new depth frame.

It's interesting that the RGB threads never have a major issue - the parsing code is slightly different, and the RGB code checks for magic numbers etc in the JPEG data while the depth code seems to rely on the length of the received data to find the packet footer.

This is on a relatively low-powered laptop, and what's interesting is that the problem happens *much* more frequently with a slow debug build than release. A release build runs well for 5-15s, but the debug build will stall within about half a second of starting the feed. Cutting out everything but the depth feed makes it run much more reliably, and I get anything up to a minute or two of good video... so it seems to occur randomly, but more frequently with high CPU use. I was thinking it's probably caused by a threading issue because of this, but I haven't been able to make much progress trying to fix it :(

Testing with depth/IR disabled works flawlessly, but with the depth threads still crashing and burning in the background.

floe, 2015-07-24:
This sounds like a performance issue, are you using the default depth packet processor (CPU)? If yes, try running `Protonect cl` (for OpenCL) or `Protonect gl` (for OpenGL), both should be significantly faster.

AliShug, 2015-07-24:
This is with OpenGL! It seems to be performance triggered, but it's isolated to the depth stream parser. The depth processor isn't even getting new frames at that point, that's what stalls the application.

floe, 2015-07-24:
So while it runs, it does so smoothly? I don't think this is an issue with the parser, but perhaps rather with USB3 again... Right now, based on your description, I could only suggest to either try a different computer or a different USB3 controller :-/

AliShug, 2015-07-24:
On a release build yeah, it'll run smoothly right up until the crash. There's only one USB3 port, so it's a dedicated controller :-/ I'll have a chance to try a different computer/controller on Monday - but if I'm the only one seeing the problem I might as well try and fix it. I'm guessing the code just doesn't deal too well with dropped packets.

xlz, 2015-07-24:
Can you try redirect output of Protonect to NUL? `Protonect.exe > NUL`. I think scrolling in Windows console is blocking and very slow, and once the depth stream parser prints some warnings, it become slower for depth processing.

AliShug, 2015-07-24:
Yeah the console output probably doesn't help; I'll give it a go when I'm back in the lab.

floe, 2015-11-09:
Is this fixed with 0.1-rc1? If not, please re-open.

closed 2015-11-09 by floe

harpreetsatwal, 2016-08-09:
Hi Floe & Xlz,

I am getting sime error.

Was this fixed??

Abbyls, 2016-08-31:
@harpreetsatwal  Hi I am getting same error.

Was this fixed??