OpenKinect/libfreenect2 #349: Debugging OpenGL on Intel Graphics

HoxMoxNox, 2015-07-24:
Is there a debug option for the ./Protonect gl option? I have cl and the cpu options working and displaying feeds on all the windows, but only have RGB working on the OpenGL option, so I doubt it's a USB3 problem. I need the opengl version to work for a separate program, so I'm trying to figure out the issue it's having.

xlz, 2015-07-24:
The OpenGL feature is known to have trouble with Intel graphics. Just use OpenCL for the moment.

xlz, 2015-08-19:
I have posted all OpenGL fixes needed for Intel in #352.

xlz, 2015-08-22:
Actually there is. For more debug options, use environment variables:

```
LIBGL_DEBUG=verbose MESA_DEBUG=y INTEL_DEBUG=fbo,... ./Protonect
```
`INTEL_DEBUG` has many suboptions, see http://www.mesa3d.org/envvars.html for a full list.

floe, 2015-11-09:
Is this fixed with 0.1-rc1? If not, please re-open.

closed 2015-11-09 by floe