OpenKinect/libfreenect2 #350: [OpenGLDepthPacketProcessor] Failed to initialize GLFW

renxinhe, 2015-07-25:
I'm trying to run Protonect on an Ubuntu 14.04 laptop running with Intel graphics only this time.
I followed the build guide. Upon running `./Protonect`, I got the following output in the ssh shell:
>[Freenect2Impl] enumerating devices...
[Freenect2Impl] 12 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:3 with serial 501314743042
[Freenect2Impl] found 1 devices
[OpenGLDepthPacketProcessor] Failed to initialize GLFW.

I tried downloading [official GLFW source code](http://www.glfw.org/download.html) and compiling it with `cmake .`, `make`, and `sudo make install`. I noticed afterwards that my `/usr/include/GLFW` does exist.

Then, instead of using SSH to access, I directly logged into the computer and ran `./Protonect`.
This time I got a slightly different message:
>[Freenect2Impl] enumerating devices...
[Freenect2Impl] 12 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:3 with serial 501314743042
[Freenect2Impl] found 1 devices
[OpenGLDepthPacketProcessor] Failed to create opengl window.

Running `./Protonect cpu` will show lots of data being read, but no GUI window will pop up.
Running `./Protonect cl` will say `OpenCL pipeline is not supported!` as well as `[OpenGLDepthPacketProcessor] Failed to create opengl window.`.

Here are some system info:
`$ uname -r` --> `3.16.0-44-generic`
`$ glxinfo | grep "OpenGL version"` --> `OpenGL version string: 3.0 Mesa 10.3.2`
`$ ls /usr/include/GLFW` --> `glfw3.h  glfw3native.h`
`$ lspci` -->
>00:00.0 Host bridge: Intel Corporation 2nd Generation Core Processor Family DRAM Controller (rev 09)
00:01.0 PCI bridge: Intel Corporation Xeon E3-1200/2nd Generation Core Processor Family PCI Express Root Port (rev 09)
00:02.0 VGA compatible controller: Intel Corporation 2nd Generation Core Processor Family Integrated Graphics Controller (rev 09)
00:16.0 Communication controller: Intel Corporation 6 Series/C200 Series Chipset Family MEI Controller #1 (rev 04)
00:19.0 Ethernet controller: Intel Corporation 82579LM Gigabit Network Connection (rev 04)
00:1a.0 USB controller: Intel Corporation 6 Series/C200 Series Chipset Family USB Enhanced Host Controller #2 (rev 04)
00:1b.0 Audio device: Intel Corporation 6 Series/C200 Series Chipset Family High Definition Audio Controller (rev 04)
00:1c.0 PCI bridge: Intel Corporation 6 Series/C200 Series Chipset Family PCI Express Root Port 1 (rev b4)
00:1c.1 PCI bridge: Intel Corporation 6 Series/C200 Series Chipset Family PCI Express Root Port 2 (rev b4)
00:1c.3 PCI bridge: Intel Corporation 6 Series/C200 Series Chipset Family PCI Express Root Port 4 (rev b4)
00:1c.4 PCI bridge: Intel Corporation 6 Series/C200 Series Chipset Family PCI Express Root Port 5 (rev b4)
00:1d.0 USB controller: Intel Corporation 6 Series/C200 Series Chipset Family USB Enhanced Host Controller #1 (rev 04)
00:1f.0 ISA bridge: Intel Corporation QM67 Express Chipset Family LPC Controller (rev 04)
00:1f.2 SATA controller: Intel Corporation 6 Series/C200 Series Chipset Family 6 port SATA AHCI Controller (rev 04)
00:1f.3 SMBus: Intel Corporation 6 Series/C200 Series Chipset Family SMBus Controller (rev 04)
01:00.0 VGA compatible controller: NVIDIA Corporation GF119M [Quadro NVS 4200M] (rev a1)
03:00.0 Network controller: Intel Corporation Centrino Wireless-N 1000 [Condor Peak]
05:00.0 USB controller: ASMedia Technology Inc. ASM1042 SuperSpeed USB Host Controller
0d:00.0 System peripheral: Ricoh Co Ltd MMC/SD Host Controller (rev 08)

xlz, 2015-07-25:
You need `DISPLAY=:0 ./Protonect` over ssh. OpenGL needs access to X to create the context.

renxinhe, 2015-07-25:
>The OpenGL feature doesn't quite work with Intel graphics.
If Intel graphics doesn't support OpenGL, is there anyway that I can run Protonect? I tried default and cpu executions, both fail to load up the windows.

>You didn't build OpenCL support.
I have the headers installed. How do I build OpenCL support?

xlz, 2015-07-25:
> glxinfo | grep "OpenGL version" --> OpenGL version string: 3.0 Mesa 10.3.2

This is not useful. You need `glxinfo | grep OpenGL` to determine core profile version. It uses core profile, like this: `OpenGL core profile version string: 3.3 (Core Profile) Mesa 10.5.9`

> [OpenGLDepthPacketProcessor] Failed to create opengl window.

Now, there is one thing you can do to extract more information about this error.

Add 
```
static void error_callback(int error, const char* description)
{
    std::cerr << "GLFW: (" << error << ") " << description << std::endl;
}
```
to [opengl_depth_packet_processor.cpp#L769](
https://github.com/OpenKinect/libfreenect2/blob/master/examples/protonect/src/opengl_depth_packet_processor.cpp#L769)

Then add
```
    glfwSetErrorCallback(error_callback);
```

to [opengl_depth_packet_processor.cpp#L773](https://github.com/OpenKinect/libfreenect2/blob/master/examples/protonect/src/opengl_depth_packet_processor.cpp#L773).

This prints GLFW errors.

> I have the headers installed

You have which headers installed? Please read https://github.com/OpenKinect/libfreenect2#debianubuntu-1404-perhaps-earlier

renxinhe, 2015-07-26:
`glxinfo | grep OpenGL`
>OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) Sandybridge Mobile 
OpenGL core profile version string: 3.1 (Core Profile) Mesa 10.3.2
OpenGL core profile shading language version string: 1.40
OpenGL core profile context flags: (none)
OpenGL core profile extensions:
OpenGL version string: 3.0 Mesa 10.3.2
OpenGL shading language version string: 1.30
OpenGL context flags: (none)
OpenGL extensions:

I've implemented the GLFW error message print-out method:
>[Freenect2Impl] enumerating devices...
[Freenect2Impl] 12 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:3 with serial 501314743042
[Freenect2Impl] found 1 devices
GLFW: (65543) GLX: Failed to create context: GLXBadFBConfig
[OpenGLDepthPacketProcessor] Failed to create opengl window.

-------------------------------------------------------------------------------------------------------------------------------
>You have which headers installed?

I believe I have ran both `sudo apt-get install beignet-dev` and `sudo apt-get install opencl-headers`.

xlz, 2015-07-26:
Your core profile version is 3.1. Your OpenGL driver is not new enough.

beignet-dev 1.0+ from the PPA is required. Do you have 1.0?

renxinhe, 2015-07-26:
I ran the following commands: `sudo apt-add-repository ppa:pmjdebruijn/beignet-testing`, `sudo apt-get update`, and `sudo apt-get install beignet-dev`. After rebooting, I got the same `glxinfo | grep OpenGL` message.

Should I install `beignet-opencl-icd` instead of `beignet-dev`?

xlz, 2015-07-26:
OpenGL driver requires other packages.

beignet-opencl-icd is a dependency of beignet-dev.

renxinhe, 2015-07-26:
I do have beignet 1.0+ installed. I used the `pmjdebruijn/beignet-testing` PPA.
[image]

I looked into upgrading the Intel graphics driver to update OpenGL, but it seems that [Intel no longer supports 14.04](https://01.org/linuxgraphics/forum/graphics-installer-discussions/do-not-use-ubuntu-14.04). Is there another way that I can update OpenGL?

>OpenGL driver requires other packages.

What packages do I still need to install?

xlz, 2015-07-26:
> What packages do I still need to install?

http://stackoverflow.com/questions/21565680/how-to-enable-opengl-3-3-using-mesa-10-1-on-ubuntu

floe, 2015-11-09:
Is this fixed with 0.1-rc1? If not, please re-open.

closed 2015-11-09 by floe

jacksonkr, 2016-04-27:
`rtabmap` works fine with kinect v1 but not kinect v2. Both `Protonect cl` and `Protonect cpu` work but `Protonect gl` fails to render depth so you only see IR and RGB. 

```
$ rtabmap
[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65543 The requested client API version is unavailable
[Error] [OpenGLDepthPacketProcessor] Failed to create opengl window.
jackson@jackson-MacBookPro:~/rtabmap/build$ glxinfo | grep "OpenGL version"
OpenGL version string: 4.5.13399 Compatibility Profile Context 15.201.1151
$ lspci | grep "VGA"
01:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Venus XT [Radeon HD 8870M / R9 M270X/M370X] (rev 83)
```

mbp2015, Ubuntu 14.04 x64