OpenKinect/libfreenect2 #353: Calling delete on the frame in OnNewFrame() causes null ptr dereference. When and how to free frames?

chipweinberger, 2015-07-27:
How and when do we release frames? Doing this causes null ptr dereference =(
My listener looks like this:

bool PipedListener::onNewFrame(Frame::Type typ, Frame *f){

	HANDLE pipe;
	DWORD byte_width = f->width * f->height * f->bytes_per_pixel;

	switch (typ){
	case Frame::Type::Color:
		pipe = color_pipe;
		break;
	case Frame::Type::Depth:
		pipe = depth_pipe;
		break;
	default:
		return true;
	}

	DWORD bytes_written;
	WriteFile(
		pipe, //hFile,
		f->data,//lpBuffer,
		byte_width,
		&bytes_written, //lpNumberOfBytesWritten,
		NULL //lpOverlapped
		);

	//delete f;

}

christiankerl, 2015-07-27:
You should return true if you take ownership of the frame e.g. call delete on it. Otherwise, the caller will delete or reuse the frame... So your default case should probably return false and at the end of your method return true

chipweinberger, 2015-07-27:
Oh, thank you!! thank you!! Is documented anywhere? I feel stupid for not knowing this.

floe, 2015-07-31:
Not really documented, we should probably at least add a comment in the header file.

closed 2015-08-18 by chipweinberger

chipweinberger, 2015-12-30:
This still needs to be added to the header file btw.

floe, 2015-12-31:
Thanks for the heads-up, added: https://github.com/OpenKinect/libfreenect2/commit/4a2dfd99c873f35b08cc7754a33476b6ef39f4c6