OpenKinect/libfreenect2 #355: Building libfreenect2 make and make install failure

Mosa-, 2015-07-30:
Hi I can't make and make install the libfreenect2.

If I do the final step: "make && sudo make install" than this failure appears:

[  4%] Built target generate_resources_tool
[  9%] Building CXX object CMakeFiles/freenect2.dir/src/opencl_depth_packet_processor.cpp.o
In file included from /home/felix/libfreenect2/examples/protonect/src/opencl_depth_packet_processor.cpp:48:0:
/usr/include/CL/cl.hpp: In function ‘void cl::detail::fence()’:
/usr/include/CL/cl.hpp:1041:38: error: ‘_mm_mfence’ was not declared in this scope
     inline void fence() { _mm_mfence(); }
                                      ^
make[2]: *** [CMakeFiles/freenect2.dir/src/opencl_depth_packet_processor.cpp.o] Fehler 1
make[1]: *** [CMakeFiles/freenect2.dir/all] Fehler 2
make: *** [all] Fehler 2

Can someone help pls?

floe, 2015-07-31:
It looks like your OpenCL library isn't installed properly. Please report your GCC version, your GPU and the version of your OpenCL development packages first.

xlz, 2015-07-31:
> error: ‘_mm_mfence’ was not declared in this scope

This message was reported in #250, FYI.

Mosa-, 2015-08-03:
GCC version:
```
gcc (Ubuntu 4.8.4-2ubuntu1~14.04) 4.8.4
```

GPU:
```
VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Caicos PRO [Radeon HD 7450] [1002:677b] (prog-if 00 [VGA controller])
	Subsystem: Hewlett-Packard Company Device [103c:90b7]
	Flags: bus master, fast devsel, latency 0, IRQ 47
	Memory at e0000000 (64-bit, prefetchable) [size=256M]
	Memory at f7e20000 (64-bit, non-prefetchable) [size=128K]
	I/O ports at e000 [size=256]
	Expansion ROM at f7e00000 [disabled] [size=128K]
	Capabilities: <access denied>
	Kernel driver in use: fglrx_pci
```
OpenCL version:
```
clinfo --version
Number of platforms:				 1
  Platform Profile:				 FULL_PROFILE
  Platform Version:				 OpenCL 1.2 AMD-APP (1214.3)
  Platform Name:				 AMD Accelerated Parallel Processing
  Platform Vendor:				 Advanced Micro Devices, Inc.
  Platform Extensions:				 cl_khr_icd cl_amd_event_callback cl_amd_offline_devices

  Platform Name:				 AMD Accelerated Parallel Processing
Number of devices:				 2
  Device Type:					 CL_DEVICE_TYPE_GPU
  Vendor ID:					 1002h
  Board name:					 AMD Radeon HD 7400 Series
  Device Topology:				 PCI[ B#1, D#0, F#0 ]
  Max compute units:				 2
  Max work items dimensions:			 3
    Max work items[0]:				 256
    Max work items[1]:				 256
    Max work items[2]:				 256
  Max work group size:				 256
  Preferred vector width char:			 16
  Preferred vector width short:			 8
  Preferred vector width int:			 4
  Preferred vector width long:			 2
  Preferred vector width float:			 4
  Preferred vector width double:		 0
  Native vector width char:			 16
  Native vector width short:			 8
  Native vector width int:			 4
  Native vector width long:			 2
  Native vector width float:			 4
  Native vector width double:			 0
  Max clock frequency:				 625Mhz
  Address bits:					 32
  Max memory allocation:			 134217728
  Image support:				 Yes
  Max number of images read arguments:		 128
  Max number of images write arguments:		 8
  Max image 2D width:				 16384
  Max image 2D height:				 16384
  Max image 3D width:				 2048
  Max image 3D height:				 2048
  Max image 3D depth:				 2048
  Max samplers within kernel:			 16
  Max size of kernel argument:			 1024
  Alignment (bits) of base address:		 2048
  Minimum alignment (bytes) for any datatype:	 128
  Single precision floating point capability
    Denorms:					 No
    Quiet NaNs:					 Yes
    Round to nearest even:			 Yes
[... 95 lines cut ...]
    Out-of-Order:				 No
    Profiling :					 Yes
  Platform ID:					 0xb739dd00
  Name:						 Intel(R) Core(TM) i5-3470 CPU @ 3.20GHz
  Vendor:					 GenuineIntel
  Device OpenCL C version:			 OpenCL C 1.2 
  Driver version:				 1214.3 (sse2,avx)
  Profile:					 FULL_PROFILE
  Version:					 OpenCL 1.2 AMD-APP (1214.3)
  Extensions:					 cl_khr_fp64 cl_amd_fp64 cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_gl_sharing cl_ext_device_fission cl_amd_device_attribute_query cl_amd_vec3 cl_amd_printf cl_amd_media_ops cl_amd_media_ops2 cl_amd_popcnt
```

OpenGL version:
```
OpenGL version string: 4.4.13374 Compatibility Profile Context 15.20.1013
```

I read this issue https://github.com/OpenKinect/libfreenect2/issues/250 and add 
in /libfreenect2/examples/protonect/CMakeLists.txt  at line 18 and 20
```
    SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -msse -msse2 -msse3")
  ELSEIF(COMPILER_SUPPORTS_CXX0X)
    SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++0x -msse -msse2 -msse3")
```
It works, I can start protonect, but if I used it with ROS (kinect2_bridge)  after a certain time I get a segmentation fault.

floe, 2015-08-06:
OK, so since protonect now works and the leftover issues are related to ROS, please continue discussion there as we're not familiar with the ROS adapter.

closed 2015-08-06 by floe

xlz, 2015-09-07:
This bug, with #139 and #250 can be attributed to the same issue: `cl.hpp` requires SSE intructions `mfence` (`_mm_mfence()`), but somehow SSE is not enabled by default in the compiler, or it is simply not supported by the platform (ARM).

This is an issue in cl.hpp 1.2.6 shipped by Ubuntu 14.04. cl.hpp 1.2.7 has fixed it. Plus, Debian testing no longer ships cl.hpp. I'd say let's include `cl.hpp` at `include/internal/CL/cl.hpp`. Its license should be compatible.