OpenKinect/libfreenect2 #358: grab one frame at a time?

pammirato, 2015-08-07:
Is there a way to grab one frame at a time instead of constantly streaming?

I am putting many kinects on a robot, and need to grab a frame(RGBD), then move, then grab another frame, etc. 
When constantly streaming from multiple kinects things run slow(I am running a motion planner, SLAM, etc. while getting data from the kinects.). It would be great to be able to "take a picture" instead of getting a constant stream

xlz, 2015-08-08:
An option of Protonect to save a snapshot and exit. Is that what you want?

pammirato, 2015-08-08:
id like to save a snapshot, then another, then another, etc. I think exiting would require me to restart, which is too much overhead.

xlz, 2015-08-08:
The only way to implement a query like interface is restarting the stream.

xlz, 2016-01-01:
I think this is already possible for users given the current interfaces. You can call waitForNewFrame() and then dev->stop() and do other things. When you need a new frame you call waitForNewFrame() again.

closed 2016-01-04 by floe