OpenKinect/libfreenect2 #359: X server found. dri2 connection failed!

zfxw0206, 2015-08-10:
When I run "./Protonect cl" , I got the following message:

~/libfreenect2/examples/protonect/bin$ ./Protonect 
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 14 usb devices connected
[Freenect2Impl] found valid Kinect v2 @2:3 with serial 503384541942
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 08 12 0e 47 4d 46 31 35 39 2e 31    31332e2e2e2e2e2e474d463135392e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  7f 2e 00 00                                        7f2e2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  7f 2e 00 00                                        7f2e2e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
device serial: 503384541942
device firmware: 4.3.3912.0.7
[RgbPacketStreamParser::onDataReceived] packetsize or sequence doesn't match!
[DepthPacketStreamParser::onDataReceived] not all subsequences received 0
[DepthPacketStreamParser::onDataReceived] skipping depth packet
[RgbPacketStreamParser::onDataReceived] skipping rgb packet!
[OpenGLDepthPacketProcessor] avg. time: 8.54584ms -> ~117.016Hz
[TurboJpegRgbPacketProcessor] avg. time: 19.8282ms -> ~50.4332Hz
^C[Freenect2DeviceImpl] stopping...
[Freenect2DeviceImpl] disabling usb transfer submission...
[Freenect2DeviceImpl] canceling usb transfers...
[Freenect2DeviceImpl] stopped
[Freenect2DeviceImpl] closing...
[Freenect2DeviceImpl] releasing usb interfaces...
[Freenect2DeviceImpl] deallocating usb transfer pools...
[Freenect2DeviceImpl] closing usb device...
[Freenect2DeviceImpl] closed
[Freenect2DeviceImpl] closing...
[Freenect2DeviceImpl] already closed, doing nothing
visiongm@visiongm-desktop:~/libfreenect2/examples/protonect/bin$ ./Protonect cl
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 14 usb devices connected
[Freenect2Impl] found valid Kinect v2 @2:3 with serial 503384541942
[Freenect2Impl] found 1 devices
X server found. dri2 connection failed! 
Trying to open directly...Device open failed

Then I run "ldd Protonect|grep usb", I got this
	libusb-1.0.so.0 => /home/visiongm/libfreenect2/examples/protonect/lib/libusb-1.0.so.0 (0x00007fcb7fa6c000)

how to fix this problem?

xlz, 2015-08-10:
Need output of `ls -l /dev/dri/` and `id`. You should be in the `video` group and `/dev/dri/*` should have rw access for `video` group. If not, add yourself to `video` and relogin.

If this does not work, run `strace ./Protonect cl` and post the end of the log.

zfxw0206, 2015-08-11:
Thank you very much. I think there may be something wrong of the system. Then I reinstall the system,   it works fine now.

closed 2015-08-12 by floe

melodybinbin, 2018-06-26:
sudo apt-get remove beignet*

Try this command. It is works for me.