OpenKinect/libfreenect2 #363: Installation Error in Ubuntu 14.04

mgandhi93, 2015-08-19:
I am up to the installation part involving building the Protonect example. When I run cmake it successfully builds. However, when I run make, I get this error:

Linking CXX executable ../bin/Protonect
/usr/bin/ld: CMakeFiles/Protonect.dir/viewer.cpp.o: undefined reference to symbol 'glfwInit'
/usr/local/lib/libglfw.so.3: error adding symbols DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [./bin/Protonect] Error 1
make[1]: *** [CMakeFiles/Protonect.dir/all] Error 2
make: *** [all] Error 2

Any ideas on how to fix this error? Thanks.

xlz, 2015-08-20:
How is libglfw installed? `/usr/local/lib/libglfw.so.3` does not look like you followed README.md.

mgandhi93, 2015-08-20:
I followed README.md exactly unless I overlooked something. I double-checked by uninstalling everything and attempting to reinstall and got the same error. I did try installing on another Ubuntu computer and it worked perfectly on that one, which is a bit peculiar. Does this command "sudo dpkg -i libglfw3*_3.0.4-1_*.deb" install libglfw? In that case I copied that command exactly while in libfreenect2/depends. Thanks.

xlz, 2015-08-20:
Where does `/usr/local/lib/libglfw.so.3` come from? It definitely conflicted with the /usr/lib one.

mgandhi93, 2015-08-20:
Thank you for pointing that out. I am using a shared computer so it looks like someone had already installed libglfw. I removed it from that location and that specific error has been solved. However, now I am receiving the error: 
/usr/bin/ld: CMakeFiles/Protonect.dir/viewer.cpp.o: undefined reference to symbol 'glDrawArrays'
//usr/bin/nvidia-340/libGL.so.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [../bin/Protonect] Error 1
make[1]: *** CMakeFiles/Protonect.dir/all] Error 2
make: *** [all] Error 2

I did install the latest nvidia drivers via Software Center. I was looking into uninstalling that but I am having trouble finding it in my Software Center (it does not even show up in the history for some reason).

xlz, 2015-08-20:
Why is `libGL.so.1` found in `/usr/bin` ?! There is something horribly wrong with your graphics libraries.

mgandhi93, 2015-08-20:
I found  and removed the nvidia driver and reinstalled the latest version from ppa:xorg-edgers as stated in the README. I also uninstalled and reinstalled libfreenect2, again closely following all directions in README and this time received this error:

/usr/bin/ld: CMakeFiles/Protonect.dir/viewer.cpp.o: undefined reference to symbol 'glfwInit'
//usr/lib/x86_64-linux-gnu/libglfw.so.3: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [../bin/Protonect] Error 1
make[1]: *** [CMakeFiles/Protonect.dir/all] Error 2
make: *** [all] Error 2

xlz, 2015-08-20:
Output of `cmake ..` and relevant bits in `make VERBOSE=1` would help.

mgandhi93, 2015-08-20:
Output of cmake .. is 
-- using tinythread as threading library
Linking with these libraries: usb-1.0;/usr/lib/x86_64-linux-gnu/libturbojpeg.so.0;pthread;glfw;/usr/lib/x86_64-linux-gnu/libGL.so;/usr/lib/x86_64-linux-gnu/libOpenCL.so
-- Configuring done
-- Generating done
-- Build files have been written to /home/me/Desktop/libfreenect2/examples/protonect/build

output of make VERBOSE=1 is
/usr/local/bin/cmake -H/home/me/Desktop/libfreenect2/examples/protonect -B/home/me/Desktop/libfreenect2/examples/protonect/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/local/bin/cmake -E cmake_progress_start /home/me/Desktop/libfreenect2/examples/protonect/build/CMakeFiles /home/me/Desktop/libfreenect2/examples/protonect/build/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory `/home/me/Desktop/libfreenect2/examples/protonect/build'
make -f CMakeFiles/generate_resources_tool.dir/build.make CMakeFiles/generate_resources_tool.dir/depend
make[2]: Entering directory `/home/me/Desktop/libfreenect2/examples/protonect/build'
cd /home/me/Desktop/libfreenect2/examples/protonect/build && /usr/local/bin/cmake -E cmake_depends "Unix Makefiles" /home/me/Desktop/libfreenect2/examples/protonect /home/me/Desktop/libfreenect2/examples/protonect /home/me/Desktop/libfreenect2/examples/protonect/build /home/me/Desktop/libfreenect2/examples/protonect/build /home/me/Desktop/libfreenect2/examples/protonect/build/CMakeFiles/generate_resources_tool.dir/DependInfo.cmake --color=
make[2]: Leaving directory `/home/me/Desktop//libfreenect2/examples/protonect/build'
make -f CMakeFiles/generate_resources_tool.dir/build.make CMakeFiles/generate_resources_tool.dir/build
make[2]: Entering directory `/home/me/Desktop/libfreenect2/examples/protonect/build'
make[2]: Nothing to be done for `CMakeFiles/generate_resources_tool.dir/build'.
make[2]: Leaving directory `/home/me/Desktop/libfreenect2/examples/protonect/build'
/usr/local/bin/cmake -E cmake_progress_report /home/me/Desktop/libfreenect2/examples/protonect/build/CMakeFiles  26
[  3%] Built target generate_resources_tool
make -f CMakeFiles/freenect2.dir/build.make CMakeFiles/freenect2.dir/depend
make[2]: Entering directory `/home/me/Desktop/libfreenect2/examples/protonect/build'
cd /home/me/Desktop/libfreenect2/examples/protonect/build && /usr/local/bin/cmake -E cmake_depends "Unix Makefiles" /home/me/Desktop/libfreenect2/examples/protonect /home/me/Desktop/libfreenect2/examples/protonect /home/me/Desktop/libfreenect2/examples/protonect/build /home/me/Desktop/libfreenect2/examples/protonect/build /home/me/Desktop/libfreenect2/examples/protonect/build/CMakeFiles/freenect2.dir/DependInfo.cmake --color=
make[2]: Leaving directory `/home/me/Desktop/libfreenect2/examples/protonect/build'
make -f CMakeFiles/freenect2.dir/build.make CMakeFiles/freenect2.dir/build
make[2]: Entering directory `/home/me/Desktop/libfreenect2/examples/protonect/build'
make[2]: Nothing to be done for `CMakeFiles/freenect2.dir/build'.
make[2]: Leaving directory `/home/me/Desktop/libfreenect2/examples/protonect/build'
/usr/local/bin/cmake -E cmake_progress_report /home/me/Desktop/libfreenect2/examples/protonect/build/CMakeFiles  4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25
[ 88%] Built target freenect2
make -f CMakeFiles/freenect2shared.dir/build.make CMakeFiles/freenect2shared.dir/depend
make[2]: Entering directory `/home/me/Desktop/libfreenect2/examples/protonect/build'
cd /home/me/Desktop/libfreenect2/examples/protonect/build && /usr/local/bin/cmake -E cmake_depends "Unix Makefiles" /home/me/Desktop/libfreenect2/examples/protonect /home/me/Desktop/libfreenect2/examples/protonect /home/me/Desktop/libfreenect2/examples/protonect/build /home/me/Desktop/libfreenect2/examples/protonect/build /home/me/Desktop/libfreenect2/examples/protonect/build/CMakeFiles/freenect2shared.dir/DependInfo.cmake --color=
make[2]: Leaving directory `/home/me/Desktop/libfreenect2/examples/protonect/build'
make -f CMakeFiles/freenect2shared.dir/build.make CMakeFiles/freenect2shared.dir/build
make[2]: Entering directory `/home/me/Desktop/libfreenect2/examples/protonect/build'
make[2]: Nothing to be done for `CMakeFiles/freenect2shared.dir/build'.
make[2]: Leaving directory `/home/me/Desktop/libfreenect2/examples/protonect/build'
/usr/local/bin/cmake -E cmake_progress_report /home/me/Desktop/libfreenect2/examples/protonect/build/CMakeFiles 
[ 88%] Built target freenect2shared
make -f CMakeFiles/Protonect.dir/build.make CMakeFiles/Protonect.dir/depend
make[2]: Entering directory `/home/me/Desktop/libfreenect2/examples/protonect/build'
cd /home/me/Desktop/libfreenect2/examples/protonect/build && /usr/local/bin/cmake -E cmake_depends "Unix Makefiles" /home/me/Desktop/libfreenect2/examples/protonect /home/me/Desktop/libfreenect2/examples/protonect /home/me/Desktop/libfreenect2/examples/protonect/build /home/me/Desktop/libfreenect2/examples/protonect/build /home/me/Desktop//libfreenect2/examples/protonect/build/CMakeFiles/Protonect.dir/DependInfo.cmake --color=
make[2]: Leaving directory `/home/Desktop/libfreenect2/examples/protonect/build'
make -f CMakeFiles/Protonect.dir/build.make CMakeFiles/Protonect.dir/build
make[2]: Entering directory `/home/me/Desktop/libfreenect2/examples/protonect/build'
Linking CXX executable ../bin/Protonect
/usr/local/bin/cmake -E cmake_link_script CMakeFiles/Protonect.dir/link.txt --verbose=1
[... cut ...]
/usr/bin/ld: CMakeFiles/Protonect.dir/viewer.cpp.o: undefined reference to symbol 'glDrawArrays'
//usr/lib/nvidia-346/libGL.so.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [../bin/Protonect] Error 1
make[2]: Leaving directory `/home/me/Desktop/libfreenect2/examples/protonect/build'
make[1]: *** [CMakeFiles/Protonect.dir/all] Error 2
make[1]: Leaving directory `/home/me/Desktop/libfreenect2/examples/protonect/build'
make: *** [all] Error 2

xlz, 2015-08-20:
`warning: libglfw.so.3, needed by ../lib/libfreenect2.so, not found` Why?

You should have `/usr/lib/x86_64-linux-gnu/libglfw.so.3` installed, yes?

mgandhi93, 2015-08-20:
I accidentally moved it while removing duplicates, but it is now installed there. The error I am still getting, however, when I run make VERBOSE=1 is

usr/bin/ld: CMakeFiles/Protonect.dir/viewer.cpp.o: undefined reference to symbol 'glfwInit' 
//usr/lib/x86_64-linux-gnu/libglfw.so.3: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [../bin/Protonect] Error 1
make[2]: Leaving directory '/home/me/Desktop/libfreenect2/examples/protonect/build'
make[1]: *** [CMakeFiles/Protonect.dir/all] Error 2
make[1]: Leaving directory 'home/me/Desktop/libfreenect2/examples/protonect/build'
make: *** [all] Error 2

xlz, 2015-08-21:
`-lusb-1.0 /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0 -lpthread -lglfw -lGL` is supposed to appear after `-rdynamic ../lib/libfreenect2.so`.

What is your CMake version?

mgandhi93, 2015-08-21:
2.8.12

xlz, 2015-08-21:
Can you try 2.8.12.2? This is the Ubuntu 14.04 version. Yours seems to be some local version in `/usr/local/bin`

mgandhi93, 2015-08-21:
That fixed it! Thanks so much!

xlz, 2015-08-22:
This bug can be reproduced with cmake 2.8.12-0ubuntu2, not with 2.8.12.1-0ubuntu1. Thus we need to raise the cmake version requirement.

closed 2015-11-09 by floe