OpenKinect/libfreenect2 #366: Protonect freezes on waitForNewFrame

mbk74, 2015-08-21:
Hi guys
I have built protonect using instructions from the description. Platform is Windows 8.1 x64. Kinect was found successfully and then the app froze on waitForNewFrame. 
I paused the app and took a look on the threads. 
I added screenshot. Is anybody know why it happened?
[image]

xlz, 2015-08-21:
Do you have the console output?

Alex-Canales, 2015-08-21:
Not sure if I have the same problem than mbk74. I am using an Ubuntu  14.04  LTS with nearly nothing installed so I don't have access to a debugger.

However, when I try Protonect, I have a window with a black screen. I suppose it could be the same problem than him.

Here what the console output displays:

        [Freenect2Impl] enumerating devices...
        [Freenect2Impl] 15 usb devices connected
        [Freenect2Impl] found valid Kinect v2 @3:3 with serial 502698541942
        [Freenect2Impl] found 1 devices
        [Freenect2DeviceImpl] opening...
        [Freenect2DeviceImpl] opened
        [Freenect2DeviceImpl] starting...
        [Freenect2DeviceImpl] ReadData0x14 response
        92 bytes of raw data
        0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
        0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
        0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
        0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
        0x0040:  31 33 00 00 00 03 01 13 47 4d 46 39 31 38 2e 31    31332e2e2e2e2e2e474d463931382e31
        [Freenect2DeviceImpl] ReadStatus0x090000 response
        4 bytes of raw data
        0x0000:  01 26 00 00                                        2e262e2e

        [Freenect2DeviceImpl] ReadStatus0x090000 response
        4 bytes of raw data
        0x0000:  03 26 00 00                                        2e262e2e

        [Freenect2DeviceImpl] enabling usb transfer submission...
        [Freenect2DeviceImpl] submitting usb transfers...

        [TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO

 The same last message repeat until:

        [Freenect2DeviceImpl] started
        device serial: 502698541942
        device firmware: 2.3.3912.0.7
        [TurboJpegRgbPacketProcessor] avg. time: 21.9224ms -> ~45.6155Hz
        [TurboJpegRgbPacketProcessor] avg. time: 21.6293ms -> ~46.2336Hz
        [TurboJpegRgbPacketProcessor] avg. time: 21.3963ms -> ~46.737Hz

The same output repeat until I kill the process using `sudo pkill Protonect`.

 Also, by trying again and again this program, I "manage" to "make an different USB port crash", nothing could be used on this port but an other USB port still worked.
I also had (for the first time of my lire) a kernel panic when I killed Protonect (fatal exception in interruption).

Alex-Canales, 2015-08-21:
I also try with two models of kinect: one for Windows, and other for Xbox.

xlz, 2015-08-21:
@Alex-Canales Can you please open a new issue and post your problem there (and delete your comments here)? It is quite hard to keep track of two issues in one issue.

Alex-Canales, 2015-08-21:
Sorry, I just added comments in this issue here because I thought it could be related. I open a [new issue](https://github.com/OpenKinect/libfreenect2/issues/367).

xlz, 2015-08-22:
Can you set up an environment variable `LIBUSB_DEBUG=4` before running, and post the console output?

mbk74, 2015-08-22:
Here it is
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 16 usb devices connected
[Freenect2Impl] found valid Kinect v2 @3:2 with serial 501774441942
[Freenect2Impl] found 1 devices
[OpenCLDepthPacketProcessor::listDevice]  devices:
  0: Intel(R) Core(TM) i5-4570S CPU @ 2.90GHz (CPU)[Intel(R) Corporation]
  1: Intel(R) HD Graphics 4600 (GPU)[Intel(R) Corporation]
  2: Intel(R) Core(TM) i5-4570S CPU @ 2.90GHz (CPU)[Intel(R) Corporation]
[OpenCLDepthPacketProcessor::init]  selected device: Intel(R) HD Graphics 4600 (
GPU)[Intel(R) Corporation]
[OpenCLDepthPacketProcessor::buildProgram] building OpenCL program...
[OpenCLDepthPacketProcessor::buildProgram] OpenCL program built successfully
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e
2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e
2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e
2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e
802e2e2e2e2e
0x0040:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e
2e2e2e2e2e2e
0x0050:  00 00 00 00 00 00 00 00 00 00 00 00                2e2e2e2e2e2e2e2e2e2e
2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  00 20 00 00                                        2e202e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  00 20 00 00                                        2e202e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
device serial: 501774441942
device firmware: 4.3.3912.0.7

xlz, 2015-08-22:
This is a libusb issue. Need console output with environment variable `LIBUSB_DEBUG=4` (probably redirect it to a file because it will be large).

mbk74, 2015-08-23:
This error repeats many times
only  "74" is changing
[ 0.209700] [000015f8] libusbx: debug [libusb_get_next_timeout] next timeout in 0.996822s
[ 0.209700] [000015f8] libusbx: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.209700] [000007b8] libusbx: debug [winusbx_submit_iso_transfer] matched endpoint 84 with interface 1
[ 0.209700] [000015f8] libusbx: debug [handle_events] poll() 74 fds with timeout in 100ms
[ 0.209700] [000007b8] libusbx: debug [winusbx_submit_iso_transfer] reading 8 iso packets
[ 0.209700] [000015f8] libusbx: debug [handle_events] poll() returned -1
[ 0.209700] [000015f8] libusbx: error [handle_events] poll failed -1 err=5

xlz, 2015-08-23:
`poll failed -1 err=5`

Error 5 is `EIO` input/output error. This comes from `usbi_poll()` in `libusb/os/poll_windows.c`.

`EIO` is not very informative. You can do something:

* Try updating to the latest libusb, `git fetch origin; git checkout origin/master; git merge joshblake/winiso`. If that doesn't work,
* In `libusb/os/poll_windows.c`, uncomment `DEBUG_POLL_WINDOWS`, and add `poll_dbg("GetLastError() = %#x\n", GetLastError())` after `errno = EIO;` in `usbi_poll()`. And see what you get.

mbk74, 2015-08-23:
I  did it
Output repeat following:
[ 0.348719] [00001710] libusb: debug [usbi_remove_pollfd] remove fd 32
[ 0.348719] [00001710] libusb: debug [usbi_write] set pipe event (fd = 0, thread = 00001710)
[ 0.348719] [00001710] libusb: debug [windows_transfer_callback] handling I/O completion with errcode 0, size 0
[ 0.348719] [00001710] libusb: debug [usbi_handle_transfer_completion] transfer 0000004C7DDE1EC0 has callback 00007FF835C02B5D
[ 0.348719] [00001710] libusb: debug [libusb_submit_transfer] transfer 0000004C7DDE1EC0
[ 0.348719] [00001710] libusb: debug [winusbx_submit_iso_transfer] matched endpoint 84 with interface 1
[ 0.348719] [00001710] libusb: debug [winusbx_submit_iso_transfer] reading 8 iso packets
[ 0.348719] [00001710] libusb: debug [usbi_add_pollfd] add fd 32 events 1
[ 0.348719] [00001710] libusb: debug [libusb_get_next_timeout] next timeout in 0.982340s
[ 0.348719] [00001710] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.348719] [00001710] libusb: debug [handle_events] poll fds modified, reallocating
[ 0.348719] [00001710] libusb: debug [usbi_read] clr pipe event (fd = 0, thread = 00001710)
[ 0.348719] [00001710] libusb: debug [handle_events] poll() 81 fds with timeout in 100ms
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[0]=0: (overlapped=0000004C7842BD80) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[1]=1: (overlapped=0000004C79FDA160) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[2]=2: (overlapped=0000004C79FDA2E0) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[3]=3: (overlapped=0000004C79FDA3A0) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[4]=4: (overlapped=0000004C79FDA520) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[5]=5: (overlapped=0000004C79FDA640) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[6]=6: (overlapped=0000004C784FD6F0) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[7]=7: (overlapped=0000004C784FD0F0) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[8]=8: (overlapped=0000004C784FD150) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[9]=9: (overlapped=0000004C784FD810) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[10]=10: (overlapped=0000004C027A6F40) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[11]=11: (overlapped=0000004C027A68E0) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[12]=12: (overlapped=0000004C027A6A00) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[13]=13: (overlapped=0000004C027A6AC0) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[14]=14: (overlapped=0000004C027A7000) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[15]=15: (overlapped=0000004C027A6B20) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[16]=16: (overlapped=0000004C027A6B80) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[17]=17: (overlapped=0000004C027A6E20) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[18]=18: (overlapped=0000004C027A6CA0) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[19]=19: (overlapped=0000004C027A6E80) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[20]=20: (overlapped=0000004C02C50840) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[21]=33: (overlapped=0000004C02C3A060) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[22]=34: (overlapped=0000004C02C3A360) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[23]=35: (overlapped=0000004C02C3A540) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[24]=36: (overlapped=0000004C02C3A480) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[25]=37: (overlapped=0000004C02C39AC0) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[26]=38: (overlapped=0000004C02C3A3C0) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[27]=39: (overlapped=0000004C02C39F40) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[28]=40: (overlapped=0000004C02C397C0) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[29]=41: (overlapped=0000004C02C3A660) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[30]=42: (overlapped=0000004C02C3A420) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[31]=43: (overlapped=0000004C02C398E0) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[32]=44: (overlapped=0000004C02C39940) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[33]=45: (overlapped=0000004C02C39B80) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[34]=46: (overlapped=0000004C02C39DC0) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[35]=47: (overlapped=0000004C02C39E80) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[36]=48: (overlapped=0000004C02C3A000) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[37]=49: (overlapped=0000004C02C3A180) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[38]=50: (overlapped=0000004C02C3A300) got events 0001
[ 0.348719] [00001710] libusb: debug [usbi_poll] fd[39]=51: (overlapped=0000004C7DA30520) got events 0001
[... cut ...]
[ 0.349722] [00001710] libusb: debug [windows_handle_events] checking fd 16 with revents = 0000
[ 0.349722] [00001710] libusb: debug [windows_handle_events] checking fd 17 with revents = 0000
[ 0.349722] [00001710] libusb: debug [windows_handle_events] checking fd 18 with revents = 0000
[ 0.349722] [00001710] libusb: debug [windows_handle_events] checking fd 19 with revents = 0000
[ 0.349722] [00001710] libusb: debug [windows_handle_events] checking fd 20 with revents = 0000
[ 0.349722] [00001710] libusb: debug [windows_handle_events] checking fd 33 with revents = 0001

mbk74, 2015-08-23:
May be something wrong with libusbK driver?

xlz, 2015-08-23:
I think the only other information you can get is from `GetLastError()`. Otherwise I don't know.

mbk74, 2015-08-25:
This is never happens in  usbi_poll()
	if ((handles_to_wait_on == NULL) || (handle_to_index == NULL)) {
		errno = ENOMEM;
		poll_dbg("GetLastError() = %#x\n", GetLastError());
		triggered = -1;
		goto poll_exit;
	}

xlz, 2015-08-25:
Ehh,

> after errno = EIO; in usbi_poll()

mbk74, 2015-08-25:
I reinstalled all and it works now. Thank you for help.

closed 2015-08-28 by mbk74