OpenKinect/libfreenect2 #367: Protonect display a black screen (and seems to be freezed)

Alex-Canales, 2015-08-21:
Not sure if I have the same problem than [mbk74](https://github.com/OpenKinect/libfreenect2/issues/366). I am using an Ubuntu  14.04  LTS with nearly nothing installed so I don't have access to a debugger.

However, when I try Protonect, I have a window with a black screen.
Here what the console output displays:

        [Freenect2Impl] enumerating devices...
        [Freenect2Impl] 15 usb devices connected
        [Freenect2Impl] found valid Kinect v2 @3:3 with serial 502698541942
        [Freenect2Impl] found 1 devices
        [Freenect2DeviceImpl] opening...
        [Freenect2DeviceImpl] opened
        [Freenect2DeviceImpl] starting...
        [Freenect2DeviceImpl] ReadData0x14 response
        92 bytes of raw data
        0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
        0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
        0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
        0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
        0x0040:  31 33 00 00 00 03 01 13 47 4d 46 39 31 38 2e 31    31332e2e2e2e2e2e474d463931382e31
        [Freenect2DeviceImpl] ReadStatus0x090000 response
        4 bytes of raw data
        0x0000:  01 26 00 00                                        2e262e2e

        [Freenect2DeviceImpl] ReadStatus0x090000 response
        4 bytes of raw data
        0x0000:  03 26 00 00                                        2e262e2e

        [Freenect2DeviceImpl] enabling usb transfer submission...
        [Freenect2DeviceImpl] submitting usb transfers...

        [TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO

 The same last message repeat until:

        [Freenect2DeviceImpl] started
        device serial: 502698541942
        device firmware: 2.3.3912.0.7
        [TurboJpegRgbPacketProcessor] avg. time: 21.9224ms -> ~45.6155Hz
        [TurboJpegRgbPacketProcessor] avg. time: 21.6293ms -> ~46.2336Hz
        [TurboJpegRgbPacketProcessor] avg. time: 21.3963ms -> ~46.737Hz

The same output repeat until I kill the process using `sudo pkill Protonect`.

 Also, by trying again and again this program, I "manage" to "make an different USB port crash", nothing could be used on this port but an other USB port still worked.
I also had (for the first time of my lire) a kernel panic when I killed Protonect (fatal exception in interruption).

I also tried with two models of kinect: one for Windows and one for Xbox.

xlz, 2015-08-21:
What is the output of `ldd ./Protonect | grep libusb`?

xlz, 2015-08-22:
If the path starts with /usr/lib then you have linked wrong libusb. If not, use `LIBUSB_DEBUG=4 ./Protonect` and post log. Do not use sudo!!!

Alex-Canales, 2015-08-24:
Unfortunately I do not think I will be able to use anymore this machine (my tests were at work and we will certainly just used Windows for the moment since this is just a prototype project).
Thank you for your help and sorry to have wasted your time, I am closing this issue.

closed 2015-08-24 by Alex-Canales