OpenKinect/libfreenect2 #369: Low frame rate kinect v.2

jpalbae, 2015-08-22:
Hello to all. 

I'm a new user of kinect v2.0.

I got the problem of low frame rate and even at low velocity the sensor loss the following.

I have a new NVIDIA QUADRO K4200, because although is not very quick, it provides the highest driver version of Direct X and also the stereo syncronization with 3D vision kit to edit in 3D.

My motherboard is a GIGABYTE X79 UD3, which provides PCI X16 of third generation, and I think that PCI x1 is of second generation (until 5Gps).

Also I got a dedicated USB 3.0 card: ORICO USB3.0 Expansion Card (PVU3-2O2I)

When I connect my kinect to the motherboard integrated USB 3.0, the color works, but I get low frame rate in the verifier and in the 3D builder program.

When I connect it through the dedicated card, with its drivers instaled (for the XHCI controller), i get low rate and no color (it is worse than in the above case). I tested this card also in a PCI X16 slot, but it is not recognized by my system. I als activated both: EHCI and XHCI on my BIOS. But it does't solve my problem.

Do you know where is my problem?, the quadro GPU it is correct?, maybe buying the TS-PDU 3 USB 3.0 card sugested in the foro is OK (social.msdn.microsoft.com/Forums/es-ES/bb379e8b-4258-40d6-92e4-56dd95d7b0bb/confirmed-list-of-usb-30-pcie-cardslaptopsconfigurations-which-work-for-kinect-v2-during?forum=kinectv2sdk)

Thank you in advance.

xlz, 2015-08-23:
"low frame rate in the verifier"? Which part do you refer to, and how low is it? Do you have logs?

jpalbae, 2015-08-23:
Thank you by your interest with my issue xlz. 
I've disconected and packed the kinect v.2. If it's necessary I'll conect it and give you more details.

As far as I remember, when conecting the kinect to the integrate USB port, both images (color and depth) appears in the verifier. But the rate might not to overcome 20 fps and some times it does appears as disconected for less than some seconds. At the same time the tick dissapears and appears when it reconects. 

When conecting the kinect to the dedicated USB card, I can only watch the depth image (black an white) and it also gives a low rate message from time to time.

I didn't say that my CPU is i7-3820 @3.6 GHz (4 cores).

Thank you in advance if you can help me.

floe, 2015-09-02:
As indicated in the FAQ, try running `Protonect cl` or `Protonect gl` to use a faster depth packet processor.

jpalbae, 2015-09-02:
Where ca I find these FAQ?, I don't know how to run Potonect, it is an option or program inside my Open GL drivers?

floe, 2015-09-02:
It seems that you're using the Microsoft Kinect SDK, not libfreenect2. Please ask in the Microsoft forum.

closed 2015-09-02 by floe

jpalbae, 2015-09-02:
Excactly, thank you, I understand know.