OpenKinect/libfreenect2 #370: Unable to install mesa-common-dev

cpaniagua1, 2015-08-24:
Hello,
I'm trying to install libfreenect2 to work with my new Kinect2. I'm following the installation instructions for Ubuntu 14.04 but when I want to install mesa-common-dev I get the following error:

The following packages have unmet dependencies:
 libcheese-gtk23 : Depends: libclutter-gtk-1.0-0 (>= 0.91.8) but it is not going to be installed
                   Depends: libcogl15 (>= 1.15.8) but it is not going to be installed
 libcheese7 : Depends: libclutter-gst-2.0-0 (>= 0.10.0) but it is not going to be installed
              Depends: gstreamer1.0-clutter but it is not going to be installed
 libclutter-1.0-0 : Depends: libcogl-pango15 (>= 1.15.8) but it is not going to be installed
                    Depends: libcogl15 (>= 1.15.8) but it is not going to be installed

Can somebody help me with this?

xlz, 2015-08-24:
You have installed unusual versions of packages in the graphics stack previously. "it is not going to be installed" this is because it conflicts with something. You need to manually resolve the conflict. I think an usual way is update everything to newest.

cpaniagua1, 2015-08-24:
Thank you for your answer. I've updated everything to newest but I have the same problem.

floe, 2015-09-02:
Sometimes, `apt-get install -f` can be used to fix this situation.

cpaniagua1, 2015-09-02:
Thank you very much. It fixed my problem.

closed 2015-09-02 by cpaniagua1