OpenKinect/libfreenect2 #371: [Windows] Multi kinect error

moonsh, 2015-08-27:
Hello .

I am using

Windows 8.1
Visual studio 2013
Kinect V2 for windows 

I installed single kinect and it was successful .

However, when I connect second kinect it shows error like below.  I found a stuck point in code "dev1->start()" but I can't find reason.

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
[Freenect2DeviceImpl] starting...
[RgbPacketStreamParser::onDataReceived] packetsize or sequence doesn't match!
[CommandTransaction::send] bulk transfer failed! libusb error -1: LIBUSB_ERROR_I
O
[CommandTransaction::send] sent number of bytes differs from expected number! ex
pected: 20 got: 0
[CommandTransaction::send] bulk transfer failed! libusb error -1: LIBUSB_ERROR_I
O
[CommandTransaction::send] sent number of bytes differs from expected number! ex
pected: 20 got: 0
[Freenect2DeviceImpl] ReadData0x14 response
0 bytes of raw data

[CommandTransaction::send] bulk transfer failed! libusb error -1: LIBUSB_ERROR_I
O
[CommandTransaction::send] sent number of bytes differs from expected number! ex
pected: 24 got: 0
[Freenect2DeviceImpl] serial number reported by libusb 506898642542 differs from
 serial number 羲羲 in device protocol!

I changed "Protonect.cpp" code like below . I added dev1 for second kinect.

int main(int argc, char *argv[])
{
  std::string program_path(argv[0]);
  size_t executable_name_idx = program_path.rfind("Protonect");

  std::string binpath = "/";

  if(executable_name_idx != std::string::npos)
  {
    binpath = program_path.substr(0, executable_name_idx);
  }

  libfreenect2::Freenect2 freenect2;
  libfreenect2::Freenect2Device *dev = 0;
  libfreenect2::PacketPipeline *pipeline = 0;

  libfreenect2::Freenect2Device *dev1 = 0;
  libfreenect2::PacketPipeline *pipeline1 = 0;

  if(freenect2.enumerateDevices() == 0)
  {
    std::cout << "no device connected!" << std::endl;
    return -1;
  }

  std::cout << freenect2.getDeviceSerialNumber(0) << std::endl;  // 0  default kinect  
  std::cout << freenect2.getDeviceSerialNumber(1) << std::endl;  // 1   Second kinect

  std::string serial = freenect2.getDeviceSerialNumber(0);
  std::string serial1 = freenect2.getDeviceSerialNumber(1);

  for(int argI = 1; argI < argc; ++argI)
  {
    const std::string arg(argv[argI]);

    if(arg == "cpu")
    {
      if(!pipeline)
        pipeline = new libfreenect2::CpuPacketPipeline();

	}
    else if(arg == "gl")
    {
#ifdef LIBFREENECT2_WITH_OPENGL_SUPPORT
      if(!pipeline)
        pipeline = new libfreenect2::OpenGLPacketPipeline();
	
#else
      std::cout << "OpenGL pipeline is not supported!" << std::endl;
#endif
    }
    else if(arg == "cl")
    {
#ifdef LIBFREENECT2_WITH_OPENCL_SUPPORT
      if(!pipeline)
        pipeline = new libfreenect2::OpenCLPacketPipeline();
#else
      std::cout << "OpenCL pipeline is not supported!" << std::endl;
#endif
    }
    else if(arg.find_first_not_of("0123456789") == std::string::npos) //check if parameter could be a serial number
    {
      serial = arg;
	 
    }
    else
    {
      std::cout << "Unknown argument: " << arg << std::endl;
    }
  }

  for (int argI = 1; argI < argc; ++argI)
  {
	  const std::string arg1(argv[argI]);

	  if (arg1 == "cpu")
	  {
		  if (!pipeline1)
			
		  pipeline1 = new libfreenect2::CpuPacketPipeline();

	  }
	  else if (arg1 == "gl")
	  {
#ifdef LIBFREENECT2_WITH_OPENGL_SUPPORT
		  if (!pipeline1)
			 
		  pipeline1 = new libfreenect2::OpenGLPacketPipeline();
#else
		  std::cout << "OpenGL pipeline is not supported!" << std::endl;
#endif
	  }
	  else if (arg1 == "cl")
	  {
#ifdef LIBFREENECT2_WITH_OPENCL_SUPPORT
		  if (!pipeline1)
			
		  pipeline1 = new libfreenect2::OpenCLPacketPipeline();
#else
		  std::cout << "OpenCL pipeline is not supported!" << std::endl;
#endif
	  }
	  else if (arg1.find_first_not_of("0123456789") == std::string::npos) //check if parameter could be a serial number
	  {
		  serial1 = arg1;
	  }
	  else
	  {
		  std::cout << "Unknown argument: " << arg1 << std::endl;
	  }
  }

  if(pipeline)
  {
    dev = freenect2.openDevice(serial, pipeline);

  }
  else
  {
    dev = freenect2.openDevice(serial);
  }

  if (pipeline1)
  {
	  dev1 = freenect2.openDevice(serial1, pipeline1);

  }
  else
  {
	  dev1 = freenect2.openDevice(serial1);
  }

  if(dev == 0)
  {
    std::cout << "failure opening device!" << std::endl;
    return -1;
  }

  if (dev1 == 0)
  {
	  std::cout << "failure opening device1!" << std::endl;
	  return -1;
  }

  signal(SIGINT,sigint_handler);
  protonect_shutdown = false;

  libfreenect2::SyncMultiFrameListener listener(libfreenect2::Frame::Color | libfreenect2::Frame::Ir | libfreenect2::Frame::Depth);
  libfreenect2::FrameMap frames;
  libfreenect2::Frame undistorted(512, 424, 4), registered(512, 424, 4);

  libfreenect2::SyncMultiFrameListener listener1(libfreenect2::Frame::Color | libfreenect2::Frame::Ir | libfreenect2::Frame::Depth);
  libfreenect2::FrameMap frames1;
  libfreenect2::Frame undistorted1(512, 424, 4), registered1(512, 424, 4);

  dev->setColorFrameListener(&listener);
  dev->setIrAndDepthFrameListener(&listener);
  dev->start();

  dev1->setColorFrameListener(&listener1);
  dev1->setIrAndDepthFrameListener(&listener1);
  dev1->start();

  std::cout << "device serial: " << dev->getSerialNumber() << std::endl;
  std::cout << "device firmware: " << dev->getFirmwareVersion() << std::endl;

  std::cout << "device serial: " << dev1->getSerialNumber() << std::endl;
  std::cout << "device firmware: " << dev1->getFirmwareVersion() << std::endl;

  libfreenect2::Registration* registration = new libfreenect2::Registration(dev->getIrCameraParams(), dev->getColorCameraParams());
  libfreenect2::Registration* registration1 = new libfreenect2::Registration(dev1->getIrCameraParams(), dev1->getColorCameraParams());

#ifdef LIBFREENECT2_WITH_OPENGL_SUPPORT
  Viewer viewer;
  viewer.initialize();
#endif

  while(!protonect_shutdown)
  {
    listener.waitForNewFrame(frames);
    libfreenect2::Frame *rgb = frames[libfreenect2::Frame::Color];
    libfreenect2::Frame *ir = frames[libfreenect2::Frame::Ir];
    libfreenect2::Frame *depth = frames[libfreenect2::Frame::Depth];

	listener.waitForNewFrame(frames1);
	libfreenect2::Frame *rgb1 = frames1[libfreenect2::Frame::Color];
	libfreenect2::Frame *ir1 = frames1[libfreenect2::Frame::Ir];
	libfreenect2::Frame *depth1 = frames1[libfreenect2::Frame::Depth];

        registration->apply(rgb, depth, &undistorted, &registered);
	registration1->apply(rgb1, depth1, &undistorted1, &registered1);

#ifdef LIBFREENECT2_WITH_OPENGL_SUPPORT
    viewer.addFrame("RGB", rgb);
    viewer.addFrame("ir", ir);
    viewer.addFrame("depth", depth);
    viewer.addFrame("registered", &registered);

    protonect_shutdown = viewer.render();
#else
    protonect_shutdown = true;
#endif

    listener.release(frames);

	listener1.release(frames);

    //libfreenect2::this_thread::sleep_for(libfreenect2::chrono::milliseconds(100));
  }

  // TODO: restarting ir stream doesn't work!
  // TODO: bad things will happen, if frame listeners are freed before dev->stop() :(
  dev->stop();
  dev->close();

  dev1->stop();
  dev1->close();

  delete registration;

  return 0;
}

Thank you for reading

yuanmingze, 2015-09-13:
Create libfreenect2::Freenect2 object for each libfreenect2::Freenect2Device. 
*dev and  *dev1 created by different  libfreenect2::Freenect2 object.

```javascript
libfreenect2::Freenect2 freenect2_0;
libfreenect2::Freenect2Device *dev0 = 0;
dev0 = freenect2_00.openDevice(serial0, pipeline0);

libfreenect2::Freenect2 freenect2_1;
libfreenect2::Freenect2Device *dev1 = 0;
dev1 = freenect2_01.openDevice(serial1, pipeline1);
```

Multi kinect will work .But I don't sure it's best way.

moonsh, 2015-09-17:
I am so appreciate your help.
Now second kinect is started but I faced another error message like below..

[DepthPacketStreamParser::onDataReceived] not all subsequences received 1022
[RgbPacketStreamParser::onDataReceived] packetsize or sequence doesn't match!
[RgbPacketStreamParser::onDataReceived] packetsize or sequence doesn't match!
[DepthPacketStreamParser::onDataReceived] not all subsequences received 512
[RgbPacketStreamParser::onDataReceived] packetsize or sequence doesn't match!
[DepthPacketStreamParser::onDataReceived] not all subsequences received 511
[RgbPacketStreamParser::onDataReceived] packetsize or sequence doesn't match!
[DepthPacketStreamParser::onDataReceived] not all subsequences received 960

It seems packetsize is too big to send .

yuanmingze, 2015-09-17:
Maybe the PC have not enough computing capability to get the depth. You can try to use libfreenect2::OpenCLPacketPipeline() or chage a powerful computer.
Do you use USB3.0 pcie card for each kinect ? 2 Kinects use one USB3.0 card will limited by USB3.0 hub speed bound.

moonsh, 2015-09-18:
Ah!  I used kinects at one USB3.0 card. My computer has two usb 3.0 slots so I thought it is possible to control kinects together.  I am thinking buy USB 3.0 card which has two slot. In this case, is it available to control two kinects ? or Only one USB3.0 card can control one kinect ? Here is usb 3.0 card I would buy . (SEDNA - PCI Express 2 Port USB 3.0 + 1 Port PeSATA Adapter with Low Profile Bracket - (NEC / Renesas uPD720202 chipset))

xlz, 2015-09-18:
`[RgbPacketStreamParser::onDataReceived] packetsize or sequence doesn't match!` this message is not from latest code.

moonsh, 2015-09-23:
I downloaded latest code and I got same result ;;;;
[Error]  [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Debug] [DepthPacketStreamParser] not all subsequences received 975..
...

floe, 2015-11-09:
Is this fixed with 0.1-rc1? If not, please re-open.

closed 2015-11-09 by floe

moonsh, 2016-02-13:
It works well thank you.