OpenKinect/libfreenect2 #375: [OpenGLDepthPacketProcessor] Failed to create opengl window

cpaniagua1, 2015-08-31:
Hello,
I installed libfreenect2 with its dependencies and I had any problem. However, when I run the example ./Proctonect gl to test the OpenGL support I get the follow error:

[OpenGLDepthPacketProcessor] Failed to create opengl window

I am using a Intel Nuc with Ubuntu 14.04 and Ros Indigo. Graphics are Gallium 0.4 on llvmpipe (LLVM 3.5, 256 bits).

Can somebody help me with this?

xlz, 2015-08-31:
The combination of OpenGL/Intel/Linux does not work right now. This will soon be fixed.

Anyway, your glxinfo should show Mesa 10.x instead of Gallium 0.4 on llvmpipe.

cpaniagua1, 2015-09-01:
Thanks for your answer.

My glxinfo shows:
OpenGL vendor string: VMware, Inc.
OpenGL renderer string: Gallium 0.4 on llvmpipe (LLVM 3.5, 256 bits)
OpenGL version string: 1.4 (3.0 Mesa 10.3.2)
OpenGL extensions: 

When you say Intel, are you talking about the processor or the graphics?

xlz, 2015-09-01:
Intel graphics is integrated. They are not differentiated.

I think you are using VNC? Or you did not install xserver-xorg-video-intel which is the proper driver for Intel graphics.

cpaniagua1, 2015-09-02:
I'm not using VNC. The driver for Intel graphics which is installed in my Nuc is xserver-xorg-video-intel-lts-utopic. Is there any problem with it?

floe, 2015-09-02:
No, the driver package is the correct one, but `OpenGL vendor string: VMware, Inc.` indicates that you also have a VMWare driver installed which somehow gets used instead. Or are you running this within a VM?

cpaniagua1, 2015-09-02:
No, I'm not running within a VM and I don't know how to uninstalled the VMWare driver. I tried sudo apt-get remove vm* but it didn't find anything to remove.

floe, 2015-09-02:
Try `sudo apt-get remove xserver-xorg-video-vmware-*` instead. No guarantees, though.

cpaniagua1, 2015-09-02:
Thank you. That worked. I removed the xserver-xorg-video-vmware-* but my glxinfo continues showing :

OpenGL vendor string: VMware, Inc.
OpenGL renderer string: Gallium 0.4 on llvmpipe (LLVM 3.5, 256 bits)
OpenGL version string: 3.0 Mesa 10.3.2
OpenGL shading language version string: 1.30
OpenGL context flags: (none)
OpenGL extensions:

And when I run the unity support test I get:

OpenGL vendor string:   VMware, Inc.
OpenGL renderer string: Gallium 0.4 on llvmpipe (LLVM 3.5, 256 bits)
OpenGL version string:  3.0 Mesa 10.3.2

Not software rendered:    no
Not blacklisted:          yes
GLX fbconfig:             yes
GLX texture from pixmap:  yes
GL npot or rect textures: yes
GL vertex program:        yes
GL fragment program:      yes
GL vertex buffer object:  yes
GL framebuffer object:    yes
GL version is 1.4+:       yes

Unity 3D supported:       no

xlz, 2015-09-02:
`grep intel /var/log/Xorg.0.log`?

xlz, 2015-09-02:
Also, have you installed libgl1-mesa-dri(-*)?

cpaniagua1, 2015-09-03:
Yes, I have installed libgl1-mesa-dri-lts-utopic.

grep intel  /var/log/Xorg.0.log:

[     2.350] (==) Matched intel as autoconfigured driver 0
[     2.350] (II) LoadModule: "intel"
[     2.350] (II) Loading /usr/lib/xorg/modules/drivers/intel_drv.so
[     2.351] (II) Module intel: vendor="X.Org Foundation"
[     2.351] (II) intel: Driver for Intel(R) Integrated Graphics Chipsets:
[     2.351] (II) intel: Driver for Intel(R) HD Graphics: 2000-6000
[     2.351] (II) intel: Driver for Intel(R) Iris(TM) Graphics: 5100, 6100
[     2.351] (II) intel: Driver for Intel(R) Iris(TM) Pro Graphics: 5200, 6200, P6300

xlz, 2015-09-04:
Is nouveau interfering? Well at this stage, I'm of no better help than Google.

xlz, 2015-09-04:
Here is another thing you can try:

```
LIBGL_DEBUG=verbose glxinfo
```
And see which shared library is loaded.

Also, you can report your graphics model `lspci | grep VGA` and `grep model.name /proc/cpuinfo`.

floe, 2015-11-09:
Is this fixed with 0.1-rc1? If not, please re-open.

closed 2015-11-09 by floe