OpenKinect/libfreenect2 #377: /lib/libfreenect2.so: undefined reference in Odroid XU3

oskr1709, 2015-09-02:
Hello!

I have an Odroid XU3 and I want to use Kinect v2 with it. I´m following the 4 steps of this entry; http://forum.odroid.com/viewtopic.php?f=93&t=12207 but at the end of the "make" step I´m having the next errors:

`
../lib/libfreenect2.so: undefined reference to 'XRRSetCrtcGamma'`
`
../lib/libfreenect2.so: undefined reference to 'XRRGetOutputInfo'`
`
../lib/libfreenect2.so: undefined reference to 'XcursorImageLoadCursor'`
`
../lib/libfreenect2.so: undefined reference to 'XRRGetScreenResourcesCurrent'`
`
../lib/libfreenect2.so: undefined reference to 'XRRAllocGamma'`
`
../lib/libfreenect2.so: undefined reference to 'XRRQueryVersion'`
`
../lib/libfreenect2.so: undefined reference to 'XRRFreeOutputInfo'`
`
../lib/libfreenect2.so: undefined reference to 'XRRFreeScreenResources'`
`
../lib/libfreenect2.so: undefined reference to 'XRRGetOutputPrimary'`
`
../lib/libfreenect2.so: undefined reference to 'XRRFreeGamma'`
`
../lib/libfreenect2.so: undefined reference to 'XineramaQueryScreens'`
`
../lib/libfreenect2.so: undefined reference to 'XRRQueryExtension'`
`
../lib/libfreenect2.so: undefined reference to 'XineramaIsActive'`
`
../lib/libfreenect2.so: undefined reference to 'XineramaQueryExtension'`
`
../lib/libfreenect2.so: undefined reference to 'XRRGetScreenResources'`
`
../lib/libfreenect2.so: undefined reference to 'XRRGetCrtcGamma'`
`
../lib/libfreenect2.so: undefined reference to 'XISelectEvents'`
`
../lib/libfreenect2.so: undefined reference to 'XRRFreeCrtcInfo'`
`
../lib/libfreenect2.so: undefined reference to 'XRRUpdateConfiguration'`
`
../lib/libfreenect2.so: undefined reference to 'XRRGetCrtcInfo'`
`
../lib/libfreenect2.so: undefined reference to 'XcursorImageCreate'`
`
../lib/libfreenect2.so: undefined reference to 'XIQueryVersion'`
`
../lib/libfreenect2.so: undefined reference to 'XRRSetCrtcConfig'`
`
../lib/libfreenect2.so: undefined reference to 'XcursorImageDestroy'`
`
../lib/libfreenect2.so: undefined reference to 'XRRSelectInput'`
`
../lib/libfreenect2.so: undefined reference to 'XRRGetCrtcGammaSize'`

Does anyone know what is the problem?

floe, 2015-09-02:
Well, it looks like your installation is missing some Xorg libraries, e.g. `libXrandr`, `libXi`, etc. Try installing `libxrandr-dev`, `libxi-dev` and so on.

oskr1709, 2015-09-03:
Hi floe, thanks for your reply. I tried with `sudo apt-get install xorg openbox` and with some packages of this link: http://packages.ubuntu.com/trusty/x11/xorg-dev (It seems I already had them) and the same problem

floe, 2015-09-03:
Can you post the full build output, e.g. on pastebin? And please use `make VERBOSE=1`.

oskr1709, 2015-09-04:
Floe, this is the build output: http://paste.ubuntu.com/12268944/

floe, 2015-09-04:
I'm noticing two weird things:

```
make[2]: Warning: File `../lib/libfreenect2.so' has modification time 8.4e+07 s in the future
make[2]: Nothing to be done for `CMakeFiles/freenect2shared.dir/build'.
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
```
Looks like something with your clock is wrong, maybe you're trying to link against an older library version by accident?

And the build command that's actually failing is explicitly using `/usr/local/lib` as a library source directory. Maybe something weird is installed there, messing up the build?

oskr1709, 2015-09-05:
I rebuild it changing with the correct date and time and deleting some old .so files of libglfw in the /usr/local/lib (that I installed before installing libglfw3). Now I don´t have the warnings that you mentioned me but still is the same problem at the end: http://paste.ubuntu.com/12286073/

floe, 2015-09-06:
Hm, you could try editing `CMakeLists.txt` and explicitly adding `xrandr` to the `SET(LIBRARIES ...` command. I can't really think of anything else...

xlz, 2015-09-07:
Your cmake configured `glfw3`. This is not expected. `glfw` is expected. That is, you should have `-lglfw` instead of `-lglfw3` during linking.

The config comes from `cat /usr/lib/*/pkgconfig/glfw3.pc`, which should give you `Libs: -L${libdir} -lglfw`. In the binary package there is only `libglfw.so.3.x`, not `libglfw3.so.x`. So you must have the wrong libglfw3 package installed.

OK, so there is actually no libglfw3(-dev) packages for Ubuntu armhf. Can't say we are really supporting ARM platform except Jetson TK1. Maybe you can drive this if you are interested.

I took a look, it seems you should be downloading libglfw3_*_armhf.deb from http://ports.ubuntu.com/ubuntu-ports/pool/universe/g/glfw3/ instead of http://archive.ubuntu.com/ubuntu/...

oskr1709, 2015-09-07:
Hi Lingzhu,
i installed libglfw3_3.1.1-1_armhf.deb as you told me but I have this linking error: 

[ 88%] Built target freenect2
Scanning dependencies of target freenect2shared
Linking CXX shared library ../lib/libfreenect2.so
/usr/bin/ld: cannot find -lglfw3
collect2: error: ld returned 1 exit status
make[2]: *** [../lib/libfreenect2.so] Error 1
make[1]: *** [CMakeFiles/freenect2shared.dir/all] Error 2
make: *** [all] Error 2

How can I change the linking to  -lglfw instead of -libglw3 as you told me?

With cat /usr/lib/*/pkgconfig/glfw3.pc I have Libs: -L${libdir} -lglfw

xlz, 2015-09-07:
You need to find out why `-lglfw3` is generated. I don't know your systems specifics. Basically it's investigating the configuration of GLFW3 in CMakeLists.txt. 

In CMakeLists.txt, `GLFW3_LIBRARIES` should have `glfw` not `glfw3`. Further in, in FindGLFW3.cmake, `PKG_CONFIG_FOUND` should be true, and running `pkg-config --libs glfw3` should return `-lglfw` not `-lglfw3`. Otherwise I can't help you.

oskr1709, 2015-09-10:
I came back a little bit because I think I know why the building is messed up.  I downloaded libglfw3_3.0.4-1_armf.deb    and installed it ( sudo dpkg -i libglfw3_3.0.4-1_armf.deb  )

When I´m building libfreenect2 I have the next error: fatal error: GLFW/glfw3.h: No such file or directory #include <GLFW/glfw3.h> . I was fixing it using this link: http://stackoverflow.com/questions/17768008/how-to-build-install-glfw-3-and-use-it-in-a-linux-project  but I think that because of this I´m getting the undefined references and the other errors. So, is there any other way to fix this problem?

xlz, 2015-09-10:
Did you install libglfw3-dev too?

oskr1709, 2015-09-10:
I´m not sure when I installed it, but It seems I have it

xlz, 2015-09-10:
You had wrong version installed. So you'd better find all glfw installations, uninstall them, and install libglfw3 and libglfw3-dev from http://ports.ubuntu.com/ubuntu-ports/pool/universe/g/glfw3/.

`GLFW/glfw3.h: No such file or directory` is obviously a result of some wrong version libglfw3-dev **still not being uninstalled**.

oskr1709, 2015-09-11:
Thanks! That solved the problem! But now I have a new one when I want to run the Protonect program: [Error] [Freenect2Impl] Your libusb does not support large iso buffer! No device connected!

xlz, 2015-09-11:
Do you have the correct libusb linked? Please post output of `ldd ./Protonect | grep libusb`.

oskr1709, 2015-09-11:
This is the output: libusb-1.0.so.0 => /lib/arm-linux-gnueabihf/libusb-1.0.so.0 (0xb6b7e000)

xlz, 2015-09-11:
This is not the correct libusb.

Have you run `install_libusb.sh`? You need the locally built libusb with proper USB3 support. ldd output should point to something like `/home/oskr1709/src/libfreenect2/depends/libusb/lib/libusb-1.0.so.0`.

oskr1709, 2015-09-11:
I run install_libsusb.sh. Now It´s recognizing the kinect v2! I´m searching a solution for this now:
 [Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65543 The requested client API version is unavailable 
[Error] [OpenGLDepthPacketProcessor] Failed to create opengl window.

xlz, 2015-09-11:
> GLFW_VERSION_UNAVAILABLE

> The requested OpenGL or OpenGL ES version (including any requested context or framebuffer hints) is not available on this machine.

Please post output of `glxinfo | grep OpenGL`. You need OpenGL 3.1+.

oskr1709, 2015-09-11:
This is the output: 

OpenGL vendor string: Mesa Project
OpenGL renderer string: Software Rasterizer
OpenGL version string: 2.1 Mesa 10.1.0
OpenGL shading language version string: 1.20
OpenGL extensions:

xlz, 2015-09-11:
ODROID-XU3 includes `Mali-T628 MP6(OpenGL ES 3.0/2.0/1.1 and OpenCL 1.1 Full profile)`.

I don't think OpenGL ES driver is supported.

floe, 2015-09-12:
@xlz good point... definitely would need some extra work to support GLES. Since we don't use many legacy OpenGL features anymore AFAICT, it shouldn't be too difficult. However, this is certainly something outside the scope of 0.1.

floe, 2015-09-12:
@oskr1709 in fact, you could simply try running `Protonect -noviewer cl`. This will just use OpenCL to process the depth data, and won't show any output, but you can at least check the basic functionality.

oskr1709, 2015-09-14:
Thanks floe. Without displaying at the beginning looks fine but then I´m having a lot of warnings like: skippping depth packet and skipping rgb packet!. The complete output is this: http://paste.ubuntu.com/12409274/ . Is that normal? And another question: I guess  there is a way to get the data using Openni, pcl or opencv to process it without using opengl, isn´t it?

xlz, 2015-09-14:
According to the log, you are not using the latest master. "not all subsequences received" should be a debug level message in the latest master.

These messages are not important. I'm surprised there is actually OpenCL driver for your Mali-T628. You should look for performance output from OpenCL and see the processing rate.

If you have OpenCL support then OpenGL is not required.

oskr1709, 2015-09-16:
Ok. Thank you xlz! I´ll try to do an application with pcl, hopping that I won´t have problems with OpenCL, instead of OpenGL. Thanks to both!

xlz, 2015-10-15:
@xlz create a PR for `install_ubuntu.sh` to use ports.ubuntu.com for armhf

floe, 2015-10-23:
Will be closed by #429 AFAICT.

floe, 2015-11-06:
Should be fixed by #429, please re-test with latest master and re-open if necessary.

closed 2015-11-06 by floe

CrazyPopLin, 2015-11-20:
hi @oskr1709 , are you using libfreenect2 with OpenCL successfully ?

oskr1709, 2015-11-25:
Yes CrazyPopLin. Just I´m not sure how to check the performance output from OpenCL and see the processing rate, as xlz commented.