OpenKinect/libfreenect2 #378: switch to modern C++

JKleinHub, 2015-09-03:
Is there a good reason, that old compilers must still be supported? Making C++11 (or later) mandatory would allow to clean the code up significantly. For example, the library is in general not good documented and at many points you have to exchange pointers with the interface, and it is not clear, who is responsible for freeing them. Consistently using unique_ptr would solve this problem even without adding any documentation and would enforce correct usage. And this is really just one example of what could be enhanced, if support for older compilers was dropped.

I just started using the library and consider committing some patches, but I really would want to use modern C++ for this.

fran6co, 2015-09-03:
+1 for this

HenningJ, 2015-09-03:
I like C++11, too. But I have to work with an old compiler (Visual Studio 2008), so I appreciate the fact that not too much stuff from C++11 is being used. So there is a case for not using C++11. :-)

But I still think you're right, especially the part about enforcing the memory management. It would be worth switching to C++11 just for (I ran into that exact problem before).

fran6co, 2015-09-03:
Can't you have both? I mean, libfreenect2 should offer a C interface to make it available to as many bindings, wrappers, compilers available but internally stick with one standard.

JKleinHub, 2015-09-03:
This would require somebody actually writing the C-Interface (and then it would be still less comfortable than an 'old' C++ interface).
Supporting two branches at the same time is most likely too time consuming, as such a change would affect the interface in really a lot places; so it is basically all or nothing.

fran6co, 2015-09-03:
it's not 2 branches, the C interface is just a wrapper and is already there in https://github.com/OpenKinect/libfreenect2/blob/master/include/libfreenect2.h it just needs to be implemented.

JKleinHub, 2015-09-03:
Oh, i wasn't clear about the branches, a edited it accordingly.
Anyway, it is probably the decision of the person in charge, as I think we would all agree on "it would be a good thing, but some people might not be able to use it anymore".

xlz, 2015-09-03:
Yes, libfreenect2 shall support old compilers. In fact, it painstakingly provides workarounds to support old OpenGL libraries and OpenCL loaders despite they have major bugs. It is necessary to provide as much support as possible because there are indeed that many users with heterogeneous environments, and most of the users will not have enough time to investigate why libfreenect2 does not work for them. (Hundreds of issues are being reported despite being clearly documented in README. Better not let them happen in the first place.)

C++11 is not backward ABI compatible. Requiring C++11 features would probably prevent the library from working with non-C++11 libraries. The main target distro Ubuntu 14.04 ships all non-C++11 libraries. It needs to be supported.

I am all for clean code. But cleanup will break working functionalities, and there are few people and little testing effort to support that level of maintenance. Each change needs to be tested under Linux/Windows/MacOSX by various test plans. If it does not provide any user visible improvement and breaks working functionalities, it's probably not worth it. If you have a cleanup PR that is tested against all of Linux, Windows, and MacOSX, it should be all right.

closed 2015-09-07 by JKleinHub