OpenKinect/libfreenect2 #381: [Question] Device Recovery/Reconnect

bennilson, 2015-09-08:
I'm wondering if there is a way to reconnect to the kinect after it has been unplugged and replugged in with out the need to restart the code. Or has somthing at the OS level changed so we can't do that because the kinect is usb.

xlz, 2015-09-09:
Hotplug is supported in libusb http://libusb.sourceforge.net/api-1.0/hotplug.html on some not all platforms.

But it is not implemented yet.

xlz, 2016-02-04:
Is unplugging and replugging something that is required in your environment? Can you explain why you have to do it?

bennilson, 2016-02-05:
It's a maintenance task. If the camera or cable needs to be replaced because it gets damaged, it's simpler to just plug in the Kinect and the driver auto recover then having to connect to the pc that's harder to reach and restart the driver. Also some IP based device drivers have some kind of connection retry protocol which I have found useful in this scenario.

xlz, 2016-02-05:
It is easy to implement waiting for a device to open. The restarting part is hard to implement with the full hotplugging maintenance in the library.

The user holds a pointer to a Freenect2Device object. Wen restarting, there are some policy problems which are not easy to decide on our side: does the user want to claim the same device with the same serial number, or any new device in the same port, or just any new device? Handling device policy seems something under udev's purview. Also, since the user holds the reference to the object, the library can't delete it and must replace its internal states rather than create a new object. This requires a lot of refactoring in the core usb code.

Right now I can provide a new interface to allow you to wait on a device, and notify you when the device is lost. But the restarting mechanism probably has to be done outside of the library.