OpenKinect/libfreenect2 #384: OpenCL depth processing is not available!

zfxw0206, 2015-09-11:
I have one issue:
Now I could run kinect2_bridge.launch, it works well, but it uses opengl as default depth method. I want to use opencl, so I changed the launch file to "opencl". After that, I got the error"OpenCL depth processing is not available!". Shown as following:

~$ roslaunch kinect2_bridge kinect2_bridge.launch 
... logging to /home/vision-gm/.ros/log/94d658fe-58a3-11e5-8c44-54ee7568ff38/roslaunch-visiongm-ThinkPad-W541-24737.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://visiongm-ThinkPad-W541:44469/

SUMMARY
========

PARAMETERS
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/vision-gm/w...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: opencl
 * /kinect2_bridge/edge_aware_filter: True
 * /kinect2_bridge/fps_limit: -1.0
 * /kinect2_bridge/jpeg_quality: 90
 * /kinect2_bridge/max_depth: 12.0
 * /kinect2_bridge/min_depth: 0.1
 * /kinect2_bridge/png_level: 1
 * /kinect2_bridge/publish_tf: True
 * /kinect2_bridge/queue_size: 5
 * /kinect2_bridge/reg_devive: -1
 * /kinect2_bridge/reg_method: default
 * /kinect2_bridge/sensor: 
 * /kinect2_bridge/use_png: False
 * /kinect2_bridge/worker_threads: 4
 * /kinect2_points_xyzrgb_hd/queue_size: 5
 * /kinect2_points_xyzrgb_qhd/queue_size: 5
 * /kinect2_points_xyzrgb_sd/queue_size: 5
 * /rosdistro: indigo
 * /rosversion: 1.11.13

NODES
  /
    kinect2 (nodelet/nodelet)
    kinect2_bridge (nodelet/nodelet)
    kinect2_points_xyzrgb_hd (nodelet/nodelet)
    kinect2_points_xyzrgb_qhd (nodelet/nodelet)
    kinect2_points_xyzrgb_sd (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[kinect2-1]: started with pid [24755]
process[kinect2_bridge-2]: started with pid [24756]
[ INFO] [1441989538.866589445]: Initializing nodelet with 8 worker threads.
[ INFO] [1441989538.874139467]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
process[kinect2_points_xyzrgb_sd-3]: started with pid [24795]
process[kinect2_points_xyzrgb_qhd-4]: started with pid [24809]
process[kinect2_points_xyzrgb_hd-5]: started with pid [24827]
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/vision-gm/wspace/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: default
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 5
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: true
     base_name_tf: kinect2
   worker_threads: 4

OpenCL depth processing is not available!
Initialization failed!

Does anybody know how to solve this problem?

zfxw0206, 2015-09-11:
I checked the "kinect2_bridge.cpp", it says:

#ifdef LIBFREENECT2_WITH_OPENCL_SUPPORT
    depthDefault = "opencl";
#endif

So, how to define LIBFREENECT2_WITH_OPENCL_SUPPORT?

floe, 2015-09-11:
This is set at configure/compile time by `cmake`. You need a working OpenCL development environment first (i.e. if you're on Debian or Ubuntu, check http://wiki.tiker.net/OpenCLHowTo).

zfxw0206, 2015-09-11:
Yes, I use Ubuntu and my nvidia driver version is 346.82. I already installed "nvidia-libopencl1-346-updates", "nvidia-opencl-icd-346-updates","beignet" and "opencl-header". 

 Now "intel.icd" and "nvidia.icd" are in "/etc/OpenCL/Vendors" . But there is no "libOpenCL.so" in "/usr/lib/x86_64-linux-gnu".

Besides this, shall I change the cmake file in libfreenect/example/protonect?

xlz, 2015-09-12:
libOpenCL.so is in ocl-icd-libopencl1.

zfxw0206, 2015-09-14:
After installing ocl-icd-libopencl1, libOpenCL.so comes out. Thank you!
But the error "OpenCL depth processing is not available!" is still there

zfxw0206, 2015-09-14:
I tried " sudo ./Protonect cl", it works. It seems that it is permission issue.

wiedemeyer, 2015-09-14:
For other permission problems with nvidia and OpenCL it helps to install `sudo apt-get install nvidia-modprobe`. Maybe it helps for your problem as well.

zfxw0206, 2015-09-14:
I already install it. It does not work

xlz, 2015-09-14:
> he error "OpenCL depth processing is not available!" is still there.

Have you rebuilt libfreenect2, or have you fixed/verified the CMake configuration of OpenCL?

> crw-rw----+ 1 root video 226, 0 Sep 14 08:52 card0

But are you in `video` group?

zfxw0206, 2015-09-14:
What do you mean of " in video group"?  How could I check it?

xlz, 2015-09-14:
`id` will show your groups. There are quite a few discussion about "dri2 connection failed" if you Google it. If it works with sudo and does not without, it's permission issues.

zfxw0206, 2015-09-14:
Yes, that is what I thought. But I do not know how to solve it.

zfxw0206, 2015-09-14:
I find that installing "beignet" will lead to "x server found" problem. After I remove it, it works fine.
But when I change the depth method tp "opencl", the "OpenCL depth processing is not available!" is still there. How should I change the CMake file to solve it?

xlz, 2015-09-14:
Well, just `adduser zfxw0206 video` to add yourself to video group.

Not available just literally means it's not made available. CMake output shows whether your OpenCL is configured. How is it?

zfxw0206, 2015-09-15:
Now I still could not use opencl as depth method. I find that when I run "./Protonect cl", it says " Opencl pipline is not supported". How to fix this problem?

xlz, 2015-09-15:
I asked for CMake output. Without it I'm unable to provide further help.

zfxw0206, 2015-09-15:
~/libfreenect2/examples/protonect/build$ cmake ..
-- using tinythread as threading library
Linking with these libraries: usb-1.0;/usr/lib/x86_64-linux-gnu/libturbojpeg.so.0;pthread;glfw;/usr/lib/x86_64-linux-gnu/libGL.so;/usr/lib/x86_64-linux-gnu/libOpenCL.so
-- Configuring done
-- Generating done
-- Build files have been written to: /home/vision-gm/libfreenect2/examples/protonect/build

Is this what you want?

xlz, 2015-09-15:
```
~/libfreenect2/examples/protonect/build$ rm -r *
~/libfreenect2/examples/protonect/build$ cmake ..
```
You should see this line:
```
-- Found OpenCL: /usr/lib/x86_64-linux-gnu/libOpenCL.so  
...
~/libfreenect2/examples/protonect/build$ grep OPENCL ../include/libfreenect2/config.h
```
You should see this:
```
#define LIBFREENECT2_WITH_OPENCL_SUPPORT
~/libfreenect2/examples/protonect/build$ make
...
~/libfreenect2/examples/protonect/build$ touch ../Protonect.cpp
~/libfreenect2/examples/protonect/build$ make VERBOSE=1
...
[ 92%] Building CXX object CMakeFiles/Protonect.dir/Protonect.cpp.o
/usr/bin/c++   -DRESOURCES_INC -fno-exceptions -O2 -g -DNDEBUG -I/home/xlz/src/libfreenect2/examples/protonect/include -I/home/xlz/src/libfreenect2/examples/protonect/src/tinythread -I/home/xlz/src/libfreenect2/depends/libusb/include/libusb-1.0 -I/usr/include/libdrm    -o CMakeFiles/Protonect.dir/Protonect.cpp.o -c /home/xlz/src/libfreenect2/examples/protonect/Protonect.cpp
Linking CXX executable ../bin/Protonect
...
```
Change the arguments from -c to -E and grep:
```
~/libfreenect2/examples/protonect/build$ /usr/bin/c++   -DRESOURCES_INC -fno-exceptions -O2 -g -DNDEBUG -I/home/xlz/src/libfreenect2/examples/protonect/include -I/home/xlz/src/libfreenect2/examples/protonect/src/tinythread -I/home/xlz/src/libfreenect2/depends/libusb/include/libusb-1.0 -I/usr/include/libdrm -E /home/xlz/src/libfreenect2/examples/protonect/Protonect.cpp | grep OpenCL
class OpenCLDepthPacketProcessorImpl;
class OpenCLDepthPacketProcessor : public DepthPacketProcessor
  OpenCLDepthPacketProcessor(const int deviceId = -1);
  virtual ~OpenCLDepthPacketProcessor();
  OpenCLDepthPacketProcessorImpl *impl_;
class OpenCLPacketPipeline : public BasePacketPipeline
  OpenCLPacketPipeline(const int deviceId = -1);
  virtual ~OpenCLPacketPipeline();
        pipeline = new libfreenect2::OpenCLPacketPipeline();
```

If `config.h` is in the include path of Protonect.cpp, and the macro `LIBFREENECT2_WITH_OPENCL_SUPPORT` is defined  it is impossible for Protonect to output "OpenCL pipeline is not supported!" So, which step went wrong for you?

zfxw0206, 2015-09-15:
The last step: Change the arguments from -c to -E and grep:
after input the command, I got :
c++: error: /home/xlz/src/libfreenect2/examples/protonect/Protonect.cpp: No such file or directory
c++: fatal error: no input files
compilation terminated.

xlz, 2015-09-15:
`/home/$your_own_username/libfreenect2`

zfxw0206, 2015-09-15:
Yes, every thing goes well now. Thank you very much. Then  I got the error:
[OpenCLDepthPacketProcessor] could not initialize OpenCLDepthPacketProcessor.

I solve this problem by installing "nvidia-modprobe"
But this cause a new problem, I could not start my laptop after restart.

xlz, 2015-09-15:
I think you did not clean up cmake and reconfigure, that is why you had this error persisting.

Installing graphics driver is out of the scope of my help.

zfxw0206, 2015-09-15:
How to clean up cmake and reconfigure? Is it "rm -r *" and "cmake .." again?

xlz, 2015-09-15:
Yes.

zfxw0206, 2015-09-15:
Thank you very much for your help. 
I reinstall the ubuntu and download the libfreenect2 again. The problem is still there. Even though I run "rm -r * ", it does not solve the problem. So I think it may be the nvidia driver problem.

wkentaro, 2015-10-18:
maybe I have same problem with:
* Ubuntu 14.04
* intel gpu

I followed the instruction in README, and installed:
* beignet-dev

but there are no line about opencl at `cmake ..` in `libfreenect/build`

```
% cmake ..
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- using tinythread as threading library
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26")
-- checking for module 'libusb-1.0'
--   found libusb-1.0, version 1.0.19
-- Performing Test TURBOJPEG_WORKS
-- Performing Test TURBOJPEG_WORKS - Success
-- Found TurboJPEG: /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0
-- checking for module 'glfw3'
--   found glfw3, version 3.0.4
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/x86_64-linux-gnu/libX11.so
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so
-- Could NOT find OpenCL (missing:  OPENCL_INCLUDE_DIRS)
-- RPATH set to /home/baxter/libfreenect2/depends/libusb/lib
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Linking with these libraries: usb-1.0;/usr/lib/x86_64-linux-gnu/libturbojpeg.so.0;pthread;glfw;/usr/lib/x86_64-linux-gnu/libGL.so
-- Configurating examples
-- Using in-tree freenect2 target
-- Configuring done
-- Generating done
-- Build files have been written to: /home/baxter/libfreenect2/build
```

I have `/etc/OpenCL/vendors/intel.icd` and `/usr/lib/x86_64-linux-gnu/libOpenCL.so`.

wkentaro, 2015-10-18:
I have both intel and nvidia gpu.
so now I'm trying using nvida one.
```
...
-- Linking with these libraries: usb-1.0;/usr/lib/x86_64-linux-gnu/libturbojpeg.so.0;pthread;glfw;/usr/lib/x86_64-linux-gnu/libGL.so;/usr/lib/x86_64-linux-gnu/libOpenCL.so
...
```
and got lien about `libOpenCL.so` in the output of `cmake ..`.

But in the log of `./bin/Protonect cl`, 
```
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel(R) Core(TM) i7-3920XM CPU @ 2.90GHz (CPU)[GenuineIntel]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Intel(R) Core(TM) i7-3920XM CPU @ 2.90GHz (CPU)[GenuineIntel]
```
It sees Intel CPU not nvidia's gpu, do you have any idea to fix this?

wkentaro, 2015-10-18:
I had `intel.icd` in `/etc/OpenCL/vendors` and after moving that file, I have 

```
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Error] [OpenCLDepthPacketProcessor] could not initialize OpenCLDepthPacketProcessor
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] disabling usb transfer submission...
[Info] [Freenect2DeviceImpl] canceling usb transfers...
[Error] [OpenCLDepthPacketProcessor] could not initialize OpenCLDepthPacketProcessor
[Error] [OpenCLDepthPacketProcessor] could not initialize OpenCLDepthPacketProcessor
[Info] [Freenect2DeviceImpl] stopped
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing
```
when `./bin/Protonect cl` and no images and points come.

`./bin/Protonect` does work.

xlz, 2015-10-18:
It is not very clear what you did between these error messages. Please just post the commands you used.

First, `-- Could NOT find OpenCL (missing:  OPENCL_INCLUDE_DIRS)` says it did not find `CL/cl.h`, so where was `cl.h` and what did you do to make it find `cl.h`?

`now I'm trying using nvida one.` What did you do to use nvidia one?

Now it is still not clear you are using Intel or Nvidia OpenCL with the information you provided. If you use Nvidia OpenCL, you need to uninstall beignet-dev, and install opencl-headers and ocl-icd-opencl-dev. If you use Intel OpenCL, you need to uninstall nvidia-opencl-dev, nvidia-libopencl1 or other "nvidia opencl" packages, and install beignet-dev, and extra steps may also depend on your kernel version (did you set `/sys/module/i915/parameters/enable_cmd_parser`?, did you look at known issues at http://www.freedesktop.org/wiki/Software/Beignet/ ?).

If you are not using what you have been installing and uninstalling, `CL/cl.h` and `libOpenCL.so` has to be compatible. What is your `dpkg -S cl.h` and `dpkg -S libOpenCL.so` and `ls -l /usr/lib/x86_64-linux-gnu/libOpenCL.so`?

closed 2015-12-23 by floe