OpenKinect/libfreenect2 #386: OS X 10.10.4 gives error: 'Failed to create OpenGL window"

mostgood, 2015-09-12:
I'm running Protect on a 15" MacBook Pro Retina Mid-2012, running 10.10.4.

I've got everything installed but when I run ./bin/Protonect I get the following output:

[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @20:10 with serial 031113540747
[Info] [Freenect2Impl] found 1 devices
[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65543 NSGL: The targeted version of OS X does not support OpenGL 3.0 or 3.1
[Error] [OpenGLDepthPacketProcessor] Failed to create opengl window.

Can anyone help me with this? It appears that I have the wrong OpenGL version perhaps?

xlz, 2015-09-12:
Can you try to apply this change in `opengl_depth_packet_processor.cpp` and tell us what happens?

```diff
-  glfwWindowHint(GLFW_CONTEXT_VERSION_MINOR, 1);
+  glfwWindowHint(GLFW_CONTEXT_VERSION_MINOR, 3);
```

My guess is on Mac OS X, we must request OpenGL 3.3 core profile forward compatible. Not 3.1, not any profile.

robnsngh, 2015-09-16:
The above path is working smooth. Tested on OSX 10.10.3

xlz, 2015-09-16:
@robnsngh 
Did you mean it works with the above patch, and it does not work without the above patch?

robnsngh, 2015-09-16:
I had same issue and used the patch you mentioned in the above. Recompiled and the ./Protonect ran smoothly.

xlz, 2015-09-16:
This should be fixed after #393.

closed 2015-09-17 by floe

NikosEfthias, 2018-09-04:
this is still an issue

commit 70cea30e (2015-09-17): Merge pull request #393 from xlz/macosx-opengl

Fix GLFW setup on Mac OSX, closes #386