OpenKinect/libfreenect2 #387: Unity 3D/Unreal engine 4

DreaMmMer, 2015-09-14:
Hello,
is possible to use libfreenect2 with Unreal 4 or Unity 3D on Linux, please ?
How ? Are there any advanced solutions ? 

Thanks
M.

floe, 2015-09-14:
It kind of depends on what you actually want to do, but there are no ready-made solutions/plugins etc. at the moment, you'd have to write one first.

hanyazou, 2015-09-17:
Though I have no experience with Unity nor UE4, these combination might help you.

Kinect/PS with OpenNI 2
https://www.assetstore.unity3d.com/en/#!/content/10693

libfreenect2 OpenNI2 driver (I wrote this to use libfreenect2 with Point Cloud Library on Mac)
https://github.com/hanyazou/libfreenect2/tree/pull-request/drivers/OpenNI2-Freenect2Driver

closed 2016-03-07 by xlz