OpenKinect/libfreenect2 #388: MSCVR120D.dll is missing, then launch error, why is it requesting a debug library?

mikezucc, 2015-09-14:
Hi All,

I have been wrestling with the install instructions, which have thus far been smooth. I can't run Protonect.exe because i get this error: "The program can't start because MSCVR120D.dll is missing from your computer ( I have 2015 VS installed). Try reinstalling the program to fix this problem."

So then I tried to grab the dll from another development machine i have that is running a 64 bit 2013 VS. It fixed that error but then only gave me a launch error of "Unable to start correctly (0xc000007b)." I assume that is from mismatched architecture, but everything I did was according to the installing instructions and I assume everything is 64 bit.

What could I be doing wrong?

xlz, 2015-09-14:
Are you building a debug profile?

If not sure, also see if it is the problem of CMAKE_DEBUG_POSTFIX in CMakeLists.txt.

mikezucc, 2015-09-16:
Ok I removed the CMAKE block and rebuilt the thing and the build was fine. However I am still getting the error. Can't seem to understand what is particularly wrong. I tried to expand the scope of environment variables to include every possible file it would need. Do I need to adjust permissions on the file? Or could something be wrong with my Kinect configuration? I may not be using the driver installation correctly. Is that an issue that you see a lot?

xlz, 2015-09-16:
MSCVR120D.dll is probably something from your runtime environment. What about building a Release profile?

I haven't heard of any similar report of missing dlls. Also, I don't have Windows to test right now.

mikezucc, 2015-09-16:
I fixed the debug library issue, which is no longer needed since I am not building a debug profile. I will look into the source of this issue some more. For now I really can't discern any more relevant information on why this is crashing.

xlz, 2015-09-16:
Missing DLL surely prevents program from running. The problem is why the particular DLL is referenced.

How did you fix the debug library issue? I don't see it. Is it copying MSCVR120D.dll from elsewhere, or deleting CMAKE_DEBUG_POSTFIX, or not building a debug profile that made the difference?

mikezucc, 2015-09-16:
Anyway, this could all just be an issue with my usb 3 buses on my Asus laptop not actually able to support the bandwidth necessary and thus crashing libusb or something. I had a problem with this when running stereo IR cameras a few months back. I think im going to buy a new workstation soon.

I deleted the CMAKE_DEBUG_POSTFIX block completely, and then altered the name of the dll file in sys32 just to make sure that was not the issue. I just copied the MSCVR120D.dll from another computer I had which was running 2013 visual studio. I am using the 2015 community studio version available for free from microsoft

mikezucc, 2015-09-16:
It would be nice if someone just hosted all of the binaries for the protonect so that we could just do a sanity check on hardware and driver set up. Is there a place to find them?

xlz, 2015-09-16:
I know you did these things. My question is which one of them made Protonect work (, one without which Protonect does not work).

mikezucc, 2015-09-16:
Protonect (Release 64 bit) is still giving me the exception with the address. The one error with the missing dll was easy, just found and copied the MSCVR120D.dll.

xlz, 2015-09-16:
You said you fixed the debug library issue. But you then said it still does not work. So you did not fix the debug library issue?

Sorry, I am unable understand what you are saying.

`Unable to start correctly (0xc000007b` anyway you can't simply copy dlls from anywhere. Your assumption was correct, versions would mismatch.

MSCVR120d.dll is presumably the debug version of MSCVR120.dll. So if you were building a debug profile, it would surely link with MSCVR120d.dll; release profile with MSCVR120.dll. So which way did you do?

If this is something wrong with your runtime environment setup, there is nothing libfreenect2 can do.

closed 2015-11-09 by floe