OpenKinect/libfreenect2 #389: [windows]Runtime error occurs instantly after execution

pierceh89, 2015-09-15:
Hi,
I have libfreenect2 installed using instructions on readme file. My environment is windows 8.1 and VS2013.
I had no errors during the build. I put all necessary dlls in the folder I guess (freenect2.dll, libusb-1.0.dll, glfw3.dll). However once I run the program, error occurs and the program stops working.
So I compiled them again in debug mode, and I got this message.
[image]

Thank you in advance for answer!

pierceh89, 2015-09-15:
I have compiled the source downloaded on 18th Aug and it worked well, so it seemed there is a problem in the latest version of the source

floe, 2015-09-15:
If you have the most recent master, can you try commenting out these lines: https://github.com/OpenKinect/libfreenect2/blob/master/examples/protonect/Protonect.cpp#L98-L100 and re-run Protonect? /cc @xlz

pierceh89, 2015-09-15:
Thank you very much! It resolved the problem!

closed 2015-09-15 by pierceh89

xlz, 2015-09-15:
I didn't find anything in docs that says `std::ofstream::ofstream(NULL)` is invalid. But if MSVC does this check, I think I have to err on the safe side and add that NULL check.