OpenKinect/libfreenect2 #390: libusb-1.0.20 release and PPA

floe, 2015-09-15:
As pointed out by @xlz, libusb has a new release 1.0.20. Luckily, this now does work with libfreenect2, at least on Intel controllers.

I've been so fed up with the whole libusb-multiple-version-RPATH mess that I quickly created an Ubuntu PPA for the new version: https://launchpad.net/~floe/+archive/ubuntu/libusb

Not sure how to handle this for Debian, though - suggestions?

xlz, 2015-09-15:
I submitted a bug report to Debian for the new release. https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=799069

Hopefully the new release is packaged within one or two months, then Debian unstable, testing will have it. Stable probably will not have it. In the mean time, the build-from-source instructions are still useful. Once a binary package is released, users can download that.

xlz, 2015-09-15:
Debian has updated their release. Very fast.

floe, 2015-09-16:
OK, to summarize how to get libusb-1.0.20 ATM:
* Ubuntu: use my PPA on 14.x, or download .deb files from Launchpad
* Debian: use unstable/testing, or download .deb files manually
* MacOSX: install via `brew` (?)
* Windows: get binaries from libusb homepage

xlz, 2015-09-16:
Windows: still have to build from source with joshblake branch. libusb 1.0.20 does not seem to include Windows usb3 iso support.

JoshBlake, 2015-09-16:
> libusb 1.0.20 does not seem to include Windows usb3 iso support.

(╯°□°）╯︵ ┻━┻

metropt, 2015-09-17:
How do I check if I have the newest version installed?

xlz, 2015-09-17:
@metropt `apt-add-repository ppa:floe/libusb && apt-get update && apt-get install libusb-1.0-0-dev`

floe, 2015-09-18:
Actually installed version can be shown with `dpkg -l libusb-1.0*`. If you have 1.0.20 there, then you can also delete the `libfreenect2/depends/libusb*` folders, maybe it's still picking up the previous self-compiled version.

floe, 2015-09-18:
I should add that right now, this repo only publishes packages for 14.04 (trusty). I'll change that soon.

ahundt, 2015-09-19:
can this update also be verified to make its way in on OS X's homebrew?

ahundt, 2015-09-19:
looks like someone submitted a pull request over there but there is a small bug that would have to be resolved before it is merged https://github.com/Homebrew/homebrew/pull/44038

fran6co, 2015-09-24:
Just merged for OS X.

floe, 2015-09-24:
Great, but does it actually work with 1.0.20 on OSX? Has anyone tested yet?

fran6co, 2015-09-24:
I'll test it tomorrow.

floe, 2015-09-29:
Any news for OSX? I've updated my PPA so it now also supports vivid (15.04).

fran6co, 2015-10-05:
Sorry for the delay but I couldn't get a hold of a Kinect2. I tried the 1.0.20 in Mac OSX 10.11, it mostly works but it throws some errors that the bundled 1.0.19 doesn't and at some point hangs:

```
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_OTHER Other error
...
[Debug] [DepthPacketStreamParser] subpacket too large
[Debug] [DepthPacketStreamParser] not all subsequences received 64
[Debug] [DepthPacketStreamParser] not all subsequences received 128
[Debug] [DepthPacketStreamParser] not all subsequences received 8
[Debug] [DepthPacketStreamParser] not all subsequences received 128
[Debug] [DepthPacketStreamParser] subpacket too large
```

fran6co, 2015-10-05:
Got a bit more data about it with ```LIBUSB_DEBUG=4```:

```
[ 1.276235] [00002803] libusb: debug [libusb_get_next_timeout] next timeout in 0.990067s
[ 1.276252] [00002803] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 1.276257] [00002803] libusb: debug [handle_events] poll() 1 fds with timeout in 100ms
[ 1.276290] [0000050f] libusb: debug [libusb_submit_transfer] transfer 0x7fb281c1af28
[ 1.276299] [0000050f] libusb: debug [ep_to_pipeRef] converting ep address 0x84 to pipeRef and interface
[ 1.276303] [0000050f] libusb: debug [ep_to_pipeRef] pipe 1 on interface 1 matches
[ 1.276416] [0000050f] libusb: debug [libusb_submit_transfer] transfer 0x7fb281e154a8
[ 1.276427] [0000050f] libusb: debug [ep_to_pipeRef] converting ep address 0x84 to pipeRef and interface
[ 1.276431] [0000050f] libusb: debug [ep_to_pipeRef] pipe 1 on interface 1 matches
[ 1.276585] [0000050f] libusb: debug [libusb_submit_transfer] transfer 0x7fb281e15648
[ 1.276611] [0000050f] libusb: debug [ep_to_pipeRef] converting ep address 0x84 to pipeRef and interface
[ 1.276615] [0000050f] libusb: debug [ep_to_pipeRef] pipe 1 on interface 1 matches
[ 1.276753] [0000050f] libusb: debug [libusb_submit_transfer] transfer 0x7fb281e157e8
[ 1.276768] [0000050f] libusb: debug [ep_to_pipeRef] converting ep address 0x84 to pipeRef and interface
[ 1.276772] [0000050f] libusb: debug [ep_to_pipeRef] pipe 1 on interface 1 matches
[ 1.276934] [0000050f] libusb: debug [libusb_submit_transfer] transfer 0x7fb281e15988
[ 1.276943] [0000050f] libusb: debug [ep_to_pipeRef] converting ep address 0x84 to pipeRef and interface
[ 1.276947] [0000050f] libusb: debug [ep_to_pipeRef] pipe 1 on interface 1 matches
[ 1.277064] [00001907] libusb: debug [darwin_async_io_callback] an async io operation has completed
[ 1.277080] [0000050f] libusb: debug [libusb_submit_transfer] transfer 0x7fb281e15b28
[ 1.277091] [0000050f] libusb: debug [ep_to_pipeRef] converting ep address 0x84 to pipeRef and interface
[ 1.277095] [0000050f] libusb: debug [ep_to_pipeRef] pipe 1 on interface 1 matches
[ 1.277115] [00002803] libusb: debug [handle_events] poll() returned 1
[ 1.277121] [00002803] libusb: debug [handle_events] caught a fish on the event pipe
[ 1.277124] [00002803] libusb: debug [darwin_handle_transfer_completion] handling isoc completion with kernel status 0
[ 1.277127] [00002803] libusb: debug [usbi_handle_transfer_completion] transfer 0x7fb281e130e8 has callback 0x10e3c5830
[ 1.277131] [00002803] libusb: debug [libusb_submit_transfer] transfer 0x7fb281e130e8
[ 1.277138] [00002803] libusb: debug [ep_to_pipeRef] converting ep address 0x84 to pipeRef and interface
[ 1.277141] [00002803] libusb: debug [ep_to_pipeRef] pipe 1 on interface 1 matches
[ 1.277220] [0000050f] libusb: debug [libusb_submit_transfer] transfer 0x7fb281e15cc8
[ 1.277229] [0000050f] libusb: debug [ep_to_pipeRef] converting ep address 0x84 to pipeRef and interface
[ 1.277234] [0000050f] libusb: debug [ep_to_pipeRef] pipe 1 on interface 1 matches
[ 1.277251] [00002803] libusb: error [submit_iso_transfer] isochronous transfer failed (dir: In): unknown error (0xe00002ee)
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_OTHER Other error
```

xlz, 2015-10-05:
`0xe00002ee` means kIOReturnIsoTooOld "isochronous I/O request for distant past!". I found a similar thread http://thread.gmane.org/gmane.comp.lib.libusb.devel.general/10637. Will take a look later.

xlz, 2015-10-05:
Depending on how many "[usb::TransferPool] failed to submit transfer" errors you saw. There is this transfer pool containing 40 transfers. If all of them failed, I suppose there would be no data to process, and it would appear hanging.

xlz, 2015-10-09:
I didn't find anything useful about kIOReturnIsoTooOld. But I noticed you have Mac OSX 10.11 - El Capitan. This is a recent release. Does libusb 1.0.20 have the same problem with OSX 10.10? This is one factor to investigate.

The other thing left to do is bisect between libusb 1.0.19 and 1.0.20.

fran6co, 2015-10-10:
I don't have 10.10 to test it out =/

Maybe it's just the new Kinect I got, I'm getting those errors with libusb bundled with libfreenect but with less frequency.

xlz, 2015-10-10:
Looking at https://github.com/libusb/libusb/commit/51b10191033ca3a3819dcf46e1da2465b99497c2, I think this patch is Linux-only and has nothing to do with Mac OSX?! I don't remember why Mac OSX was required to use `install_libusb.sh` to install an unreleased version of libusb.

I think it's totally fine or even better to install from homebrew with whatever version that works. So, if libusb 1.0.19 release works, then just use that. If libusb 1.0.20 does not work, probably report a bug at libusb or do nothing.

So the libusb installation readme would be:
* Linux - use libusb 1.0.20 binary (Ubuntu: use floe's ppa or from official repo if 1.0.20; Debian: use testing or sid); if no binary from distro, build the 1.0.20 release version and point cmake to it.
* Mac OSX - use homebrew/macports to install 1.0.19 or 1.0.20 (the latter of which might not work on 10.11 - El Capitan)
* Windows - git clone libusb, git merge joshblake, use visual studio to build (hopefully in 1.0.21 binary release includes the usbdk backend for usb3 iso transfer)

I'm not sure what are the steps to install old version 1.0.19 with homebrew/macports. Perhaps you can create a PR to update Mac OSX docs about this.

fran6co, 2015-10-10:
I think 1.0.20 is ok for mac osx, it's working more or less the same as 1.0.19.

xlz, 2015-10-10:
Yeah but what about your LIBUSB_ERROR_OTHER?

fran6co, 2015-10-10:
It's happening in 1.0.19 too, just a bit less

closed 2016-01-27 by xlz

BrownRiceRice, 2016-02-04:
Any chance the PPA can be updated to willy (15.10)?

xlz, 2016-02-04:
You can pull it from xenial directly http://packages.ubuntu.com/xenial/libusb-1.0-0. The dependencies are the same.

qzwelt, 2016-03-07:
I want to install kinect2.0 on linux, which "libusb" should I use?
I find a lesson about Ubuntu14.04+kinect2.0,but my system is Ubuntu15.10.
What can I do?
Looking forward your reply.

BrownRiceRice, 2016-03-07:
I think you need to use libusb >= 1.0.20. On 15.10, you can get it from the link xlz posted right above you.

qzwelt, 2016-03-07:
Thank you for your reply.
I have found "libusb-1.0.20 for Ubuntu 14.04",but it doesn't work. 
If I follow this lesson,could I success to install kinect2.0 on Ubuntu 15.10?
https://github.com/OpenKinect/libfreenect2/blob/master/README.md#linux

qzwelt, 2016-03-11:
Sorry，I still can't install "libusb" for Ubuntu 15.10.
Everytime I run "update",it tells me "Failed to fetch http://ppa.launchpad.net/floe/libusb/ubuntu/dists/wily/main/binary-armhf/Packages 404 Not Found.".
When I follow  the link xlz posted,there has a erro like "Unable to locate package packages.ubuntu.com/xenial;Couldn't find any package by regex 'packages.ubuntu.com/xenial'; ".
Which command I could do to make it right?

BrownRiceRice, 2016-03-11:
So, adding the PPA won't work for 15.10, since the PPA only has packages for trusty and vivid, not wily.  That's what the 404 comes from. There's not a URL with ".../wily/..." in it.   As for xlz's link, did you try to add that link directly as a PPA? I don't think that'd work, I just downloaded the deb from that page and call `sudo dpgk -i /path/to/deb` and then `sudo apt-get install -f`

qzwelt, 2016-03-11:
I have tried that thread,but it doesn't wok.
I call "sudo apt-get install http://packages.ubuntu.com/xenial/libusb-1.0-0",there comes the erro "couldn't find the package".
And where is "deb"?  I can't find it.

BrownRiceRice, 2016-03-11:
Here's everything: Go to http://packages.ubuntu.com/xenial/libusb-1.0-0. Under the section that says "Download libusb-1.0-0," click on which architecture you have (amd64 for 64 bit and i386 for 32 bit). Then choose which mirror is best for you, and it should download the .deb file, which is just a binary package for ubuntu. Find where it downloaded to your system, cd to that directory, and then type those two commands I mentioned earlier.
(`sudo dpgk -i /path/to/deb` and `sudo apt-get install -f`). That should install the library.