OpenKinect/libfreenect2 #394: USB droping connection

metropt, 2015-09-17:
Hello,

I'm having problems to make it work on 2 computers of 5 I tried. I tried on two desktops and a laptop and it worked but on two mini itx motherboard I can't make it work, I think it could be related.

This is the debug I got:
```
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 13 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:3 with serial 505160442542
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] ReadData0x14 response
[Debug] [Freenect2DeviceImpl] 92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 07 14 0f 47 4d 46 36 33 31 2e 31    31332e2e2e2e2e2e474d463633312e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e
 
[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  1d 26 00 00                                        2e262e2e
 
[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  03 26 00 00                                        2e262e2e
 
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 505160442542
device firmware: 4.3.3912.0.7
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NO_DEVICE No such device (it may have been disconnected)
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NO_DEVICE No such device (it may have been disconnected)
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERR....
```
The following pastebin is about what I got on dmesg http://pastebin.com/QqMKFaC1 and http://pastebin.com/SkLRfNkE

Finally the USB and libusb info:

00:14.0 USB controller: Intel Corporation 8 Series/C220 Series Chipset Family USB xHCI (rev 05)

-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
-- checking for module 'libusb-1.0'
--   found libusb-1.0, version 1.0.19

All the PCs has the same Ubuntu 14.10 64bits version.

I tried to run Microsoft Kinect SDK for windows and it worked so the USB controller work with it. Could it be some USB power definition because it is a mini itx board?

xlz, 2015-09-17:
This is similar to the previously reported #216. What we know or we can do about this is really limited.

Some more things you can try:

`LIBUSB_DEBUG=4 ./Protonect` and post to pastebin.

Use a new kernel (which includes updated USB drivers), perhaps http://packages.ubuntu.com/wily/linux-image-4.2.0-10-generic.

metropt, 2015-09-18:
Hi,

Here is what i have got with LIBUSB_DEBUG=4 ./Protonect:
http://pastebin.com/A1qPQFSc

I upgraded my kernel but the keyboard & mouse stopped working so i had to install ubuntu again.
I also tried to upgrade to libusb 1.0.20 but it keeps showing the previous version:
-- checking for module 'libusb-1.0'
-- found libusb-1.0, version 1.0.19

xlz, 2015-09-18:
I suppose you can always select older kernel in grub during boot.

You need to remove libusb in depends directory.

xlz, 2015-09-18:
Well, beyond these I don't really know what is the issue. My best guess is this is a kernel/USB driver/hardware support issue.

closed 2015-11-09 by floe