OpenKinect/libfreenect2 #395: [OpenGLDepthPacketProcessor] fragment shader fails to compile

wjvds, 2015-09-17:
When I try to run the protonect example with the OpenGL packet processor (on Windows 7, AMD FirePro GPU), my log shows the following message:

[Error] [ShaderProgram] failed to compile fragment shader!
Fragment shader failed to compile with the following errors:
ERROR: 2:91: error(#203) More than one matching overloaded function are found mix
ERROR: 2:92: error(#203) More than one matching overloaded function are found mix
ERROR: 2:93: error(#203) More than one matching overloaded function are found mix
ERROR: error(#273) 3 compilation errors.  No code generated

Looking around in the source code, I find a TODO comment just above line 91 in the shader/filter1.fs file, followed by three lines with a call to mix. I tried to change these lines, but the message remained the same - maybe something was cached somewhere? I am not familiar enough with this codebase to figure this out myself. Since I can't get the OpenCL backend to work with my GPU either, I am not sure how I can use this library now.

xlz, 2015-09-17:
In which ways did you try to change these lines that still did not work?

I would try it with `mix(dist.x, 0.0f, c1.x)`, since it is complaining about overloading problem.

wjvds, 2015-09-17:
I tried commenting out the lines entirely, and the error still remained - either it came from a different shader or the shader is cached somewhere.

xlz, 2015-09-17:
Did you rebuild after changing the shader files? The files are embedded into the binary during building.

wjvds, 2015-09-18:
I have cleaned and rebuilt the entire solution and the error does not change.

xlz, 2015-09-18:
The only explanation is the changed shader file was not built into your library. I don't know how.

wjvds, 2015-09-18:
I have noticed something possibly related: if I change the line "Failed to compile fragment shader!" in the opengl_depth_packet_processor.cpp file to something different and rebuild the solution, the log still shows the old string. I feel like the recompilation is not working correctly somehow, but I cannot figure out what is exactly going wrong.

wjvds, 2015-09-18:
It turned out the recompilation went just fine, but I forgot to copy the new freenect2(d).dll file to my bin folder, so it used the old build every time, even when launched from visual studio.
Now, for whatever reason, the old fragment shader does compile without any changes so I am not sure what is going on there but it works. Thanks for your help anyway.

closed 2015-09-18 by wjvds

floe, 2015-09-18:
@xlz just curious, is this something requiring a general workaround in the shader code?

xlz, 2015-09-18:
@floe OP could not reproduce anymore. I don't see anything obviously wrong.