OpenKinect/libfreenect2 #396: libusb error while trying to run multiple kinects

vaheta, 2015-09-18:
Hi!
I'm trying to run two Kinects v2 on the same machine (ubuntu 14.04). The one is connected to motherboard USB3 host, and the other to PCI-E NEC host. Both work fine separately, but if I run one and then run another while the first is running, I get the following error for the second kinect:

[Error] [protocol::CommandTransaction] bulk transfer failed! libusb error -1: LIBUSB_ERROR_IO

I also tried some other PCI-E hosts - that didn't help. Does anyone know how to fix it?

Thanks!

floe, 2015-09-18:
Please check the FAQ: https://github.com/OpenKinect/libfreenect2#can-i-use-multiple-kinect-v2-devices-on-one-machine - did you already try increasing USBFS memory buffers?

vaheta, 2015-09-18:
Yes, we use the NEC controller as in the FAQ and played a lot with memory buffers, putting values from 16 to 512 mb - no result.

floe, 2015-09-18:
Hmm, not sure if you might have to specify it at the kernel commandline, IIRC usbcore is already built into the kernel on Ubuntu.

Two other things to check:
* get a recent kernel, something >= 3.19 (should be available directly as Ubuntu LTS package)
* see if your PCI-E card is, by chance, in a PCI-E x1 slot. I had a similar issue recently, and it vanished once I plugged the extension card into an x16 slot.

vaheta, 2015-09-18:
Thanks for ideas! 
But unfortunately, the kernel is already 3.19, and the card is in x16 slot. Just to mention also, the motherboard is ASUS Z87-PLUS with ASMedia USB3 controller. It works fine with single Kinect.

vaheta, 2015-09-18:
Ok, problem solved. 
You were right, I didn't set the memory buffer correctly. Edited grub, now works fine! 
Thank you!

closed 2015-09-18 by vaheta