OpenKinect/libfreenect2 #398: Crash after a few minutes

wjvds, 2015-09-22:
When I run the Protonect example for a longer continuous period of time (generally it happens after around 2 minutes), the window will freeze and the program will hang. The log file will at this point do something like this:

*Lots of skipping depth packets before this*
[Debug] [DepthPacketStreamParser] skipping depth packet
[Info] [TurboJpegRgbPacketProcessor] avg. time: 21.0809ms -> ~47.4364Hz
[Debug] [DepthPacketStreamParser] subpacket too large
[Info] [TurboJpegRgbPacketProcessor] avg. time: 20.8406ms -> ~47.9832Hz
[Debug] [DepthPacketStreamParser] image data too short!
[Debug] [DepthPacketStreamParser] subpacket too large
*Lots of subpackets too large after this*

This problem occurs literally every time I try it, both using the debug and release builds.
I am using the OpenGL depth packet parsing backend on an AMD FirePro M4000 GPU.

xlz, 2015-09-22:
Cpu throttling? Memory leak? Do you observe anything on system level?

Also, `ulimit -c umlimited; pkill -QUIT Protonect` to get a core dump and inspect where it was hanging.

ronin8600, 2015-09-22:
Are you running on Windows 7?

I'm having the same issue when I run Protonect in Windows 7.  If I run Protonect on the same computer but with Linux (Fedora 22) or Windows 8 Protonect runs perfectly.

Could this be a driver problem?  

I'm using the latest Intel USB 3 Drivers for the Intel 8 Series Chipset / C220

xlz, 2015-09-29:
@wjvds any update?

wjvds, 2015-10-01:
Updating my USB drivers (Intel 7 Series C216) and GPU drivers seems to have improved it a little bit. It seems to skip depth packets a lot less now too. CPU, GPU and memory is not used very intensively. I cannot test on different operating systems from this machine right now so I'm kind of stuck here.

xlz, 2015-10-01:
@wjvds Can you try `Protonect.exe >NUL` to discard console output. It might be the slow console scrolling that made it hanging.

xlz, 2015-10-02:
@wjvds Also, can you run a debug build from Visual Studio and see where it crashes?

xlz, 2015-10-13:
This is a duplicate issue with #414. I will follow in #414.

floe, 2015-10-23:
Closing as duplicate.

closed 2015-10-23 by floe