OpenKinect/libfreenect2 #406: Potential crash in cpu_depth_packet_processor.cpp

alberth, 2015-09-25:
While looking into simplifying the *cpu_depth_packet_processor.cpp* file, in particular the *processPixelStage2* method, I ran into a potential crash, due to division by zero.
(git commit caf7d8679674c5129d27e896fbe9aaed71f19fee which is current or very recent master)

The problem starts with 'phase' becoming 0, which can happen either by a simple assingment in one branch, or by a long computation in another branch:
````c
658: phase = 0;

704: float mask = t9 >= 0.0f ? 1.0f : 0.0f;
705: t10 *= mask;
720: float mask2 = ir_x >= norm ? 1.0f : 0.0f;
722: float t11 = t10 * mask2;
// Omitting mask3 lines that changes t10
726: phase = true/*(modeMask & 2) != 0*/ ? t11 : t10;
```
Then in the depth calculation starting at line 730:
````c
734: phase = 0 < phase ? phase + params.phase_offset : phase;  // False condition, phase stays 0
737: float max_depth = phase * params.unambigious_dist * 2; // 0 due to phase being 0
741: xmultiplier = (xmultiplier * 90) / (max_depth * max_depth * 8192.0); // Crash on division by zero.
````

xlz, 2015-09-25:
I don't see any crash.

`max_depth = 0` then `xmultiplier = INFINITY;`.

Then `depth_fit = xxx / (-INFINITY + 1) = -0.0f;`.

Then `depth_fit = depth_fit < 0 ? 0 : depth_fit;` converts it back to 0.

alberth, 2015-09-28:
I was thinking too much regular mathematics apparently.

Thank you for pointing out my error, and great that it was actually just a false positive. Closing issue.

closed 2015-09-28 by alberth