OpenKinect/libfreenect2 #407: Computing depth/IR values from Frame's data field

gardaud, 2015-09-25:
I'm using the libfreenect2 C++ APIs to get color/depth/IR data from a Kinect v2 and display them in an application/do some post processing on it.

I listen for Color and IR/Depth frames, and am getting the frames properly. The color frames I can display in my application just fine.

But I'm having trouble extracting meaningful values from the depth/IR frames.

For example for depth, I'm reading that the Kinect is encoding depth data on the 11 highest bits of the 2 least significant bytes. So every time I get a new depth frame, I call the following:

``` 
int* bwValues = (int*)malloc(frame->width * frame->height); //this is where I want the full depth values to be stored

for (int i = 0; i < frame->width * frame->height; i++){
        unsigned char a = frame[i*4];
        unsigned char r = frame[i*4 + 1];
        unsigned char g = frame[i*4 + 2];
        unsigned char b = frame[i*4 + 3];

        int value = (g << 8) + b; //get the 2 LSB as a single value
        value = (value >> 5); //we only care about the 11 highest bits
        
        bwValues[i] = value;
}
``` 

If I map the values back to [0-255] and display them as a B&W image, it looks almost quite right, but I'm clearly missing something. The depth data exhibits some artefacts, and the IR data gets modulation banding.

What am I missing?
Thanks!

xlz, 2015-09-25:
For Depth/IR frames, use
```cpp
float *frame_data = (float *)frame->data;
float value = frame_data[y*width+x];
```

gardaud, 2015-09-25:
This did the trick, thanks! I now get [0-4500] values for depth, and [0-65535] values for IR.

closed 2015-09-25 by gardaud

jav-solo, 2019-03-20:
> For Depth/IR frames, use
> 
> ```c++
> float *frame_data = (float *)frame->data;
> float value = frame_data[y*width+x];
> ```

Are x and y the pixel values in this example?