OpenKinect/libfreenect2 #408: clGetDeviceID Issue

rahulraw, 2015-09-25:
I keep getting the following error when trying to run rosrun kinect2_bridge kinect2_bridge. Any clue what's going on? Using a ThinkPad with an Integrated Graphics Controller
```
modprobe: ERROR: could not insert 'fglrx': No such device
Error: Fail to load fglrx kernel module!
Error! Fail to load fglrx kernel module! Maybe you can switch to root user to load kernel module directly
modprobe: ERROR: could not insert 'fglrx': No such device
Error: Fail to load fglrx kernel module!
Error! Fail to load fglrx kernel module! Maybe you can switch to root user to load kernel module directly
modprobe: ERROR: could not insert 'fglrx': No such device
Error: Fail to load fglrx kernel module!
Error! Fail to load fglrx kernel module! Maybe you can switch to root user to load kernel module directly
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/rrawat/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel(R) Core(TM) i5-4300M CPU @ 2.60GHz (CPU)[GenuineIntel]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Intel(R) Core(TM) i5-4300M CPU @ 2.60GHz (CPU)[GenuineIntel]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Warning] [OpenCLDepthPacketProcessorImpl] Your ocl-icd has deadlock bugs. Update to 2.2.4+ is recommended.
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 12 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:5 with serial 005262641947
[Info] [Freenect2Impl] found 1 devices
Kinect2 devices found: 
  0: 005262641947 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened

starting kinect2

[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started

device serial: 005262641947
device firmware: 1.1.3323.0.7
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] disabling usb transfer submission...
[Info] [Freenect2DeviceImpl] canceling usb transfers...
[Info] [Freenect2DeviceImpl] stopped

default ir camera parameters: 
fx 365.688, fy 365.688, cx 258.733, cy 208.231
k1 0.0911368, k2 -0.271956, p1 0, p2 0, k3 0.0955618

default color camera parameters: 
fx 1081.37, fy 1081.37, cx 959.5, cy 539.5
using sensor defaults for color intrinsic parameters.
using sensor defaults for ir intrinsic parameters.
using defaults for rotation and translation.
using defaults for depth shift.

camera parameters used:
camera matrix color:
[1081.3720703125, 0, 959.5;
  0, 1081.3720703125, 539.5;
  0, 0, 1]
distortion coefficients color:
[0, 0, 0, 0, 0]
camera matrix ir:
[365.6881103515625, 0, 258.7327880859375;
  0, 365.6881103515625, 208.2306060791016;
  0, 0, 1]
distortion coefficients ir:
[0.09113681316375732, -0.2719560861587524, 0, 0, 0.09556184709072113]
camera matrix depth:
[365.6881103515625, 0, 258.7327880859375;
  0, 365.6881103515625, 208.2306060791016;
  0, 0, 1]
distortion coefficients depth:
[0.09113681316375732, -0.2719560861587524, 0, 0, 0.09556184709072113]
rotation:
[1, 0, 0;
  0, 1, 0;
  0, 0, 1]
translation:
[0;
  0;
  0]
depth shift:
0

[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistrationOpenCL::init] ERROR: clGetDeviceIDs(-1)
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
Initialization failed!
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing
```

xlz, 2015-09-26:
`[DepthRegistrationOpenCL::init] ERROR: clGetDeviceIDs(-1)` this error totally comes from iai_kinect2.

But let us know if this error did not happen with previous libfreenect2, in which case please provide the last commit with which iai_kinect2 does not exhibit this behavior.

rahulraw, 2015-09-27:
I believe you are correct. 
This issue was resolved after installing the appropriate Beignet v1.0 dependencies.

closed 2015-09-27 by rahulraw