OpenKinect/libfreenect2 #409: ERROR: 65543 Failed to create opengl window

rahulraw, 2015-09-28:
I'm getting the following error when I try to run ./Protonect 
I was wondering if anyone has faced a similar situation 
```
administrator@pr2test:~/catkin_ws/src/libfreenect2/depends/bin$ ./Protonect 
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 13 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:4 with serial 005262641947
[Info] [Freenect2Impl] found 1 devices
[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65543 The requested client API version is unavailable
[Error] [OpenGLDepthPacketProcessor] Failed to create opengl window.
```
FYI: I'm using a Nvidia Quadro card and install the necessary nvidia headers. 
I even downloaded opengl from source and used cmake/make/install just in case.
I'm using Ubuntu 14.04.3 Server

floe, 2015-09-29:
Hm, does it work (although probably slowly) if you run `./Protonect cpu`?

rahulraw, 2015-09-29:
it does not. I get the following output:
```
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 13 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:4 with serial 005262641947
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] ReadData0x14 response
[Debug] [Freenect2DeviceImpl] 92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 03 0e 13 47 4d 43 38 30 34 2e 31    31332e2e2e2e2e2e474d433830342e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  23 26 00 00                                        23262e2e

[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  23 26 00 00                                        23262e2e

[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 005262641947
device firmware: 1.1.3323.0.7
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
Segmentation fault (core dumped)
```

xlz, 2015-09-29:
`The requested client API version is unavailable` This asks for your OpenGL version. So what is it? `glxinfo | grep OpenGL`.

Also, `./Protonect cpu` should not segfault. Can you produce a stack trace? `ulimit -c unlimited; ./Protonect cpu; gdb ./Protonect core`

rahulraw, 2015-09-30:
Thanks for the prompt reply.
This is actually a graphics driver issue on Ubuntu. After installing I removed my old NVIDIA drivers and installed them in addition to Nouveau using Software & Updates tool found on Ubuntu.
To note the version installed is: Version 346.96 from nvidia-346 
                                Card used: NVIDIA Quadro K620

closed 2015-09-30 by rahulraw

jacksonkr, 2016-04-27:
I'm having the same issue with AMD drivers.

`rtabmap` works fine with kinect v1 but not kinect v2. Both `Protonect cl` and `Protonect cpu` work but `Protonect gl` fails to render depth so you only see IR and RGB. 

```
$ rtabmap
[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65543 The requested client API version is unavailable
[Error] [OpenGLDepthPacketProcessor] Failed to create opengl window.
jackson@jackson-MacBookPro:~/rtabmap/build$ glxinfo | grep "OpenGL version"
OpenGL version string: 4.5.13399 Compatibility Profile Context 15.201.1151

$ lspci | grep "VGA"
01:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Venus XT [Radeon HD 8870M / R9 M270X/M370X] (rev 83)

$ fglrxinfo
display: :0  screen: 0
OpenGL vendor string: Advanced Micro Devices, Inc.
OpenGL renderer string: AMD Radeon R9 (TM) M370X
OpenGL version string: 4.5.13399 Compatibility Profile Context 15.201.1151

```

mbp2015, Ubuntu 14.04 x64