OpenKinect/libfreenect2 #410: Error Freenect2Impl Your libusb does not support large iso buffer! no device connected!

Markthenuke, 2015-09-30:
Hello, 

I am having problems running Protonect. Whenever I try to run it I get the following error.

[Error] [Freenect2Impl] Your libusb does not support large iso buffer!
no device connected!

I believe I have the libfreenect2 installed correctly and I am using the libusb that was included in the project. I have also tried this with the most recent libusb available, which was libusb-1.0.20 .

Things about my system:
Ubuntu 15.04

Thanks in advance for the help!

xlz, 2015-10-01:
`ldd ./Protonect | grep libusb`?

This message says the linked libusb is 1.0.19-.

Markthenuke, 2015-10-01:
Here is the output.
mark@Lab-Rat:~/libfreenect2/build/bin$ ldd ./Protonect | grep libusb
	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007f7bf7c96000)

Should I be using libusb 1.0.19-

xlz, 2015-10-01:
You said you have libusb 1.0.20. `dpkg -l libusb-1.0-0`?

Markthenuke, 2015-10-01:
I meant that I had received the same error with both libusb 1.0.20 and libusb 1.0.0  sorry for the confusion. 

mark@Lab-Rat:~/libfreenect2/depends$ dpkg -l libusb-1.0-0
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                           Version              Architecture         Description
+++-==============================-====================-====================
ii  libusb-1.0-0:amd64             2:1.0.19-1           amd64                userspace USB programming library
ii  libusb-1.0-0:i386              2:1.0.19-1           i386                 userspace USB programming library

xlz, 2015-10-01:
You have libusb 1.0.19 installed in the system, and this version of libusb is linked to your libfreenect2. libfreenect2 does not allow old libusb to work because it produces very unintuitive errors.

I don't know which version of libusb is install in your ~/libfreenect2/depends. But it seems you never had libfreenect2 linked with libusb 1.0.20.

Markthenuke, 2015-10-02:
Thanks,
I now have  lib usb 1.0.20 installed, but I am still receiving an error. 

mark@Lab-Rat:/etc/udev/rules.d$ sudo ./Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 9 usb devices connected
[Info] [Freenect2Impl] found 0 devices
no device connected!

Markthenuke, 2015-10-02:
Sorry there was also a hardware problem. I now have it working. Thank you so much for the help!!!!!

closed 2015-10-04 by floe

djo90, 2015-11-27:
when I did rosrun kinect2_bridge kinect2_bridge I got this error message
[rospack] Error: package 'kinect2_bridge' not found

so I tried those command 
sudo -i
source /home/djo/catkin_ws/devel/setup.bash
rosrun kinect2_bridge kinect2_bridge

I got this error 
[Error] [Freenect2Impl] Your libusb does not support large iso buffer!

So I tried your command 
ldd ./Protonect | grep libusb

and I got this message 
	libusb-1.0.so.0 => /home/djo/libfreenect2/depends/libusb/lib/libusb-1.0.so.0 (0x00007f0eaebee000)

what should I do in my case to can run kinect2_bridge?

xlz, 2015-11-27:
First, please do not report iai_kinect2 and ROS bugs here. Use `Protonect` to test your bug.

Second, do not use sudo.

Now, please report again.

djo90, 2015-11-27:
How I can use Protonect? ./bin/Protonect ?

djo90, 2015-11-28:
I got this error when running Protonect. How can I solve it?

[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Error] [OpenCLDepthPacketProcessorImpl] failed to build program: -11
[Error] [OpenCLDepthPacketProcessorImpl] Build Status: -2
[Error] [OpenCLDepthPacketProcessorImpl] Build Options:	 -D

floe, 2015-11-28:
Try `./bin/Protonect cpu`. And read the README, where all of this is already described.

djo90, 2015-12-01:
I did ./bin/Protonect cpu. I got this line repeated
[debug] [DepthPacketStreamParser] not all subsequences received 0
[debug] [DepthPacketStreamParser] skipping depth packet

How can I fix it?

floe, 2015-12-01:
Do you actually see any visual output (even if slow)?

djo90, 2015-12-01:
no nothing, but when I open ./bin/Protonect in depends folder. It works fine, I see 4 screens of my face from the kinect. But when I execute ./bin/Protonect from the libfreenect2 folder, it says that they found a valid kinect2 but there is any visual output.

floe, 2015-12-01:
OK, let's discuss this in the new issue #487 .