OpenKinect/libfreenect2 #411: error in libusb_set_configuration when opening second Kinect

hurzelchen, 2015-10-01:
Hi!

we're running in weird issues here on some machines when we're trying to open two kinects. On one machine it works perfectly fine, on two others the exact same code run's into an error:

In `UsbControl::setConfiguration()` the `current_config_id` equals `desired_config_id` for the first Kinect. Then it get's opened as expected.

For the second Kinect, `current_config_id` is `0`. `libusb_set_configuration` returns with `LIBUSB_ERROR_OTHER` and `errno` returning `28` (No space left on device).

We tried the suggested kernel boot option `usbcore.usbfs_memory_mb` with values ranging from 64 to 1024 but no visible effect.

Maybe you have any ideas what to keep trying?

xlz, 2015-10-02:
`LIBUSB_DEBUG=4 ./your_app` will output libusb debug information. Also, there may be some kernel messages from `dmesg`. Is there something relevant from the two sources?

hurzelchen, 2015-10-02:
One thing beforehand, this is the output of a machine where it did work before...

`LIBUSB_DEBUG` spams the output with the transfer of the first Kinect, I'll try to find the lines where it shows the error...

`dmesg` gives the following:

    [Fr Okt  2 13:08:31 2015] usb 2-2.1: reset SuperSpeed USB device number 5 using xhci_hcd
    [Fr Okt  2 13:08:31 2015] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff880212482c00
    [Fr Okt  2 13:08:31 2015] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff880212482c48
    [Fr Okt  2 13:08:31 2015] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff880212482cd8
    [Fr Okt  2 13:08:31 2015] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff880212482c90
    [Fr Okt  2 13:08:31 2015] usb 2-1.1: reset SuperSpeed USB device number 8 using xhci_hcd
    [Fr Okt  2 13:08:31 2015] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8802123e0800
    [Fr Okt  2 13:08:31 2015] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8802123e0848
    [Fr Okt  2 13:08:31 2015] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8802123e08d8
    [Fr Okt  2 13:08:31 2015] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8802123e0890
    [Fr Okt  2 13:08:31 2015] usb 2-1.1: Not enough bandwidth for new device state.
    [Fr Okt  2 13:08:31 2015] usb 2-1.1: Busted HC?  Not enough HCD resources for old configuration.
    [Fr Okt  2 13:08:31 2015] usb 2-1.1: USB disconnect, device number 8
    [Fr Okt  2 13:08:31 2015] usb 2-1.1: Not enough bandwidth for new device state.
    [Fr Okt  2 13:08:31 2015] usb 2-1.1: Not enough bandwidth for altsetting 1
    [Fr Okt  2 13:08:31 2015] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8802123e0800
    [Fr Okt  2 13:08:31 2015] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8802123e0848
    [Fr Okt  2 13:08:31 2015] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8802123e08d8
    [Fr Okt  2 13:08:31 2015] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8802123e0890
    [Fr Okt  2 13:08:31 2015] systemd-udevd[5215]: error opening ATTR{/sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1.1/2-1.1:1.2/sound/card3/controlC3/../uevent} for writing: No such file or directory
    [Fr Okt  2 13:08:31 2015] systemd-udevd[5215]: Failed to apply ACL on /dev/snd/controlC3: No such file or directory
    [Fr Okt  2 13:08:31 2015] systemd-udevd[5215]: Failed to apply ACL on /dev/snd/controlC3: No such file or directory
    [Fr Okt  2 13:08:31 2015] systemd-udevd[5238]: Failed to apply ACL on /dev/snd/pcmC3D0c: No such file or directory
    [Fr Okt  2 13:08:31 2015] systemd-udevd[5238]: Failed to apply ACL on /dev/snd/pcmC3D0c: No such file or directory
    [Fr Okt  2 13:08:32 2015] usb 2-1.1: new SuperSpeed USB device number 9 using xhci_hcd
    [Fr Okt  2 13:08:32 2015] usb 2-1.1: New USB device found, idVendor=045e, idProduct=02c4
    [Fr Okt  2 13:08:32 2015] usb 2-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=4
    [Fr Okt  2 13:08:32 2015] usb 2-1.1: Product: Xbox NUI Sensor
    [Fr Okt  2 13:08:32 2015] usb 2-1.1: Manufacturer: Microsoft
    [Fr Okt  2 13:08:32 2015] usb 2-1.1: SerialNumber: 036788334847
    [Fr Okt  2 13:08:32 2015] usb 2-1.1: Not enough bandwidth for new device state.
    [Fr Okt  2 13:08:32 2015] usb 2-1.1: can't set config #1, error -28
    [Fr Okt  2 13:08:32 2015] xhci_hcd 0000:00:14.0: WARN Event TRB for slot 7 ep 2 with no TDs queued?

Yes, i do see the `Not enough bandwidth for new device state`, but I do wonder why it did work before.

xlz, 2015-10-24:
Not enough bandwidth for the second Kinect. Maybe the other two machines literally do not have enough bandwidth.

Bandwidth requirement for each Kinect is 3 Gbps.

closed 2015-12-23 by floe