OpenKinect/libfreenect2 #412: On Ubuntu cmake fails for CMakeLists.txt is missing

jonnius, 2015-10-01:
OS: Ubuntu Gnome 15.10
Graphics: NVIDIA GeForce 840M (nvidia-346)

I was following the install instructions but at step 4. Build the actual protonect executable I encounter the following error message:

 $ cmake ..
 CMake Error: The source directory "/home/user/libfreenect2/depends" does not appear to contain CMakeLists.txt.
 Specify --help for usage, or press the help button on the CMake GUI.

What's wrong?

hurzelchen, 2015-10-01:
Have a closer look at the [installation instructions for Ubuntu](https://github.com/OpenKinect/libfreenect2#debianubuntu-1404-perhaps-earlier).

You're currently in `libfreenect2/depends` which is used in step 2.

cmake has to be called in step 4 in `libfreenect2/examples/protonect/build`.

jonnius, 2015-10-01:
You are right, sorry. But now that I am in the right directory, the last command does not succeed:

    $ sudo make install
    make: *** No rule to make target 'install'.  Schluss.

hurzelchen, 2015-10-01:
Did `cmake ..` and `make` finish without errors?

jonnius, 2015-10-01:
Yes, cmake finished with:
    -- Build files have been written to: /home/jonatan/.Programmieren/libfreenect2/build
and make didn't give any output.

floe, 2015-10-04:
OK, this is an error in the README, you don't actually have to run `make install`, the Protonect executable will already be in `bin` after just running `make`.

jonnius, 2015-10-04:
Actually no bin directory is beeing created by make. Therefore I cannot find the Protonect executable...

floe, 2015-10-05:
Can you put the full output from a clean `cmake` and `make` run on pastebin and post the link?

jonnius, 2015-10-05:
I cloned a fresh copy and this time it is working as expected. It seems to me that git downloaded more files this time. Maybe files were missing the first time I tried. Anyway, thanks for your help!

closed 2015-10-05 by jonnius