OpenKinect/libfreenect2 #413: Is the frametime generated on the kinect? Or from the computer time?

chipweinberger, 2015-10-02:
You guys have been making awesome progress on this library. Kudos.

When it makes sense to add, it'd be great to have more documentation to negate the need for questions like these =) Ill check out the source and see if I can answer it myself as well.

chipweinberger, 2015-10-02:
Seems to be from the device:

src/turbo_jpeg_rgb_packet_processor.cpp 

85	  if(impl_->decompressor != 0 && listener_ != 0)
86	  {
87	    impl_->startTiming();
88	
89	    impl_->frame->timestamp = packet.timestamp;

closed 2015-10-02 by chipweinberger

xlz, 2015-10-02:
The unit of this timestamp is not clear. You may have to measure it against real time.

chipweinberger, 2015-10-02:
xlz, thanks for always being so prompt to answer. It's super helpful.

From the looks of it, it appears to be either roughly or exactly 1/10 of a millisecond. I've noticed that if I fire up 3 kinects as simultaneously as I can, they can be off as much by 30,000 of these units or roughly 3 seconds. I wonder if there is a bettery in the kinect keeping time or how they are generated.

My kinects are at ~67736347 right now. I would think this resets every day or so.

Hope this helps in any way.