OpenKinect/libfreenect2 #416: NO device Connected

Chinkalp7, 2015-10-06:
When i tried to run ./Protonect it shows

chinkal@chinkal:~/libfreenect2/build/bin$ sudo ./Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 8 usb devices connected
[Info] [Freenect2Impl] found 0 devices
no device connected!

alberth, 2015-10-06:
kinect not plugged in (unplug it, and check you get 7 devices)?
not powered?
not using patched libusb (ldd ./Protonect should not point to the system libusb)?

Chinkalp7, 2015-10-06:
when i unplugged it it shows 7 devices..
output of ldd ./Protonect is 

chinkal@chinkal:~/libfreenect2/build/bin$ ldd ./Protonect
	linux-vdso.so.1 =>  (0x00007fffc85fc000)
	libfreenect2.so => /home/chinkal/libfreenect2/build/lib/libfreenect2.so (0x00007fd4add8a000)
	libglfw.so.3 => /usr/lib/x86_64-linux-gnu/libglfw.so.3 (0x00007fd4adb77000)
	libGL.so.1 => /usr/lib/x86_64-linux-gnu/mesa/libGL.so.1 (0x00007fd4ad8e5000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fd4ad5e1000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fd4ad3cb000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fd4ad006000)
	libusb-1.0.so.0 => /home/chinkal/libfreenect2/depends/libusb/lib/libusb-1.0.so.0 (0x00007fd4acdee000)
	libturbojpeg.so.0 => /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0 (0x00007fd4acb94000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fd4ac976000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fd4ac670000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007fd4ac33b000)
	libXrandr.so.2 => /usr/lib/x86_64-linux-gnu/libXrandr.so.2 (0x00007fd4ac131000)
	libXi.so.6 => /usr/lib/x86_64-linux-gnu/libXi.so.6 (0x00007fd4abf21000)
	libXxf86vm.so.1 => /usr/lib/x86_64-linux-gnu/libXxf86vm.so.1 (0x00007fd4abd1b000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fd4abb13000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007fd4ab8e9000)
	libglapi.so.0 => /usr/lib/x86_64-linux-gnu/libglapi.so.0 (0x00007fd4ab6bf000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007fd4ab4ad000)
	libXdamage.so.1 => /usr/lib/x86_64-linux-gnu/libXdamage.so.1 (0x00007fd4ab2aa000)
	libXfixes.so.3 => /usr/lib/x86_64-linux-gnu/libXfixes.so.3 (0x00007fd4ab0a4000)
	libX11-xcb.so.1 => /usr/lib/x86_64-linux-gnu/libX11-xcb.so.1 (0x00007fd4aaea2000)
	libxcb-glx.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-glx.so.0 (0x00007fd4aac8b000)
	libxcb-dri2.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-dri2.so.0 (0x00007fd4aaa86000)
	libxcb-dri3.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-dri3.so.0 (0x00007fd4aa883000)
	libxcb-present.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-present.so.0 (0x00007fd4aa680000)
	libxcb-sync.so.1 => /usr/lib/x86_64-linux-gnu/libxcb-sync.so.1 (0x00007fd4aa47a000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007fd4aa25b000)
	libxshmfence.so.1 => /usr/lib/x86_64-linux-gnu/libxshmfence.so.1 (0x00007fd4aa059000)
	libdrm.so.2 => /usr/lib/x86_64-linux-gnu/libdrm.so.2 (0x00007fd4a9e4d000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fd4a9c49000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fd4ae177000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007fd4a9a38000)
	libXrender.so.1 => /usr/lib/x86_64-linux-gnu/libXrender.so.1 (0x00007fd4a982e000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007fd4a962a000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007fd4a9424000)
	libcgmanager.so.0 => /lib/x86_64-linux-gnu/libcgmanager.so.0 (0x00007fd4a9209000)
	libnih.so.1 => /lib/x86_64-linux-gnu/libnih.so.1 (0x00007fd4a8ff1000)
	libnih-dbus.so.1 => /lib/x86_64-linux-gnu/libnih-dbus.so.1 (0x00007fd4a8de7000)
	libdbus-1.so.3 => /lib/x86_64-linux-gnu/libdbus-1.so.3 (0x00007fd4a8ba2000)

floe, 2015-10-07:
Plug the Kinect in, run `lsusb`, and post the output. Make sure it's plugged into an USB3 port.

Chinkalp7, 2015-10-07:
Bus 004 Device 002: ID 8087:8000 Intel Corp. 
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 8087:8008 Intel Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 064e:9301 Suyin Corp. 
Bus 001 Device 003: ID 045e:02d9 Microsoft Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Chinkalp7, 2015-10-07:
In windows it works in same port i think i am connecting it in right usb 3.0 port.

xlz, 2015-10-08:
There should be 3 "Microsoft Corp." devices for a Kinect 2.

Chinkalp7, 2015-10-08:
Is there anything i can do to resolve his?

xlz, 2015-10-08:
Maybe check the adaptor? It's easy to overlook some connectors on it.

floe, 2015-11-09:
Is this fixed with 0.1-rc1? If not, please re-open.

closed 2015-11-09 by floe