OpenKinect/libfreenect2 #418: Could NOT find TurboJPEG (missing: TURBOJPEG_WORKS)

DarthSomebody, 2015-10-07:
When I try to cmake libfreenect2, I get the error "Could NOT find TurboJPEG (missing: TURBOJPEG_WORKS)".

However, TurboJPEG IS there. I checked the paths for the libs and include files and can't find out why he can't find it. I also tried to cmake it with different TurboJPEG versions, but none of them worked.

This is the CMakeError.log:
```
Performing C SOURCE FILE Test TURBOJPEG_WORKS failed with the following output:
Change Dir: D:/Documents/Arbeit/Libraries/libfreenect2/build/CMakeFiles/CMakeTmp

Run Build Command:"C:/Program Files (x86)/MSBuild/12.0/bin/MSBuild.exe" "cmTryCompileExec2473982567.vcxproj" "/p:Configuration=Debug" "/p:VisualStudioVersion=12.0"
Microsoft (R) Build Engine version 12.0.40629.0

[Microsoft .NET Framework, version 4.0.30319.34209]

Copyright (C) Microsoft Corporation. All rights reserved.

Build started 07.10.2015 15:11:56.

Project "D:\Documents\Arbeit\Libraries\libfreenect2\build\CMakeFiles\CMakeTmp\cmTryCompileExec2473982567.vcxproj" on node 1 (default targets).

PrepareForBuild:

  Creating directory "cmTryCompileExec2473982567.dir\Debug\".

  Creating directory "D:\Documents\Arbeit\Libraries\libfreenect2\build\CMakeFiles\CMakeTmp\Debug\".

  Creating directory "cmTryCompileExec2473982567.dir\Debug\cmTryCom.F68A2579.tlog\".

InitializeBuildStatus:

  Creating "cmTryCompileExec2473982567.dir\Debug\cmTryCom.F68A2579.tlog\unsuccessfulbuild" because "AlwaysCreate" was specified.

ClCompile:

  D:\Programs\Visual Studio Ultimate 2013\VC\bin\CL.exe /c /I"D:\Documents\Arbeit\Libraries\libfreenect2\depends\libjpeg-turbo64\include" /Zi /W3 /WX- /Od /Ob0 /Oy- /D WIN32 /D _WINDOWS /D TURBOJPEG_WORKS /D _DEBUG /D "CMAKE_INTDIR=\"Debug\"" /D _MBCS /Gm- /RTC1 /MT /GS /fp:precise /Zc:wchar_t /Zc:forScope /Fo"cmTryCompileExec2473982567.dir\Debug\\" /Fd"cmTryCompileExec2473982567.dir\Debug\vc120.pdb" /Gd /TC /analyze- /errorReport:queue D:\Documents\Arbeit\Libraries\libfreenect2\build\CMakeFiles\CMakeTmp\src.c

  Microsoft (R) C/C++ Optimizing Compiler Version 18.00.40629 for x86

  Copyright (C) Microsoft Corporation.  All rights reserved.

  cl /c /I"D:\Documents\Arbeit\Libraries\libfreenect2\depends\libjpeg-turbo64\include" /Zi /W3 /WX- /Od /Ob0 /Oy- /D WIN32 /D _WINDOWS /D TURBOJPEG_WORKS /D _DEBUG /D "CMAKE_INTDIR=\"Debug\"" /D _MBCS /Gm- /RTC1 /MT /GS /fp:precise /Zc:wchar_t /Zc:forScope /Fo"cmTryCompileExec2473982567.dir\Debug\\" /Fd"cmTryCompileExec2473982567.dir\Debug\vc120.pdb" /Gd /TC /analyze- /errorReport:queue D:\Documents\Arbeit\Libraries\libfreenect2\build\CMakeFiles\CMakeTmp\src.c

  src.c

Link:

  D:\Programs\Visual Studio Ultimate 2013\VC\bin\link.exe /ERRORREPORT:QUEUE /OUT:"D:\Documents\Arbeit\Libraries\libfreenect2\build\CMakeFiles\CMakeTmp\Debug\cmTryCompileExec2473982567.exe" /INCREMENTAL /NOLOGO kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib "D:\Documents\Arbeit\Libraries\libfreenect2\depends\libjpeg-turbo64\lib\turbojpeg.lib" /MANIFEST /MANIFESTUAC:"level='asInvoker' uiAccess='false'" /manifest:embed /DEBUG /PDB:"D:/Documents/Arbeit/Libraries/libfreenect2/build/CMakeFiles/CMakeTmp/Debug/cmTryCompileExec2473982567.pdb" /SUBSYSTEM:CONSOLE /TLBID:1 /DYNAMICBASE /NXCOMPAT /IMPLIB:"D:/Documents/Arbeit/Libraries/libfreenect2/build/CMakeFiles/CMakeTmp/Debug/cmTryCompileExec2473982567.lib" /MACHINE:X86 /SAFESEH  /machine:X86 /debug cmTryCompileExec2473982567.dir\Debug\src.obj

src.obj : error LNK2019: unresolved external symbol _tjInitCompress referenced in function _main [D:\Documents\Arbeit\Libraries\libfreenect2\build\CMakeFiles\CMakeTmp\cmTryCompileExec2473982567.vcxproj]

D:\Documents\Arbeit\Libraries\libfreenect2\build\CMakeFiles\CMakeTmp\Debug\cmTryCompileExec2473982567.exe : fatal error LNK1120: 1 unresolved externals [D:\Documents\Arbeit\Libraries\libfreenect2\build\CMakeFiles\CMakeTmp\cmTryCompileExec2473982567.vcxproj]

Done Building Project "D:\Documents\Arbeit\Libraries\libfreenect2\build\CMakeFiles\CMakeTmp\cmTryCompileExec2473982567.vcxproj" (default targets) -- FAILED.

Build FAILED.

"D:\Documents\Arbeit\Libraries\libfreenect2\build\CMakeFiles\CMakeTmp\cmTryCompileExec2473982567.vcxproj" (default target) (1) ->

(Link target) ->

  src.obj : error LNK2019: unresolved external symbol _tjInitCompress referenced in function _main [D:\Documents\Arbeit\Libraries\libfreenect2\build\CMakeFiles\CMakeTmp\cmTryCompileExec2473982567.vcxproj]

  D:\Documents\Arbeit\Libraries\libfreenect2\build\CMakeFiles\CMakeTmp\Debug\cmTryCompileExec2473982567.exe : fatal error LNK1120: 1 unresolved externals [D:\Documents\Arbeit\Libraries\libfreenect2\build\CMakeFiles\CMakeTmp\cmTryCompileExec2473982567.vcxproj]

    0 Warning(s)

    2 Error(s)

Time Elapsed 00:00:00.37

Source file was:
#include <turbojpeg.h>
int main(void) { tjhandle h=tjInitCompress(); return 0; }
```

elliotwoods, 2015-10-08:
getting the same error here.
trying to configure with visual studio 2015 x64

xlz, 2015-10-08:
In principle I should be looking at this. But I don't have a Windows machine right now and I don't see anything obviously wrong.

Your `link.exe` did include `"D:\Documents\Arbeit\Libraries\libfreenect2\depends\libjpeg-turbo64\lib\turbojpeg.lib"`, and `turbojpeg.lib` does include `__imp_tjInitCompress` `tjInitCompress`.

Can you try it with `-MT` removed in https://github.com/OpenKinect/libfreenect2/blob/master/cmake_modules/FindTurboJPEG.cmake?

DarthSomebody, 2015-10-08:
Thank you for your response,
Unfortunately, removing -MT didn't do anything.

floe, 2015-10-09:
Is this maybe something trivial, related to spaces in directory names?

DarthSomebody, 2015-10-09:
Thank you! It was something trivial on top of something non-trivial.
Removing -MT in FindTurboJPEG.cmake actually worked. I just forgot to delete the existing files generated by cmake in the build directory.

I was able to build the library now, thanks.

xlz, 2015-10-09:
@larshg 
`-MT` was your addition. Can you clarify the purpose of adding this?

`-MT` is for static linking, OK. I'm not sure what went wrong for this user, but I didn't have the same problem previously.

larshg, 2015-10-09:
I can see it was originally (win32) condition and I think I have only used x64 bit, so this must be an error for the x86 build :/

xlz, 2015-10-09:
@larshg 
Is it OK to remove this snippet altogether?

larshg, 2015-10-09:
Yes, I don't think it will do any harm.

xlz, 2015-10-13:
I am able to reproduce a similar but different error.

I install libjpeg-turbo64 to a random location and first run `cmake ..` without setting TurboJPEG_ROOT, and then rerun `cmake ..` after setting TurboJPEG_ROOT. Then I would have the error of TURBOJPEG_WORKS. But in my error, there is no `unresolved external symbol _tjInitCompress`.

`-MT` does not have any effect.

So any further report about this should make sure they clean the build directory before reconfiguring.

closed 2015-11-06 by floe

sajivthomas, 2015-11-26:
I am running into this issue even now following the steps here https://github.com/OpenKinect/libfreenect2. See logs below. Please advice.

C:\k4w\libfreenect2-0.1-rc1>mkdir build && cd build

C:\k4w\libfreenect2-0.1-rc1\build>cmake .. -G "Visual Studio 12 2013 Win64"
-- The C compiler identification is MSVC 18.0.21005.1
-- The CXX compiler identification is MSVC 18.0.21005.1
-- Check for working C compiler using: Visual Studio 12 2013 Win64
-- Check for working C compiler using: Visual Studio 12 2013 Win64 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler using: Visual Studio 12 2013 Win64
-- Check for working CXX compiler using: Visual Studio 12 2013 Win64 -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- using tinythread as threading library
-- Could NOT find PkgConfig (missing:  PKG_CONFIG_EXECUTABLE)
-- Found LibUSB: C:/k4w/libfreenect2-0.1-rc1/depends/libusb/MS64/dll/libusb-1.0.lib
CMake Error at C:/Program Files (x86)/CMake/share/cmake-3.4/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find TurboJPEG (missing: TurboJPEG_LIBRARIES TURBOJPEG_WORKS)
Call Stack (most recent call first):
  C:/Program Files (x86)/CMake/share/cmake-3.4/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  cmake_modules/FindTurboJPEG.cmake:66 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:69 (FIND_PACKAGE)

-- Configuring incomplete, errors occurred!
See also "C:/k4w/libfreenect2-0.1-rc1/build/CMakeFiles/CMakeOutput.log".

xlz, 2015-11-26:
@sajivthomas First, yours is not the same issue.

CMake could not find turbojpeg.lib. Where did you install turbojpeg?

sajivthomas, 2015-11-26:
Here - C:\k4w\libfreenect2-0.1-rc1\depends\libjpeg-turbo64

there is a lib folder and it has these files:
libjpeg.a
libjpeg.dll.a
libturbojpeg.a
libturbojpeg.dll.a

xlz, 2015-11-26:
You downloaded libjpeg-turbo-1.4.2-gcc64.exe? You need this one: libjpeg-turbo-1.4.2-vc64.exe.

sajivthomas, 2015-11-26:
Thanks that worked.

rmkemker, 2016-01-06:
I am also having some issues and none of the above remedies seem to apply.  This is the error message:

> Performing Test TURBOJPEG_WORKS
Performing Test TURBOJPEG_WORKS - Failed
CMake Error at C:/Program Files (x86)/CMake/share/cmake-3.4/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find TurboJPEG (missing: TURBOJPEG_WORKS)
Call Stack (most recent call first):
  C:/Program Files (x86)/CMake/share/cmake-3.4/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  cmake_modules/FindTurboJPEG.cmake:66 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:69 (FIND_PACKAGE)

I have installed libjpeg-turbo-1.4.2-vc64.exe and the -MT tag is not present in the cmake file.  Any help would be nice...installing on Windows 8.

sajivthomas, 2016-01-06:
Is the environment variable TurboJPEG_ROOT set to the path where you installed TurboJPEG. I have it here depends/libjpeg-turbo64 , in fact libusb, turbojpeg and glfw are all in the depends folder.

rmkemker, 2016-01-06:
All of mine are in the depends folder.  I did try your method, but that didn't seem to help.  cmake is finding the LIB, DLL, and INCLUDE path, but it just fails on that test.  I could arbitrarily set the variable manually, but I fear that it might be something that keeps my program from functioning.

rmkemker, 2016-01-06:
I went to the error log and this is what it said:

>"C:\libfreenect2-master\build\CMakeFiles\CMakeTmp\cmTC_dc575.vcxproj" (default target) (1) ->

>(Link target) -> 

>  C:\libfreenect2-master\depends\libjpeg-turbo64\lib\turbojpeg.lib : warning LNK4272: library machine type 'x64' conflicts with target machine type 'X86' [C:\libfreenect2-master\build\CMakeFiles\CMakeTmp\cmTC_dc575.vcxproj]

This doesn't make sense since my machine is x64.

rmkemker, 2016-01-06:
I figured it out...I was using the wrong compiler in cmake.  It defaulted to the x86 compiler instead of the x64 compiler.  Good to go...thanks!

xlz, 2016-01-06:
Yep, a result of not following the instructions.

diksito, 2017-02-01:
I got similar problem and thanks to this instructions I managed to build it successfully - http://stackoverflow.com/questions/27388699/cmake-find-package-cant-find-my-mingw-libjpeg 

> You can specify the location of lib JPEG by specifying JPEG_LIBRARY and JPEG_INCLUDE_DIR via -DJPEG_LIBRARY:PATH=C:/MinGW/lib/libjpeg.a -DJPEG_INCLUDE_DIR:PATH=C:/MinGW/include/ when you run cmake (or add these variables using the CMake GUI)

kayzersoze26, 2018-02-05:
Can you help me getting an error like below?

-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   thread
--   filesystem
--   regex
--   chrono
--   date_time
--   atomic
-- Found gflags  (include: /usr/include, library: /usr/lib/x86_64-linux-gnu/libgflags.so)
-- Found glog    (include: /usr/include, library: /usr/lib/x86_64-linux-gnu/libglog.so)
-- Found PROTOBUF Compiler: /usr/bin/protoc
-- Found lmdb    (include: /usr/include, library: /usr/lib/x86_64-linux-gnu/liblmdb.so)
-- Found LevelDB (include: /usr/include, library: /usr/lib/x86_64-linux-gnu/libleveldb.so)
-- Found Snappy  (include: /usr/include, library: /usr/lib/x86_64-linux-gnu/libsnappy.so)
CMake Error at /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find JPEGTurbo (missing: JPEGTurbo_LIBRARIES)
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  cmake/Modules/FindJPEGTurbo.cmake:18 (find_package_handle_standard_args)
  cmake/Dependencies.cmake:57 (find_package)
  CMakeLists.txt:66 (include)

-- Configuring incomplete, errors occurred!
See also "/home/salih/caffe/build/CMakeFiles/CMakeOutput.log".
See also "/home/salih/caffe/build/CMakeFiles/CMakeError.log".

Fatemeh-MA, 2019-08-08:
I think I found the solution, I had the same problem. you need to use "libjpeg-turbo-vc" version instead of "libjpeg-turbo-gcc".  it works for me. 
best luck

commit 6a83286f (2015-11-06): cmake: Fix rebuilding error with stale cache files

check_c_source_compiles would generate wrong files if the user
does not set correct variables initially even given correct values
later. Protect against this scenario.

This should fix #418.

Also remove "-MT" flags for MSVC which seems to do no good here.