OpenKinect/libfreenect2 #419: Color pixel to depth coordinates

ERHSQuizbowl, 2015-10-07:
Hi all,
  
 Does anyone have a method for converting the coordinates of the pixel(x,y) in Frame::Color to the coordinates of a point(x,y,z) in  Frame::Depth? I would really appreciate it if someone would share such working code.
  
Thanks

floe, 2015-10-08:
The `Registration` class does exactly that, see `Protonect.cpp` for an example of how to use it.

ERHSQuizbowl, 2015-10-08:
Sorry if I was not clear. What I meant was take to take a single point in the color frame and transform that to the depth frame. My project requires that do analysis on an RGB image and transform a specific point in the region to world coordinates.

xlz, 2015-10-08:
Registered color image is of the same coordinate as the depth image. No transform is needed.

floe, 2015-10-09:
Also, check out the `Registration::getPointXYZRGB` function, which provides world coordinates for a depth pixel.

pookiefoof, 2015-12-09:
I think what @ERHSQuizbowl required is registering the depth frame to the RGB frame (what libfreenect2 has now is RGB -> depth). I noticed the RGB -> depth registration is done by some "magical" tables, so what I am wondering is whether it is possible to do the depth -> RGB side?

xlz, 2015-12-09:
We don't know how to do depth to rgb registration given the builtin parameters. The builtin ones use a different formulation than the standard matrix transformation, see https://github.com/OpenKinect/libfreenect2/issues/223.

Also, those are not magic tables, just some batched together values of kinect2_color_to_depth().

floe, 2015-12-10:
You can get a reasonable approximation of depth-to-RGB if you use the optional `bigdepth` parameter to `Registration::apply(...)`.

closed 2016-01-27 by xlz

se7oluti0n, 2016-11-11:
Hi, I think you misunderstood Question.
He has a rgb pixel on **RAW** rgb image (not registered image), he want to get corresponding pixel on depth image or 3d space.