OpenKinect/libfreenect2 #420: [Question] Is there any hope for eventually adding exposure control to the color camera?

chipweinberger, 2015-10-08:


chipweinberger, 2015-10-08:
It looks like exposure is supported in the SDK so that is a good sign. Taken from the header file:

/* [propget] */ HRESULT ( STDMETHODCALLTYPE *get_ExposureTime )( 
            IColorCameraSettings * This,
            /* [annotation][out][retval] */ 
            _Out_  TIMESPAN *exposureTime);
        
        /* [propget] */ HRESULT ( STDMETHODCALLTYPE *get_FrameInterval )( 
            IColorCameraSettings * This,
            /* [annotation][out][retval] */ 
            _Out_  TIMESPAN *frameInterval);
        
        /* [propget] */ HRESULT ( STDMETHODCALLTYPE *get_Gain )( 
            IColorCameraSettings * This,
            /* [annotation][out][retval] */ 
            _Out_  float *gain);
        
        /* [propget] */ HRESULT ( STDMETHODCALLTYPE *get_Gamma )( 
            IColorCameraSettings * This,
            /* [annotation][out][retval] */ 
            _Out_  float *gamma);

HenningJ, 2015-10-08:
The official SDK only supports reading of the camera settings. So, most likely, that's the most we can hope for here.

xlz, 2015-10-08:
> The color exposure is fixed function of the camera, so changing exposure cannot be controlled by software. It is a firmware change.

-- Carmine Sirignano - MSFT

chipweinberger, 2015-10-09:
Rats!!!

closed 2016-01-27 by xlz