OpenKinect/libfreenect2 #421: modprobe: ERROR: could not insert 'fglrx': No such device

wkentaro, 2015-10-09:
I followed the installation instruction described at
https://github.com/OpenKinect/libfreenect2#debianubuntu-1404-perhaps-earlier

and had error like below:

```
% ./bin/Protonect
modprobe: ERROR: could not insert 'fglrx': No such device
Error: Fail to load fglrx kernel module!
Error! Fail to load fglrx kernel module! Maybe you can switch to root user to load kernel module directly
modprobe: ERROR: could not insert 'fglrx': No such device
Error: Fail to load fglrx kernel module!
Error! Fail to load fglrx kernel module! Maybe you can switch to root user to load kernel module directly
modprobe: ERROR: could not insert 'fglrx': No such device
Error: Fail to load fglrx kernel module!
Error! Fail to load fglrx kernel module! Maybe you can switch to root user to load kernel module directly
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 12 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:21 with serial 502489742542
[Info] [Freenect2Impl] found 1 devices
[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65543 The requested client API version is unavailable
[Error] [OpenGLDepthPacketProcessor] Failed to create opengl window.
```

do you have any idea to resolve this error?

floe, 2015-10-09:
What kind of graphics card do you have? If it's an AMD card, you should first check if it's actually installed properly and OpenGL works. See Google for multiple tutorials.

wkentaro, 2015-10-09:
I think that is not AMD.
```
% lspci -v | grep -i vga
00:02.0 VGA compatible controller: Intel Corporation 3rd Gen Core processor Graphics Controller (rev 09) (prog-if 00 [VGA controller])
01:00.0 VGA compatible controller: NVIDIA Corporation GK107GLM [Quadro K2000M] (rev a1) (prog-if 00 [VGA controller])
```

xlz, 2015-10-09:
`ERROR: could not insert 'fglrx': No such device` is irrelevant.

`GLFW error 65543 The requested client API version is unavailable` says your OpenGL version is too low. What is your `glxinfo | grep OpenGL`?

wkentaro, 2015-10-10:
that is fixed after I reinstalled nvidia-346 and nvidia-settings.
thanks.

closed 2015-10-10 by wkentaro