OpenKinect/libfreenect2 #422: Error building Protonect

tszajner, 2015-10-12:
Hello, 

When I try to build the Protonect executable, I get the following error when using cmake ..
:~/libfreenect2/examples/build$ cmake ..
CMake Error at CMakeLists.txt:33 (FIND_PACKAGE):
  By not providing "Findfreenect2.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "freenect2", but CMake did not find one.

  Could not find a package configuration file provided by "freenect2" with
  any of the following names:

    freenect2Config.cmake
    freenect2-config.cmake

  Add the installation prefix of "freenect2" to CMAKE_PREFIX_PATH or set
  "freenect2_DIR" to a directory containing one of the above files.  If
  "freenect2" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/tom/libfreenect2/examples/build/CMakeFiles/CMakeOutput.log".

I'm using Ubuntu 15.04. Have I missed some installation steps?

xlz, 2015-10-12:
Why are you building in libfreenect2/examples/build? See the docs.

tszajner, 2015-10-12:
D'oh! Yes, that worked! Thank you for your help!

closed 2015-10-12 by tszajner