OpenKinect/libfreenect2 #424: Interference between multiple Kinects

dtytgat, 2015-10-13:
I've been experimenting with using multiple Kinects (4 atm) that are all pointed at the same object, but viewed at different viewing angles (with an overlap). This all works fine, however there is quite alot of interference when doing so. The depth measurements seems to "pull" away from the camera in a periodic nature. 

I was wondering if this is a known issue, and if there is some way to resolve it. Detecting faulty values would be one solution, however even better would be a solution that prevents this from happening. My gut feeling is that it has something to do with the phase of the different IR transmitters; however I have no idea how to resolve it if that is indeed the case.

I've read on a forum somewhere (sorry but I can't retrieve the reference atm) that interference between multiple kinect v2's should be limited to dropping a frame once in a while. This is not what I'm seeing however. Might this "frame drop" be something that is not yet in libfreenect2 but is in the Windows Kinect SDK? 

Thanks.

HenningJ, 2015-10-13:
This happens with the Microsoft SDK as well and the Microsoft SDK doesn't detect it either. There are several threads about that in the official forum (e.g. [this one](https://social.msdn.microsoft.com/Forums/en-US/8e2233b6-3c4f-485b-a683-6bacd6a74d53/how-to-prevent-interference-between-multiple-kinect-v2-sensors?forum=kinectv2sdk)), but no solution.

What happens (as I understood it) is this: Each Kinect v2 continuously, more or less randomly switches between different modulation frequencies. When two Kinects switch to the same frequency range, the interference occurs. They typically gradually drift into the same range and after a while drift out of that range again. So, theoretically, you just have to wait a bit until the interference is gone. The only way I found to stop the interference immediately was to disconnect (and reconnect) the concerned Kinect from its power supply. 

But I only used 2 Kiencts at a time. I guess using 4 Kinects, you get interference so often, you can't just wait it out.

There has been talk of a beta firmware that allows for some control over the modulation frequencies, but it's unlikely that anything official will be announced anytime soon (or maybe ever).

dtytgat, 2015-10-13:
Thanks HenningJ for the good explanation and pointers. Quite unfortunate that these modulation frequencies aren't controllable at this time... Let's hope MS surprises us with that custom firmware.

Edwinzero, 2015-11-01:
I think there are couples of articles mentioned that Kinect v2 has already solved the interference by improving the hardware design. So you don't need to do some work to reduce interference like attaching a motor to generate vibration with synchronized frequency. But since I am working on huge number of Kinect v2 capturing, I think you can set the two camera has a 50 cm  (+5 or -5 cm) distance along their orientation when you put them parallel to each other. For angle, the recommended angle I got from an article mentioned that for two Kinect v2 is [0, 45) && (45, 60] degrees. At that range you can have the minimal interference.
[image]

closed 2016-01-27 by xlz

wiegerthefarmer, 2016-04-22:
Has there been any update to this?  It's such a pain with multiple kinects... and just waiting till some random "good time" doesn't work...

dtytgat, 2016-04-27:
Unfortunately I haven't found a good solution up to this point...

bitandbytes, 2017-01-17:
@dtytgat: What is the procedure that you used to get data from **multiple Kinects v2**? 
Thank you very much in-advance.

dtytgat, 2017-01-17:
@bitandbytes We're using Fedora Linux with libfreenect2. 

Be sure to take into account the measures described here: https://github.com/OpenKinect/libfreenect2/wiki/Troubleshooting . As noted there, when using multiple Kinects you probably need to alter the USBFS memory size.

bitandbytes, 2017-01-18:
@dtytgat Thank you for the info. Troubleshooting page has good corner cases. 

One more thing, are you doing skeleton tracking? If so what do you use to do tracking?

abujelala, 2017-05-24:
@bitandbytes Did you figure out how to do skeleton tracking while using multiple Kinects?

bitandbytes, 2017-06-06:
@abujelala I couldn't do that with the libfreenect2 library on Windows. I also didn't work so hard to get it working too. Since my app was on Unity Engine and by the time this problem occurred and we have already done some development using Unity, we used multiple machines.  :P