OpenKinect/libfreenect2 #425: Unable to load DLL 'KernelBuilderAPI.dll':The specified procedure could not be found.(Exception from HRESULT:0x8007007F)

lishirui, 2015-10-13:
Hi, when I open the libfreenect2.sln using VS2013 after I finished CMake procedure.  I got this error in  VS2013 dialog.  It seems like the OpenCL cause to this error. When I uninstall the OpenCL this error disappear, But I can't compile the project. Have you meet this problem?

xlz, 2015-10-16:
It seems your OpenCL SDK is not correctly set up. I also used OpenCL SDK without this issue.

> When I uninstall the OpenCL this error disappear, But I can't compile the project.

You must regenerate the cmake files. `cmake .. _G "Visual Studio 12 2013 Win64"`.

lishirui, 2015-10-18:
You mean I can compile this project without OpenCL?

xlz, 2015-10-18:
With `cmake -DENABLE_OPENCL=OFF`.

lishirui, 2015-10-19:
Ok It does works. but when I run Protonect.exe I got this:
C:\Users\li.6833\Downloads\libfreenect2-master\libfreenect2-master\build\bin\Debug>Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 13 usb devices connected
[Error] [Freenect2Impl] failed to open Kinect v2: @2:22 LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform
[Info] [Freenect2Impl] found valid Kinect v2 @2:22 with serial 502792341942
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Error] [protocol::UsbControl] failed to claim interface with IrInterfaceId(=1)! LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform. Try debugging with environment variable: export LIBUSB_DEBUG=4 .
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Error] [Freenect2] failed to open Kinect v2: @2:22
failure opening device!
I don't know why...  do it support the Windows 10 platform?

xlz, 2015-10-19:
I have not tested Windows 10. What you saw maya be a bug. Can you do as the error message says, provide the log with LIBUSB_ERROR=4?

xlz, 2015-10-25:
The followed up win10 issue is to be merged to #448.

floe, 2015-11-06:
Should be fixed by #429, please re-test with latest master and re-open if necessary.

closed 2015-11-06 by floe

zybaugur, 2016-07-01:
I have the same problem, and here is the solution : https://software.intel.com/en-us/forums/opencl/topic/603665