OpenKinect/libfreenect2 #426: Registration from image instead of Frame

xionglingfeng, 2015-10-15:
I am going to register color/depth images, but `Registration::apply()` takes only `Frame` as input, not images.

It is because I previously recorded color/depth images from Kinect 2, the color images are stored as JPG/PNG images (`cv::imwrite()`) while depth images are directly written to raw format

	Mat depth(depthFrame->height, depthFrame->width, CV_32FC1, depthFrame->data);
	fwrite(depth->data,
		depth->elemSize(),
		depth->rows * depth->cols, fp
		);

I tried to read those files and create `Frame` objects from them:

	// Read depth image
	unsigned char *depthBuf[512*424*4];
	fread(depthBuf, 4,  512*424, fp);

	// Read color image
	cv::Mat rgbMat ;
	cv::imread("xxx.jpg").convertTo(rgbMat, CV_8UC4);

	// Build frames
	libfreenect2::Frame rgb(rgbMat.cols, rgbMat.rows, 4);
	libfreenect2::Frame depth(512, 424, 4);
	rgb.data = rgbMat.data;
	depth.data = (unsigned char*)depthBuf;

	// Register
	registration->apply(&rgb, &depth, &undistorted, &registered);

But `registered` is just an black image... :-(
Anything wrong?

xionglingfeng, 2015-10-15:
Solved. By checking the color image, I found the type of `rgbMat` is `CV_8UC3`, even I wrote
```C++
cv::imread("xxx.jpg").convertTo(rgbMat, CV_8UC4);
```
The correct way to converting it to `CV_8UC4` is invoking `cvtColor`.

Ref: http://stackoverflow.com/questions/18988910/opencv-imread-gives-cv-8uc3-cant-convert-to-cv-8uc4

closed 2015-10-15 by xionglingfeng

faisalbmc, 2016-05-18:
Hi xionglingfeng,
how do you visualize `undistorted` and `registered` images as you have shown? Are you showing them through `cv::imshow`? If yes, how you are transferring the values from `libfreenect2::Frame` to `cv::Mat`. 

I am doing the following way, is it correct?

```
            cv::Mat depthImage = cv::Mat(424, 512, CV_32FC1, cv::Scalar(0));
            depthImage = frames.frames["depth_filtered"].clone();

            libfreenect2::Frame freenectDepth(512, 424, 4);
            freenectDepth.data = depthImage.data;

            libfreenect2::Frame registeredDepth(512, 424, 4);
            libfreenect2::Frame undistortedDepth(512, 424, 4);
            libfreenect2::Registration *registration = new libfreenect2::Registration(m_depth_p, m_rgb_p);
            registration->apply(&freenectContourMask, &freenectDepth, &undistortedDepth, &registeredDepth);

            cv::Mat extractedRegisteredDepth = cv::Mat(424, 512, CV_32FC1, cv::Scalar(0));
            extractedRegisteredDepth.data = registeredDepth.data;
            cv::imshow("Extracted registered depth!", extractedRegisteredDepth);
```
I only get a black image for while visualizing `undistortedDepth` or `registeredDepth`.

faisalbmc, 2016-05-19:
Could you please tell me the type of the frames `undistorted` and `registered` are? What are the types of the `cv::Mat` objects in which you are transferring the data from `libfreenect2::Frame` respectively from `undistorted` and `registered`?

xionglingfeng, 2016-05-19:
@faisalbmc  Sorry for the late reply.
For frames,

	libfreenect2::Frame undistorted(depth.width, depth.height, depth.bytes_per_pixel);
	libfreenect2::Frame registered(depth.width, depth.height, depth.bytes_per_pixel);
	registration->apply(&rgb, &depth, &undistorted, &registered);

For the matrices,

	cv::Mat depth = cv::Mat(
			depthFrame->height, depthFrame->width,
			CV_32FC1, depthFrame->data);
	cv::Mat rgb = cv::Mat(
			rgbFrame->height, rgbFrame->width,
			CV_8UC4, rgbFrame->data);

The current version of the example (Protonet) is a little bit confusing in my opinion, because the some developers decided to remove dependency of OpenCV. Try to look for a old version (checkout the repo for some versions around Feb 2015).