OpenKinect/libfreenect2 #430: System integrity protection prevents copy of OpenCL .hpp file into the OpenCL Framework in Mac OS 10.11 (El Capitan)

flipphillips, 2015-10-16:
As the title says-

The 'system integrity protection' mechanisms prevent copying the (missing)`cl.hpp` into the OpenCL framework path, even as root. Starting with Mac OS 10.11 (El Capitan)

Two options-
1) tell folks to turn off integrity protection
2) put the file somewhere else / modify the path suitably in `opencl_depth_packet_processor.cpp`

I did (2) and all is well.

xlz, 2015-10-16:
PR #429 fixes this. A copy of cl.hpp is included in the repo.

flipphillips, 2015-10-16:
Excellent. Built / runs well otherwise, both OpenCL and GL versions of depth recovery. I'm not sophisticated-enough of a git user to figure out how to check out versions that are 'ahead' of the current version as suggested in #429, so if I ever figure that out I can try it against what is proposed there too.

xlz, 2015-10-16:
You can use this
```
git fetch origin pull/429/head:pr429
git checkout pr429
```

flipphillips, 2015-10-16:
Thank you! 
Learn something new every day!

flipphillips, 2015-10-16:
NB- builds and runs as of 10.11.1beta.

flipphillips, 2015-10-16:
Also, fwiw, in the Readme.md, the equivalent command for `ldd` is `otool -L` under Mac OS X, in case you'd like to add that.

closed 2015-10-16 by flipphillips