OpenKinect/libfreenect2 #431: Building with cl2.hpp (opencl-headers 2.x) instead of cl.hpp (opencl-headers 1.x)

lopho, 2015-10-17:
Cannot build with opencl-headers 2.x as it is not backwards compatible with 1.x.
2.0 ships in current debian testing as the only opencl-headers package.

Haven't looked into it much further yet.

Gist to cl2.hpp
https://gist.github.com/lopho/aa80c477518c65d05b86

Gist to cl.hpp -> cl2.hpp diff
https://gist.github.com/lopho/569aa7d9c1029a9d70d9

xlz, 2015-10-17:
PR #429 includes a copy of cl.hpp in here. Should not be a problem after #429.

lopho, 2015-10-17:
Sounds good. Lot's of things happening hidden in PRs. Got me by suprise :)

Any plans on programming against opencl 2 in the future?

xlz, 2015-10-17:
I don't think OpenCL 2.0 is anytime soon. Most users are with Intel beignet which is OpenCL 1.2. And we are still dealing with Nvidia's OpenCL 1.1.

closed 2015-10-17 by lopho