OpenKinect/libfreenect2 #432: Ubuntu 12.04 opencl_depth_packet_processor

mall3i, 2015-10-18:
Hey everyone,

I'm trying to setup libfreenect2 under Ubuntu 12.04 but I keep running into the following issue when executing the "make" command. I have an Nvidia graphics card and have the respective drivers and opencl components installed (I think).
Thankful for a push to the solutionary direction ;)

```
user@userU:~/libfreenect2/build$ make
[  8%] Built target generate_resources_tool
[  8%] Building CXX object CMakeFiles/freenect2.dir/src/opencl_depth_packet_processor.cpp.o
/home/user/libfreenect2/src/opencl_depth_packet_processor.cpp: In member function ‘bool libfreenect2::OpenCLDepthPacketProcessorImpl::initDevice(int)’:
/home/user/libfreenect2/src/opencl_depth_packet_processor.cpp:324:59: error: no matching function for call to ‘cl::Context::Context(cl::Device&, NULL, NULL, NULL, cl_int*)’
/home/user/libfreenect2/src/opencl_depth_packet_processor.cpp:324:59: note: candidates are:
/usr/include/CL/cl.hpp:1466:5: note: cl::Context::Context(const cl::Context&)
/usr/include/CL/cl.hpp:1466:5: note:   candidate expects 1 argument, 5 provided
/usr/include/CL/cl.hpp:1464:5: note: cl::Context::Context()
/usr/include/CL/cl.hpp:1464:5: note:   candidate expects 0 arguments, 5 provided
/usr/include/CL/cl.hpp:1443:5: note: cl::Context::Context(cl_device_type, cl_context_properties*, void (*)(const char*, const void*, size_t, void*), void*, cl_int*)
/usr/include/CL/cl.hpp:1443:5: note:   no known conversion for argument 1 from ‘cl::Device’ to ‘cl_device_type {aka long unsigned int}’
/usr/include/CL/cl.hpp:1420:5: note: cl::Context::Context(const std::vector<cl::Device>&, cl_context_properties*, void (*)(const char*, const void*, size_t, void*), void*, cl_int*)
/usr/include/CL/cl.hpp:1420:5: note:   no known conversion for argument 1 from ‘cl::Device’ to ‘const std::vector<cl::Device>&’
/home/user/libfreenect2/src/opencl_depth_packet_processor.cpp: In member function ‘bool libfreenect2::OpenCLDepthPacketProcessorImpl::buildProgram(const string&)’:
/home/user/libfreenect2/src/opencl_depth_packet_processor.cpp:549:42: error: no matching function for call to ‘cl::Program::build(const char*)’
/home/user/libfreenect2/src/opencl_depth_packet_processor.cpp:549:42: note: candidate is:
/usr/include/CL/cl.hpp:2459:12: note: cl_int cl::Program::build(const std::vector<cl::Device>&, const char*, void (*)(cl_program, void*), void*) const
/usr/include/CL/cl.hpp:2459:12: note:   no known conversion for argument 1 from ‘const char*’ to ‘const std::vector<cl::Device>&’
make[2]: *** [CMakeFiles/freenect2.dir/src/opencl_depth_packet_processor.cpp.o] Error 1
make[1]: *** [CMakeFiles/freenect2.dir/all] Error 2
make: *** [all] Error 2
```

xlz, 2015-10-19:
How did you install cmake 2.18.12.2 with Ubuntu 12.04? I could only find 2.8.7 with Ubuntu precise. And with that I could not even configure it.

```
CMake Error at CMakeLists.txt:1 (CMAKE_MINIMUM_REQUIRED):
  CMake 2.8.12.1 or higher is required.  You are running version 2.8.7

-- Configuring incomplete, errors occurred!
```

mall3i, 2015-10-25:
I installed a cmake backport for ubuntu 12.04 but you should also be able to build a current cmake from source.

xlz, 2015-10-25:
I don't think we will support Ubuntu 12.04. Almost all dependencies used by libfreenect2 (kernel, beignet, glfw, grahpics stack, libusb) are seriously outdated on Ubuntu 12.04.

mall3i, 2015-10-25:
Okay, but in this post it is mentioned that it works on Hydro, 12.04 and that post is "only" one year old :/

xlz, 2015-10-25:
Which post?

xlz, 2015-10-25:
I could provide a list of dependencies for Ubuntu 12.04, but libfreenect2 can't solve your graphics driver problem.

I couldn't even test your issue because I couldn't find a place to install cmake 2.8.12.2 without messing up a whole lot of system dependencies. If you somehow managed to install it, you have already replaced a fair amount of packages with unsupported version.

About your specific warnings, I guess it is because you have outdated cl.hpp. `dpkg -S cl.hpp` to find out where it comes from. There is also a problem about whether you are using Intel graphics or Nvidia graphics for OpenCL. If you want to hear I can paste more docs here.

xlz, 2015-10-25:
In general, I do hope Ubuntu 12.04 can use libfreenect2. The problem is the required versions of packages just do not exist in Ubuntu 12.04. The only thing to do is update the packages. Build those required versions of packages, either build from source or import packages from later Ubuntu releases (which breaks the whole system if not careful). If you have these dependencies, libfreenect2 works as-is; if not, there is nothing libfreenect2 can do about them.

The dependencies are:
* kernel 3.16+
* libusb 1.0.20+
* libglfw3-dev 3.0+
* beignet-dev 1.0+

/rant

xlz, 2015-10-26:
As a temporary workaround, you can disable OpenCL during building. `-DENABLE_OPENCL=OFF`.

I think the only offending part is OpenCL. The rest of the dependencies might be easier to provide or backport to Ubuntu 12.04.

floe, 2015-11-09:
I'd say it's hard enough to support 14.04+, OSX 10.9+ and Win7+. I'm afraid this would lead to a FrankenUbuntu (?) otherwise :-)

closed 2015-11-09 by floe