OpenKinect/libfreenect2 #433: OpenCl clGetDeviceIDs(-1) error!

shamoorti, 2015-10-19:
Does anyone know how to fix this problem?
[OpenCLDepthPacketProcessor::init] ERROR: clGetDeviceIDs(-1)
terminate called after throwing an instance of 'cl::Error'
  what():  clGetDeviceIDs
Aborted (core dumped)
¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤¤4
By the way I my CLquery is like this: 
lDeviceQuery Starting...

3 OpenCL Platforms found

 CL_PLATFORM_NAME: 	Experiment Intel Gen OCL Driver
 CL_PLATFORM_VERSION: 	OpenCL 1.1 beignet 0.3
OpenCL Device Info:

 No devices found supporting OpenCL (return code -1)

clDeviceQuery, Platform Name = Experiment Intel Gen OCL Driver, Platform Version = OpenCL 1.1 beignet 0.3, NumDevs = 0
 CL_PLATFORM_NAME: 	Intel Gen OCL Driver
 CL_PLATFORM_VERSION: 	OpenCL 1.2 beignet 1.1.1
OpenCL Device Info:

 1 devices found supporting OpenCL on: Intel Gen OCL Driver

 ----------------------------------
 Device Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3
 ---------------------------------
  CL_DEVICE_NAME: 			Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3
  CL_DEVICE_VENDOR: 			Intel
  CL_DRIVER_VERSION: 			1.1.1
  CL_DEVICE_TYPE:			CL_DEVICE_TYPE_GPU
  CL_DEVICE_MAX_COMPUTE_UNITS:		48
  CL_DEVICE_MAX_WORK_ITEM_DIMENSIONS:	3
  CL_DEVICE_MAX_WORK_ITEM_SIZES:	512 / 512 / 512 
  CL_DEVICE_MAX_WORK_GROUP_SIZE:	512
  CL_DEVICE_MAX_CLOCK_FREQUENCY:	1000 MHz
  CL_DEVICE_ADDRESS_BITS:		32
  CL_DEVICE_MAX_MEM_ALLOC_SIZE:		1024 MByte
  CL_DEVICE_GLOBAL_MEM_SIZE:		2048 MByte
  CL_DEVICE_ERROR_CORRECTION_SUPPORT:	no
  CL_DEVICE_LOCAL_MEM_TYPE:		global
  CL_DEVICE_LOCAL_MEM_SIZE:		64 KByte
  CL_DEVICE_MAX_CONSTANT_BUFFER_SIZE:	131072 KByte
  CL_DEVICE_QUEUE_PROPERTIES:		CL_QUEUE_PROFILING_ENABLE
  CL_DEVICE_IMAGE_SUPPORT:		1
  CL_DEVICE_MAX_READ_IMAGE_ARGS:	128
  CL_DEVICE_MAX_WRITE_IMAGE_ARGS:	8

  CL_DEVICE_IMAGE <dim>			2D_MAX_WIDTH	 8192
					2D_MAX_HEIGHT	 8192
					3D_MAX_WIDTH	 8192
					3D_MAX_HEIGHT	 8192
					3D_MAX_DEPTH	 2048
  CL_DEVICE_PREFERRED_VECTOR_WIDTH_<t>	CHAR 16, SHORT 8, INT 4, FLOAT 2, DOUBLE 4

clDeviceQuery, Platform Name = Experiment Intel Gen OCL Driver, Platform Version = OpenCL 1.1 beignet 0.3, NumDevs = 0
Intel Gen OCL Driver, Platform Version = OpenCL 1.2 beignet 1.1.1, NumDevs = 1, Device = Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3
*** Error in `./clDeviceQuery': double free or corruption (fasttop): 0x0000000001a267f0 ***
Aborted (core dumped)

xlz, 2015-10-19:
You did not follow the README.

> Install beignet-dev 1.0+

shamoorti, 2015-10-20:
Hey xlz,
I have installed beignet and also ran all the test with only two failed. As you can see that beignet 0.3 finds no Devs and I don't know where is coming from. Anyhow beignet 1.0 had some problems so I ran beignet 1.1 and it worked with the test,
/home/buttler/beignet/build/utests/utest_run
platform number 1
platform_profile "FULL_PROFILE"
platform_name "Intel Gen OCL Driver"
platform_vendor "Intel"
platform_version "OpenCL 1.2 beignet 1.1.1"
platform_extensions "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_spir cl_khr_icd"
device_profile "FULL_PROFILE"
device_name "Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3"
device_vendor "Intel"
device_version "OpenCL 1.2 beignet 1.1.1"
device_extensions "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_fp16 cl_khr_spir cl_khr_icd"
device_opencl_c_version "OpenCL C 1.2 beignet 1.1.1"
23 image formats are supported
[CL_R CL_UNORM_INT8]
[CL_R CL_UNORM_INT16]
[CL_R CL_SIGNED_INT8]
[CL_R CL_SIGNED_INT16]
[CL_R CL_SIGNED_INT32]
[CL_R CL_UNSIGNED_INT8]
[CL_R CL_UNSIGNED_INT16]
[CL_R CL_UNSIGNED_INT32]
[CL_R CL_HALF_FLOAT]
[CL_R CL_FLOAT]
[CL_RG CL_UNORM_INT8]
[CL_RG CL_UNSIGNED_INT8]
[CL_RGBA CL_UNORM_INT8]
[CL_RGBA CL_UNORM_INT16]
[CL_RGBA CL_SIGNED_INT8]
[CL_RGBA CL_SIGNED_INT16]
[CL_RGBA CL_SIGNED_INT32]
[CL_RGBA CL_UNSIGNED_INT8]
[CL_RGBA CL_UNSIGNED_INT16]
[CL_RGBA CL_UNSIGNED_INT32]
[CL_RGBA CL_HALF_FLOAT]
[CL_RGBA CL_FLOAT]
[CL_BGRA CL_UNORM_INT8]
test_load_program_from_bin_file()    [SUCCESS]
enqueue_built_in_kernels()    [SUCCESS]
builtin_acos_float()    [SUCCESS]
builtin_acos_float2()    [SUCCESS]
builtin_acos_float4()    [SUCCESS]
builtin_acos_float8()    [SUCCESS]
builtin_acos_float16()    [SUCCESS]
builtin_acosh_float()    [SUCCESS]
builtin_acosh_float2()    [SUCCESS]
builtin_acosh_float4()    [SUCCESS]
builtin_acosh_float8()    [SUCCESS]
builtin_acosh_float16()    [SUCCESS]
builtin_acospi_float()    [SUCCESS]
builtin_acospi_float2()    [SUCCESS]
builtin_acospi_float4()    [SUCCESS]
builtin_acospi_float8()    [SUCCESS]
builtin_acospi_float16()    [SUCCESS]
builtin_asin_float()    [SUCCESS]
builtin_asin_float2()    [SUCCESS]
builtin_asin_float4()    [SUCCESS]
builtin_asin_float8()    [SUCCESS]
builtin_asin_float16()    [SUCCESS]
builtin_asinh_float()    [SUCCESS]
builtin_asinh_float2()    [SUCCESS]
builtin_asinh_float4()    [SUCCESS]
builtin_asinh_float8()    [SUCCESS]
builtin_asinh_float16()    [SUCCESS]
builtin_asinpi_float()    [SUCCESS]
builtin_asinpi_float2()    [SUCCESS]
builtin_asinpi_float4()    [SUCCESS]
builtin_asinpi_float8()    [SUCCESS]
[... 788 lines cut ...]
vload_test_char()    [SUCCESS]
vload_test_ushort()    [SUCCESS]
vload_test_short()    [SUCCESS]
vload_test_uint()    [SUCCESS]
vload_test_int()    [SUCCESS]
vload_test_float()    [SUCCESS]
runtime_use_host_ptr_buffer()    [SUCCESS]
runtime_alloc_host_ptr_buffer()    [SUCCESS]
compiler_get_sub_group_size()    [SUCCESS]
compiler_get_sub_group_id()    [SUCCESS]
compiler_sub_group_shuffle()    [SUCCESS]
compiler_overflow_add_ulong4()    [SUCCESS]
compiler_overflow_add_uint4()    [SUCCESS]
compiler_overflow_add_ushort4()    [SUCCESS]
compiler_overflow_add_uchar4()    [SUCCESS]
compiler_overflow_sub_uint4()    [SUCCESS]
runtime_climage_from_boname()    [SUCCESS]

summary:
----------
  total: 764
  run: 763
  pass: 761
  fail: 2
  pass rate: 0.997379

xlz, 2015-10-20:
I suggest you remove all beignet packages and build from source installations, and then follow the README and install beignet-dev from the ppa.

closed 2015-11-09 by floe